MarlinFirmware/Marlin #15786: [BUG] [bugfix-2.x] G29 ABL stops at step 6 of a 9 point probe

rgrugan, 2019-11-03:
### Description

The ever since commits starting in early Oct. the G29 does not probe the entire bed. Currently with a 3x3 grid the probe will get to step 6 of 9 and then just stop. No errors given just stops.
I am using bilinearand have also tried UBL with similar results.

Setup:
SKR v1.3 w/2208s
BLTouch v1.3

If I flash back to a firmware date of 9/11/19 or earlier it will run through the bilinear ABL process with no issues successfully probing all 9 points.

### Steps to Reproduce

1. G28
2. G29

**Expected behavior:** 
Complete ABL probe of entire bed. 3x3 grid so 9 points probed. 

**Actual behavior:** 
The ABL process halts at step 6 of 9 and acts like it has completed the G29 process. No errors shown or even a grid printed on the command line.

#### Additional Information
Configuration.h and Config_adv.h files in attached zip for review.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3801826/Marlin.zip)

hackebike, 2019-11-07:
Not sure if related but I cannot run ABL Bilinear (here the [bug](https://github.com/MarlinFirmware/Marlin/issues/15809) I opened), I check your configs and I are pretty much the same. Only difference is I am using sensorless homing and probing with 2209, but ABL bilinear is not working.

boelle, 2019-11-11:
@rgrugan still an issue?

rgrugan, 2019-11-12:
@boelle yes as of last week. Will dload the latest and try again.

rgrugan, 2019-11-12:
Can confirm that latest pull as of today 11/12/19 compiled and is executing properly in relation to ABL. It hits all 9 points with no issues now.

closed 2019-11-12 by rgrugan

rt1970, 2019-11-21:
"#define AUTO_BED_LEVELING_BILINEAR" works fine after uncomment "//#define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors." i think...
"#define AUTO_BED_LEVELING_UBL" stopps after point 5