MarlinFirmware/Marlin #15789: [BUG] "Motion -> Bed Leveling -> Level Corners" crashes after "Homing XYZ"

sustmi, 2019-11-03:
### Description
After merging PR https://github.com/MarlinFirmware/Marlin/pull/15760, the _Motion_ -> _Bed Leveling_ -> _Level Corners_ menu function hangs the printer (and causes E stepper to whistle) after _Homing XYZ_ step finishes.

The screen shows :
```
Next Point?
Homing XYZ

Done     **Next**
```

When I rollback the pull request, leveling works fine.

### Steps to Reproduce
1. Turn on the printer
2. Enter menu and go to _Motion_ -> _Bed Leveling_
3. Select _Level Corners_
4. Wait for Homing XYZ to finish

**Expected behavior:** Printer head should then move to the first leveling point.

**Actual behavior:** Printer head only moves up a bit, the menu is partially redrawn (still showing "Homing XYZ"), printer control wheel/button is unresponsive, E stepper starts to whistle and the only thing I can do is turn off the printer.

sustmi, 2019-11-03:
@thinkyhead: I tried to look at [your changes](https://github.com/MarlinFirmware/Marlin/pull/15760/files) and see where the problem is but the diff is too big for me to understand it.

The only difference I noticed is that `_lcd_level_bed_corners_homing()` used to call [`ui.goto_screen(menu_level_bed_corners);`](https://github.com/MarlinFirmware/Marlin/pull/15760/files#diff-f126349326f3f91ef5f520de974498e0L111) and now it contains directly the function body of `menu_level_bed_corners()`. But I do not understand the menu code enough to know whether it's a problem or not.

boelle, 2019-11-03:
will temp close this one until the information in issue template has been added

sustmi, 2019-11-03:
@boelle: Do you mean `Configuration.h` and `Configuration_adv.h` files? Or a video?
Which of these information are mandatory and which are optional?

boelle, 2019-11-03:
yes  `Configuration.h` and `Configuration_adv.h`

sustmi, 2019-11-03:
Thanks :+1: . I updated the issue and added a ZIP with those files.

Spartacus1066, 2019-11-03:
I logged in to report the SAME issue, Tevo Tornado here. Never had this happen... 
Printer resets if using Level Corners option, does fine if I disable steppers and do it maually. Mesh leveling also worked. Thanks!

[Tevo_Tornado_config_Spartacus.zip](https://github.com/MarlinFirmware/Marlin/files/3802236/Tevo_Tornado_config_Spartacus.zip)

Evg33, 2019-11-04:
SKR 1.3
https://youtu.be/V6w1NU9C0sk

Evg33, 2019-11-05:
https://github.com/MarlinFirmware/Marlin/pull/15808
Compilation: ok
Working: fail

manimac79, 2019-11-06:
Same issue here, SKR E3 mini 1.2

willy059, 2019-11-06:
Same issue, Geeetech A10.

shitcreek, 2019-11-07:
This is now fixed with this commit: https://github.com/MarlinFirmware/Marlin/commit/84ce3a6bea0d12815261ef7fa40bd2c08f2a5573

Evg33, 2019-11-07:
The problem is solved !
It works!
Thank you !

boelle, 2019-11-11:
will close this as fixed

closed 2019-11-11 by boelle