MarlinFirmware/Marlin #15790: M48 z probe repeatability test fails with "? (X,Y) out of bounds." unless bed leveling is enabled.

drmulligan, 2019-11-03:
M48 z probe repeatability test fails with "? (X,Y) out of bounds." unless bed leveling is enabled.  I have been using my BLTouch just as a z min probe while I work out some issues with my printer.  I wanted to do a probe repeatability test before making some changes but it fails echoing the message "? (X,Y) out of bounds." to the serial port.

Enabling UBL fixed the issue.

### Steps to Reproduce

1. With a z probe disable bed leveling.
2. M48

**Expected behavior:** 
Preferably perform the z probe repeatability test or at compile time tell me the settings don't make sense.  There is no reason not to allow the test as long as a probe is defined.

**Actual behavior:** 
"? (X,Y) out of bounds."

#### Additional Information
probe.h probe_min_x(), probe_min_y(), probe_max_x(), and probe_max_y() are disabled if HAS_LEVELING is false.

boelle, 2019-11-03:
will temp close this one until the information requested in the issue template has been added

drmulligan, 2019-11-03:
What is missing?  Do you want config files?  I gave the defines required to reproduce.  Do you want before and after working and not working config files?  Do you want a PR before it's considered enough info?  I even provided the code which causes the problem yet not enough information.

boelle, 2019-11-03:
well if you read the issue template then you would know

but yes the config files

drmulligan, 2019-11-03:
[Marlin_config_M48Works.zip](https://github.com/MarlinFirmware/Marlin/files/3802021/Marlin_config_M48Works.zip)
[Marlin_config_M48Fails.zip](https://github.com/MarlinFirmware/Marlin/files/3802022/Marlin_config_M48Fails.zip)

The cause is line #65 of Marlin\src\module\probe.h 
"#if HAS_LEVELING && (HAS_BED_PROBE || ENABLED(PROBE_MANUALLY))" in case the line number changes.

What else would you like?

boelle, 2019-11-03:
nothing more, configs are a clear requirement
so all is good

drmulligan, 2019-11-03:
FYI the diff between the two are "//" in front of a bed leveling mode in this case UBL.  Any one of them set HAS_LEVELING which solves the issue.

Evgeny-SPB, 2019-11-15:
I was able, to reproduce this issue with running G34 (G34 - Z Steppers Auto-Alignment) without enabling bed leveling. This function also uses probe.h functions "probe_min_x(), probe_min_y(), probe_max_x(), and probe_max_y()" to check if the position is reachable by the probe. But they are compiled only with HAS_LEVELING flag.
```
#if HAS_LEVELING && (HAS_BED_PROBE || ENABLED(PROBE_MANUALLY))
  inline float probe_min_x() {
    return _MAX(
      #if IS_KINEMATIC
        PROBE_X_MIN, MESH_MIN_X
      #else
        (X_MIN_BED) + (MIN_PROBE_EDGE_LEFT), (X_MIN_POS) + probe_offset.x
      #endif
    );
  }
  inline float probe_max_x() {
    return _MIN(
      #if IS_KINEMATIC
        PROBE_X_MAX, MESH_MAX_X
      #else
        (X_MAX_BED) - (MIN_PROBE_EDGE_RIGHT), (X_MAX_POS) + probe_offset.x
      #endif
    );
  }
  inline float probe_min_y() {
    return _MAX(
      #if IS_KINEMATIC
        PROBE_Y_MIN, MESH_MIN_Y
      #else
        (Y_MIN_BED) + (MIN_PROBE_EDGE_FRONT), (Y_MIN_POS) + probe_offset.y
      #endif
    );
  }
  inline float probe_max_y() {
    return _MIN(
      #if IS_KINEMATIC
        PROBE_Y_MAX, MESH_MAX_Y
      #else
        (Y_MAX_BED) - (MIN_PROBE_EDGE_BACK), (Y_MAX_POS) + probe_offset.y
      #endif
    );
  }
#else
  inline float probe_min_x() { return 0; };
  inline float probe_max_x() { return 0; };
  inline float probe_min_y() { return 0; };
  inline float probe_max_y() { return 0; };
#endif
```

In my opinion we can remove HAS_LEVELING precuisite, because this functions are also used in cases not related to LEVELLING, and can be used without levelling.

sjasonsmith, 2019-11-15:
It seems like that entire #else case should go away. It seems like if something can't build without them present, it probably won't work right with everything set to zero!

I already have a PR open that is touching this code, removing those default implementations may or not fit within its scope.
#15887

Evgeny-SPB, 2019-11-18:
Tested PR https://github.com/MarlinFirmware/Marlin/pull/15887 today. Seems that a problem will be solved after PR is merged.

boelle, 2019-11-24:
pr has been merged so closing

closed 2019-11-24 by boelle