MarlinFirmware/Marlin #15795: [BUG] CR-10s stock display not working on SKR 1.3 board

Ahamedo, 2019-11-03:
### Description

Upgraded my CR-10s with BigtreeTech SKR 1.3 board but the stock display is not working. it is the 2 cable display labeled EXP1 and EXP2 i tried swapping them back and forth but with no use.

### Steps to Reproduce
**Attachments:** The ZIP file which contains my `Configuration.h` and `Configuration_adv.h` files. 
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3802214/config_files.zip)

1. Plug the EXP1 cable of the display to EXP1 of SKR1.3 board.
2. Plug the EXP2 cable of the display to EXP2 of SKR1.3 board.
3. Turn on the printer

**Expected behavior:** 
The display should have power and shows the marlin splash screen, then to be able to navigate on menu items and print.

**Actual behavior:** 
The display stays off with no sign of power. However, the SKR board is powerd on and could be controlled by USB pronterface no problems

#### Display images
[image]
[image]
[image]

[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3802214/config_files.zip)

boelle, 2019-11-03:
also try ask here

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

thisiskeithb, 2019-11-04:
You have both `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` and `CR10_STOCKDISPLAY` enabled in your config. Only enable one of these.

The CR-10S uses `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` while `CR10_STOCKDISPLAY` is used for the original CR-10, Ender-3/5, etc.

> The display stays off with no sign of power.

Ensure you have EXP1 plugged into EXP1 and EXP2 plugged into EXP2. If that is correct, rotate the connectors on one end of the LCD cables 180 degrees and see if the LCD powers up.

Ahamedo, 2019-11-04:
Thank you @thisiskeithb this actually fixed the issue. i commented out CR10_STOCKDISPLAY as it should be for CR-10s, and rotated one end of the cables for exp1 and exp2 (i removed the plastic tab to make it fit rotated) and the display powered up.

closed 2019-11-04 by Ahamedo