MarlinFirmware/Marlin #15797: [BUG] [bugfix-2.x] Z-axis motion loses deceleration planning when reading SD card contents or sending commands quickly

YanMinge, 2019-11-04:
### Description
Z-axis motion loses deceleration planning when reading SD card contents or sending commands quickly

### Steps to Reproduce
【Recurrence mode 1 - SD card print】
1. use Animation_modeling.gcode for SD card print
2. Measure the signal of the step pin of Z-axis
3. When the Z axis moves, record the step signal of the Z axis.

【Recurrence mode 2 - Send the move command as follows】
1. G1 X116.66 Y100 Z200
2. delay 10-30s,  Make sure the Z axis moves to the specified position.
3. Measure the signal of the step pin of Z-axis

4. G1 X116.66 Y100
5. delay 2000
6. G1 X116.66 Y116
7. delay 2000
8. G1 X116.66 Y116 Z200.4
9. delay 10
10. G1 X116.66 Y100
11. delay 2000
12. G1 X116.66 Y116
13. delay 2000
14. G1 X116.66 Y116 Z200.6
15. delay 10
16. Repeat steps 4-15
When the Z axis moves, record the step signal of the Z axis.

**Expected behavior:** 
The movement of the Z axis should be the same as the single step movement.
[image]

**Actual behavior:** 
In the above measurement, the step of the Z axis is a waveform in which the deceleration process is lost.
[image]

This behavior has a serious flaw when using the  stealthChop mode of TMC driver. Because the step without deceleration stops causing a current surge, Taking the TMC2208 as an example, when the pulse frequency is around 9K, this inrush current may cause the drive to enter the short-circuit protection and stop the printing operation.
[image]

#### Additional Information
Configuration:
[Marlin_configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3802510/Marlin_configuration.zip)
SD card  test model：
[Animation_modeling.zip](https://github.com/MarlinFirmware/Marlin/files/3802512/Animation_modeling.zip)

YanMinge, 2019-11-06:
Is there anyone who can reproduce the same issue as me, I‘m not sure if it is only for marlin2.0, or if only in LPC1768 chip, or all Marlin are similar plans.

boelle, 2019-11-11:
@p3p does this make any bells ring?

YanMinge, 2019-11-12:
This problem is mainly related to the speed optimization of the inflection point. use junction deviation instead of traditional Jerk Limiting there will be some optimization. Since the speed of the Z axis is usually quite different from the speed of XY axis, this issue is more obvious on the Z axis.  Is it possible to optimize the inflection point without including the Z axis ?

DerAndere1, 2019-12-03:
@YanMinge 
1) Your config file looks your Marlin is from beginning of 2019. Can you check if the issue exist in current Marlin 2.0?

2)
> Since the speed of the Z axis is usually quite different from the speed of XY axis, this issue is more obvious on the Z axis. 

Does that mean that you observe loss of deceleration planning also for the X and Y axis? Then the bug title/description should be edited accordingly.

3) What software/script did you use to send the gcode commands for "Recurrence mode 2"?

boelle, 2019-12-07:
Since its 25 days since we have seen @YanMinge i will call it Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

thinkyhead, 2021-07-30:
We have sometimes seen a STST shutdown, and this has usually been due to too many DIR signal changes too close together, or just general timing being strange enough to make the TMC drivers mad. I'm not sure how to address it, except to get very deep into the acceleration and stepping stuff and pulling it apart to see where it is falling down.

Are you able to narrow down the version of Marlin or the date range where the bug starts to appear? That could help us to narrow down which change(s) led to this bad behavior.