MarlinFirmware/Marlin #15799: [BUG] x and y motor are ticking during diagonal motion

emaayan, 2019-11-04:
i was using marlin 1.1.9 and this issue seemed to have resolved in 1.1 bug fix but i"m curoius as to what caused it

when the boths motors are moving in diagonal motions they are ticking
this happens when bed leveling is enabled but only when z is under 2 mm . i have replaced card and drivers and still happens i also changed the vrefs and it still happens
i have the stripped the gcode into this
g28
M420 S1 ;

G1 Z1.1

; when these 2 lines are repeated over and over it happens

G1 X136.232 Y135.972

G1 X173.528 Y173.268

the video shows the x motor ticking and moving in a diagonal motion , then a delay , then a gcode moving only the x axis 

https://www.youtube.com/watch?v=8EyYAeQp6aY

thisiskeithb, 2019-11-04:
> i was using marlin 1.1.9 and this issue seemed to have resolved in 1.1 bug fix

So there’s no issue?

Can you confirm everything still works correctly in `bugfix-2.0.x`?

boelle, 2019-11-04:
since it was solved i will close this one

we can open of there is a issue within marlin itself

closed 2019-11-04 by boelle

emaayan, 2019-11-04:
> > i was using marlin 1.1.9 and this issue seemed to have resolved in 1.1 bug fix
> 
> So there’s no issue?
> 
> Can you confirm everything still works correctly in `bugfix-2.0.x`?

well i believe this is working on bug-fix-1.1 i don't want to move to 2.0 yet, (besides isn't it 32 bit?) , 
but from what i can tell it doesn't appear to happen on bug-fix.1.1 i didn't  know exactly how to ask it, because this issue has been plaugging me for weeks, possibly even effecting bed leveling. so i wanted to try and see if anyone could tell me about any bugs related to this topic that were closed on 1.1.9 (i used the diff on git-hub, but nothing seems to stick out)

emaayan, 2019-11-04:
> > i was using marlin 1.1.9 and this issue seemed to have resolved in 1.1 bug fix
> 
> So there’s no issue?
> 
> Can you confirm everything still works correctly in `bugfix-2.0.x`?

btw what they are describing here, sounds very similar (only my motors aren't idle, i my skew feature is turned of ) https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/clicking-from-x-y-axis-stepper-motors-when-idle-during-print/

boelle, 2019-11-04:
> (besides isn't it 32 bit?) ,

both 8 and 32 bit