MarlinFirmware/Marlin #15800: Printed Halted [CR-10S]

eliaolgiati, 2019-11-04:
Hi,
i'm new in this forum.
I've a little problem with my **CR-10S** and **Marlin 1.1.9**.
This firmware contains some interesting features that i love, for example "Bed Leveling", with nine calibration point.
The problem is that if I select .G-code file to print from the SD without calibrating the bed before, the print stops and show me the message "**Printed Halted, reset!**".
If I calibrate the bed before, the printer it works fine and start printing without any problem.
Can anyone help me?

Thank you!

boelle, 2019-11-04:
ask here instead:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

the issue list is for bugs and fr's only

advice is to use 2.0.x instead of old 1.1.9
and use arduino ide 1.9.0 beta or vscode+platform.io to compile and upload
you can use winmerge to compare old and new config and copy old settings to new config

closed 2019-11-04 by boelle

eliaolgiati, 2019-11-04:
Thank you very much for your advices!