MarlinFirmware/Marlin #15801: [BUG] Z_Endstop  doesn't work on the Z axis

hn0550, 2019-11-04:
### Description
 Z_Endstop  doesn't work on the Z axis
### Steps to Reproduce
Marlin-bugfix-2.0.x
1)  Z_Endstop is Z_mini
2)  GoTo Z home , The z_motor stops when Z_Endstop  is triggered
3)  When moving from Z100 to Z0,  the motor continues to operate after the Z_Endstop is triggered. Use the M19  to see that the  Z_Endstop  is triggered
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3804278/config.zip)

gloomyandy, 2019-11-04:
The endstops are not active during normal move operations by default, if you want them to be active all of time you need to enable ENDSTOPS_ALWAYS_ON_DEFAULT in configuration_adv.h (it is off in your files).

ruggb, 2019-11-04:
I am assuming by OPERATE you mean it continues to move Z.
If you are referring to the fact that the motor is still ON, not turning, that is normal and will deactivate after a short period, maybe 60 sec. You can configure it to turn off immediately after stepping. This is not recommended as Z will have nothing to hold its position.

Otherwise:
The question then becomes why isn't Z0 == Z Home.
If you G28 then G0 Z100 then G0 Z0 and it should stop at Z0, which will also trigger the endstop.
Unless there is something in your host (like Repetier-Host) that is setting an offset.

hn0550, 2019-11-07:
Maybe I didn't describe it very well, but what I mean is, when the Z axis moves from Z100 to Z0, use G1 Z0. During the motion if I trigger Z_Endstop, The Z axis stills to move.

ruggb, 2019-11-07:
A G0 Z0 does not use the endstops. They are only used for a homing command == G28
Software endstops are used for regular movement commands. IE, you can't G0 Z-5 or G0 Z400 on my Zmax = 300 machine.

hn0550, 2019-11-07:
I see. Thank all

closed 2019-11-07 by hn0550