MarlinFirmware/Marlin #15803: [BUG] Compilation fails with DAC steppers and screen enabled

dagorel, 2019-11-04:
Hi all,

Please find below a bug i suspect in Marlin 2.0 current revision.

Regards,
Orel

### Description

As stated in the issue title, the compilation fails when DAC steppers and screen features are enabled.

### Steps to Reproduce

Configuration files: [Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/3804767/Marlin_config.zip)

From the current revision of **bugfix-2.0.x** branch, just change the **MOTHERBOARD** to _BOARD_RIGIDBOARD_V2_ for instance (a board for which DAC steppers are enabled) and uncomment **REPRAP_DISCOUNT_SMART_CONTROLLER** in _Configuration.h_. Then compile using the default platformio environment.

**Expected behavior:** Compilation success.

**Actual behavior:** Compilation failure. See below for compilation log.

#### Additional Information

Compilation log:
```
Processing megaatmega2560 (platform: atmelavr; framework: arduino; board: megaatmega2560)
----
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 1.14.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.0.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.5.0
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit NeoPixel> 1.2.5
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Compiling .pio/build/megaatmega2560/src/src/lcd/menu/menu_advanced.cpp.o
Compiling .pio/build/megaatmega2560/src/src/lcd/ultralcd.cpp.o
Compiling .pio/build/megaatmega2560/src/src/libs/L6470/L6470_Marlin.cpp.o
In file included from Marlin/src/lcd/menu/menu_advanced.cpp:31:0:
Marlin/src/lcd/menu/menu_advanced.cpp: In function 'void menu_dac()':
Marlin/src/lcd/menu/../../inc/../core/language.h:319:15: error: expected unqualified-id before string constant
 #define MSG_X "X"
               ^
Marlin/src/lcd/menu/menu.h:354:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:386:48: note: in expansion of macro 'MENU_ITEM_P'
[... 78 lines cut ...]
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_E'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:72:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(E);
     ^
Compiling .pio/build/megaatmega2560/src/src/libs/buzzer.cpp.o
Compiling .pio/build/megaatmega2560/src/src/libs/hex_print_routines.cpp.o
Compiling .pio/build/megaatmega2560/src/src/libs/least_squares_fit.cpp.o
Compiling .pio/build/megaatmega2560/src/src/libs/nozzle.cpp.o
*** [.pio/build/megaatmega2560/src/src/lcd/menu/menu_advanced.cpp.o] Error 1
Compiling .pio/build/megaatmega2560/src/src/libs/numtostr.cpp.o
==== [FAILED] Took 3.23 seconds ====

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           FAILED    00:00:03.228
(...)
==== 1 failed, 0 succeeded in 00:00:03.228 ====
```

dagorel, 2019-11-06:
Hi again,

First of all, thanks to @boelle to add the compilation log to pastebin. I was aware that putting a log "as it is" could be annoying but i have no clue on what to do with logs in the bug template.

Regarding the potential bug, i guess the issue is around the **GET_TEXT** macro or the way the **EDIT_ITEM** macro handles labels.

I am fine to submit a pull request, but i want to be aligned with the coding rules...
It is easy to get rid of that bug in changing the _LABEL_ argument of **EDIT_ITEM** to a "valid" one, but it is not the right way to fix it.

Regards,
Orel

boelle, 2019-11-11:
@dagorel did you make a PR to fix this?

boelle, 2019-11-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

dagorel, 2019-11-23:
Hi @boelle ,

Actually it can be solved in defining new labels for each axis (MSG_DAC_PERCENT_X, MSG_DAC_PERCENT_Y, MSG_DAC_PERCENT_Z) and use MSG_DAC_PERCENT_##A in the EDIT_DAC_PERCENT macro. But i am not happy with this solution so i try to find another one.

Regards,
Orel

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

dagorel, 2019-12-10:
Hi @boelle ,

It is the very same with the new 2.0 branch so i submit a pull request on the bugfix-2.0.x branch.

Regards,
Orel

boelle, 2019-12-10:
:-) great, nice when people contribute

boelle, 2019-12-10:
let hope @thinkyhead hits merge soon

closed 2019-12-11 by thinkyhead