MarlinFirmware/Marlin #15804: [BUG] Z_MIN_PROBE_REPEATABILITY_TEST breaks EEPROM?

sme4gle, 2019-11-04:
### Description

When `Z_MIN_PROBE_REPEATABILITY_TEST` is enabled in `configuration.h` the printer functions like it should until I change a parameter in EEPROM and save that. Even then the printer still functions fine. But when I turn the printer off, and try to perform a reboot the printer will not be able to complete it's boot into Marlin

**Marlin files**
My Configuration.h and Configuration_adv.h are here
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3806102/Marlin.zip)

**Steps to reproduce**
These are the steps to reproduce:
1. Change EEPROM parameter, eg z offset
2. Save EEPROM
3. Reboot printer

**Expected behavior:** 
The printer should boot in order to get ready for printing

**Actual behavior:**
Printer gets stuck somewhere and doesn't completely boot.

ruggb, 2019-11-05:
I am not a developer, but I don't see that action on my system, but it isn't the same as your system.
But intermittently, a reboot will cause strange things to happen the first time I try to run them, like bed level. Then they run fine after that. That might indicate that something isn't getting initialized properly.
I am curious as to why you are trying to cool things with the fans running at full while you are trying to preheat bed and hotend?

sme4gle, 2019-11-05:
@ruggb Ooh I, didn't do that intentionally, I'm still configuring this printer step by step. Disabling the fans while leveling is something I just havent done yet.

brew99, 2019-11-05:
You didn't indicate which board you are using, but iaw your config, it looks like a SKR mini E3.  This issue you describe is a very common occurrence when you have gone over the max firmware.bin file size.

This board more than likely has the "RC" chip which has 256kB max, in which the EEprom is taking some.   Your max firmware.bin file size cannot be more than 224kB (229,376 bytes when compiling).

Disable some functions and try again

sme4gle, 2019-11-06:
Oh that makes perfect sense actually. I'm not at home now, but I'm pretty sure the filesize is 224 without the feature enabled, and a little more with that feature enabled.

sme4gle, 2019-11-07:
Update: It seems that my board has 512kb of flash according to this post: https://www.reddit.com/r/ender3/comments/dsmf6x/512k_flash_working_on_skr_mini_e3_v12/ I will post back on this tomorrow.

boelle, 2019-11-11:
@sme4gle any news?

sme4gle, 2019-11-16:
@boelle sorry for the late reply, my awnser 9 days ago fixed it for me. The SKR 1.2 seems to have 512kb storage

boelle, 2019-11-16:
so we can close this one?

sme4gle, 2019-11-16:
I suppose we can, but shouldn't the amount of available storage be changed for the board?

boelle, 2019-11-16:
yes, but i will open a FR (feature request) for that and link to this issue

closed 2019-11-16 by boelle