MarlinFirmware/Marlin #15806: I have used my own motherboard and 4 TMC2209 driver modules. In UART mode, I set the microstep to 16, but each time the serial port sends M122, my E-axis microstep displays 256. If I turn on the XYZ axis, then the Z axis shows a microstep of 256. Why, can you help me?

lijinlin19970813, 2019-11-05:


teemuatlut, 2019-11-05:
Communication issue.

YanMinge, 2019-11-05:
Marlin has a storage configuration for the TMC configuration(M500-M504). I think it may be that your configuration has not been modified at all. Simple code configuration is not the final configuration.

boelle, 2019-11-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2019-11-08:
I had some issues the other day where errors in SoftwareSerial libraries caused communication to appear OK, but corrupt register values were actually received.

Post your configs (including your platformio.ini) and I would be happy to take a look.

boelle, 2019-11-11:
will temp close this one as we need the configs first

ping me by adding @ and then my username when configs and platformio.ini is added

lijinlin19970813, 2019-11-21:
> “问题”模板中的要

[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/3873304/Desktop.zip)
I tried it for a long time, the problem still exists, and sometimes it is ok, but when I re-power it, I can't do it again.
[image]

lijinlin19970813, 2019-11-21:
> 前几天，我遇到了一些问题，其中SoftwareSerial库中的错误导致通信显示正常，但实际上收到了损坏的寄存器值。
> 
> 发布您的配置（包括您的platformio.ini），我很乐意看一下。

[image]
[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/3873312/Desktop.zip)
Still will show 256, although I configured 16

sjasonsmith, 2019-11-21:
Does the problem occur on random drivers, or only the last one?

If you add a 2209 on E, do you still see errors on Z, or does the failure move to E?

lijinlin19970813, 2019-11-21:
> Does the problem occur on random drivers, or only the last one?
> 
> If you add a 2209 on E, do you still see errors on Z, or does the failure move to E?

If I turn on the 2209 mode of XY, then Y shows 256. No matter how many 2209 I open, always show 256 in the last one, or the first one?

sjasonsmith, 2019-11-21:
Can you modify your platformio.ini file to use this TMCStepper, to see if the problem goes away?
You will need to change it in the LPC1768 section, as it does not use the common dependencies from the top of the file.

`  TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip`

lijinlin19970813, 2019-11-21:
> Can you modify your platformio.ini file to use this TMCStepper, to see if the problem goes away?
> You will need to change it in the LPC1768 section, as it does not use the common dependencies from [...]

[image]
[image]
I think that even if the compilation is successful, there will be problems. Now, I summed it up, my 2208 and 2209 are such problems.【terror】

sjasonsmith, 2019-11-21:
Are you using an old commit of bugfix-2.0.x? The TMCStepper dependency has not been "<1.0.0" for quite some time.

Anyway, you would need to remove the old TMCStepper line and replace it with the new line. It builds fine for me, but I am working with bugfix-2.0.x that is only about one hour old.

lijinlin19970813, 2019-11-22:
> Are you using an old commit of bugfix-2.0.x? The TMCStepper dependency has not been "<1.0.0" for quite some time.
> 
> Anyway, you would need to remove the old TMCStepper line and replace it with the new line. It builds fine for me, but I am working with bugfix-2.0.x that is only about one hour old.

Just today, I downloaded the latest marlin, but I still get the initial error.
This is the latest env:
[image]

sjasonsmith, 2019-11-22:
Now try it with the updated TMCStepper I provided. I had a similar problem when working with the SKR Pro. The branch of TMCStepper I provided is slightly modified to test whether the same issue impacts LPC1768.

lijinlin19970813, 2019-11-22:
> Now try it with the updated TMCStepper I provided. I had a similar problem when working with the SKR Pro. The branch of TMCStepper I provided is slightly modified to test whether the same issue impacts LPC1768.

I used your TMCStepper, M122 is displayed correctly, but in fact the motor movement is wrong.

lijinlin19970813, 2019-11-22:
> Now try it with the updated TMCStepper I provided. I had a similar problem when working with the SKR Pro. The branch of TMCStepper I provided is slightly modified to test whether the same issue impacts LPC1768.

I have used your TMCStepper to test it several times. It is OK. So can you help me find the difference between the two?
[image]

sjasonsmith, 2019-11-22:
You can look at the last commit in that GitHub branch to see what I changed.

@boelle, this issue should probably be re-opened. I thought this might be a problem when I had the same issue on STM32, but I hadn't heard any reports of it on LPC until this issue.

@p3p & @teemuatlut, the problem he is encountering is that the SoftwareSerial::begin function on LPC176x leaves listening enabled at the end of the function, even for half-duplex communication. This leaves the pin configured as an input, which on a lot of hardware (BTT 220x modules, specifically) pulls the line to an indeterminate state, because the driver modules have pull-downs which compete with the input pullup. This prevents the TMC driver from detecting the start bit.

It only impacts the last driver, because there is only one listener is active at a time. When listen() is called on each driver, it stops listening on other drivers.

The STM32 version of SoftwareSerial is already modified to skip calling listen() in begin if using half-duplex.

To resolve this, we either need to modify SoftwareSerial in the LPC framework, or we need to modify TMCStepper to always call stopListening() after calling begin(). Do you have preferences where we solve this?

sjasonsmith, 2019-11-22:
@lijinlin19970813, have you modified the SKR pins file for your firmware?
I just realized that I would have only expected my change to do anything for single-pin software serial. By default the SKR pins file uses 2-pin software serial.

sjasonsmith, 2019-11-24:
I have posted a pull request to TMCStepper, to see if it can include the change I had @lijinlin19970813 try out.

https://github.com/teemuatlut/TMCStepper/pull/101

sjasonsmith, 2019-11-25:
@lijinlin19970813, I have closed my pull request to TMCStepper, because I can't force this communication error on my SKR 1.3, even if I go out of my way to worsen the serial signal.

Can you answer my last question about changing the pins file? Perhaps you could zip up your entire Marlin folder so I can see if anything has changed other than the configuration files.

It is possible that my change is helping in some way, but given the current the default values in the pins file, I'm having a hard time understanding how.

lijinlin19970813, 2019-11-26:
> @lijinlin19970813, I have closed my pull request to TMCStepper, because I can't force this communication error on my SKR 1.3, even if I go out of my way to worsen the serial signal.
>
> Can you answer [...]

After I used the method you provided, my problem was solved.So it is OK now.I appreciate your help！

sjasonsmith, 2019-11-26:
> After I used the method you provided, my problem was solved.So it is OK now.I appreciate your help！

The problem is that my branch was just a test, it isn't available for everyone else to use. We don't quite understand why it fixed your problem, and can't reproduce the problem ourselves.

That's why the extra information about your board and pins file would be helpful.

lijinlin19970813, 2019-11-26:
> The problem is that my branch was just a test, it isn't available for everyone else to use. We don't quite understand why it fixed your problem, and can't reproduce the problem ourselves.
>
> That's [...]

My motherboard's X / Y / Z / E RXTX uses the same pins, so this should be the problem.

sjasonsmith, 2019-11-26:
> My motherboard's X / Y / Z / E RXTX uses the same pins, so this should be the problem.

Can you provide a copy of the pins file you are using?

boelle, 2019-12-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-07 by boelle