MarlinFirmware/Marlin #15809: [BUG] sensorless probing with ADL bilinear not working

hackebike, 2019-11-05:
SKR Pro v1.1 + TMC 2209 v1.2 using sensorless probing Auto Bed Leveling Bilinear failed to probe the bed.

I setup the sensorless probing so that the nozzle is the probe.
I than enabled auto bed leveling bilinear.
Than used the following commands:
G28 to home
G29 to start bed leveling

Extruders moved from 0,0,0 to the middle of the printer and than to the x edge (650mm) and y (50mm is what I set to be the edge of probing), with the nozzle/probe doing couple of probes without touching the bed, and than stopped. It did not continue to do the probing to other points. At the command line it said: probing failed

Here is a [video](https://www.youtube.com/watch?v=zqbWxaegkuA).

I used M111 S32 leveling debug did not add much information about the failure.

Config files: 
[MarlinCinfigs.zip](https://github.com/MarlinFirmware/Marlin/files/3809385/MarlinCinfigs.zip)
Marlin compiled from today.

Grogyan, 2019-11-05:
It is generally inadvisable to be using the nozzle itself for probing, as it can, and does, damage your printer

Patag, 2019-11-05:
@Grogyan except if you use a nozzle piezzo sensor as I do ! You would be really surprised by this system efficiency !

hackebike, 2019-11-05:
the issue is with ADL not with nozzle probing.
@Patag that's interesting, any details I can look at?

Grogyan, 2019-11-05:
@Patag That is a sensor.
In @hackebike video, you're probing onto a buildtack surface, with a volcano hotend.
Even if the hot end is not on, the softening of the buildtack surface will cause deviation of the bed surface from repeated probing

Patag, 2019-11-05:
Sure !
https://www.precisionpiezo.co.uk/
Just note I'm just a user, not a shareholder ;)

hackebike, 2019-11-05:
@Grogyan everything is custom on my printer including the pellet extruder extruder and nozzle. I did not calibrate the z-offset yet. I do not care about been precise to the micro since I am print at 1/1.2/1/5 mm. Everything is working fine beside ADL, any idea?

Grogyan, 2019-11-05:
I am unable to verify your config right now.

My concern, is that someone new to 3D printing will try this, and do something idiotic and damage their printer.

@hackebike with such layer heights, manual mesh levelling should be the way to go.

If you insist on sensorless probing, then you will need to adjust your StallGuard value
Keeping i mind that on the TMC2209  that it requires StealthCHop to be active for Stall Guard to work properly

hackebike, 2019-11-05:
My printer is huge 1200x800x500 and I would prefer not to do anything manually.
I have all those enabled in the configs and sensitivity is ok (found a good value), the probing works fine.
The problem is with bilinear ADL (or maybe with my configs(?)), it basically does not start to do the mash probing and it just says probing failed, even using M111 no details. If someone can take a look at the configs it would be great.
I will do more testing tomorrow.
thx

hackebike, 2019-11-06:
done more testing today, here is a log from M503
[logM503.txt](https://github.com/MarlinFirmware/Marlin/files/3814087/logM503.txt)

I tried to calibrate the Z-offset using the following:
Calibrating Z-Axis
1) Home your Z axis
2) M851 Z0      // Reset Z offset to 0
3) M500         // Save to EEPROM
4) M501         // Set EEPROM
5) G28 Z        // Home Z only
6) G1 F60 Z0    // Move to Z = 0
7) M211 S0      // Soft endstops off
8) Paper technique with z change to figure out correct Z0. M114  to get current pos
9) M851 Z-1.7   // Set z offset
10) M211 S1     // Softenstops back on
11) M500
12) M501

But seems that the SKR Pro does not store the new value for the z-offset to the EEPPROM.
I have un-comment the following: 
#define EEPROM_SETTINGS in configuration.h
#define FLASH_EEPROM_EMULATION // Use Flash-based EEPROM emulation in pins_BIGTREE_SKR_PRO_V1.1.h

hackebike, 2019-11-06:
in the log above the auto bed leveling has NaN value:

hackebike, 2019-11-07:
I have done couple of tests:
Set to have 1 extruder (the above configs have 2), same failed results
Set to have less probing points 3x3, same failed results

I than try to use UBL and seemed to work. So there is definitely an issue with Bilinear ABL.

dch1921, 2019-11-08:
@hackebike here are my config files you asked for. Looks like I'm working from 11/6 build of Marlin

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/3826225/Config.Files.zip)

boelle, 2019-11-11:
@hackebike still an issue?

hackebike, 2019-11-12:
tested 2 days ago, it was still an issue. cannot test now, traveling

boelle, 2019-11-21:
@hackebike when can you test again?

hackebike, 2019-11-21:
sorry cannot test for few weeks as I am traveling for work.
As soon as I get back will test and report. thx

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
is the issue still the same with all the updates in the last 33 days?

hackebike, 2019-12-24:
sorry cannot test it yet.
you can close it and when I test if is an issue I will re-open or open a new one, but hopefully is not an issue anymore.

boelle, 2020-01-02:
will close, no others have spoken up and as you say we can always reopen, if you cant then just ping me by adding @ before username

closed 2020-01-02 by boelle

hackebike, 2020-01-02:
sure I will hopefully soon.
Thx for the support!

boelle, 2020-01-02:
not much support, and i just sweep the floor :-/