MarlinFirmware/Marlin #15810: How do I set an VECT_TAB_OFFSET for Marlin2.0

FLYmaker, 2019-11-05:
I made a new 9-axis 3D printer control board with stmf407zgt6. I want to write the firmware to 0x8008000 with my bootloader, but I don't know how to modify the firmware with vscode. If you do not modify the address burned to 8000, it will not run

FLYmaker, 2019-11-05:
Later, I will save my bootloader to my repository. Stm32f4 ﹐ black can also be used

thisiskeithb, 2019-11-05:
You can use PlatformIO's extra scripting feature. Take a look at the SKR Mini line of boards:
https://github.com/MarlinFirmware/Marlin/blob/f088c6a8503907ae00dec817af381e63757713e6/platformio.ini#L294-L309

extra_scripts from above:
https://github.com/MarlinFirmware/Marlin/blob/f088c6a8503907ae00dec817af381e63757713e6/buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py#L1-L8

LDSCRIPT_PATH from above:
https://github.com/MarlinFirmware/Marlin/blob/f088c6a8503907ae00dec817af381e63757713e6/buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI.ld#L1-L14

FLYmaker, 2019-11-06:
> LDSCRIPT_PATH
Thanks.

# 
# FLY
#
[env:FLYF407ZG]
platform = ststm32@5.6.0
framework = arduino
board = FLYF407ZG
extra_scripts = pre:buildroot/share/PlatformIO/scripts/black_stm32f407vet6.py
build_flags = ${common.build_flags}
  -DSTM32F4 -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DVECT_TAB_OFFSET=0x8000
lib_deps = ${common.lib_deps}
lib_ignore = Adafruit NeoPixel, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed = 250000

  I added a line here 《DTARGET_STM32F4 -DVECT_TAB_OFFSET=0x8000》
I have to manually modify the data on 0x8008000 address: 20020000 - > 20000430. SD card can run after burning. I can modify the bootloader program and automatically modify the data on 0x8008000 address, but I think the main board program will be unstable

boelle, 2019-11-06:
will close this one as it's a question support thing

the list is for bugs only

ideal you should get support here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-06 by boelle