MarlinFirmware/Marlin #15813: [BUG] Marlin bugfix 2.0.x (UBL Nozzle far from bed. Z-Axis not moving.)

smith505, 2019-11-05:
Description
Watched a few videos on UBL can't seem to figure out if its me or Marlin. Saw a few other Bug reports on here about it but no fix. When starting a print my nozzle is way off the bed

Setup:
SKR mini e3 w/2209s
BLTouch v1.3

Start G-Code

Actual behavior:
The ABL process halts at step 6 of 9 and acts like it has completed the G29 process. No errors shown or even a grid printed on the command line.

Additional Information
Configuration.h and Config_adv.h files in attached zip for review.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3811287/Configs.zip)

Vertabreak, 2019-11-07:
generally with UBL you setup and save everything on the machine and don't use any start gcode for leveling other then maybe a G29 J but i never found the need.

smith505, 2019-11-07:
Could you possibly send me your configs so I can see what is different?

boelle, 2019-11-11:
@smith505 still an issue?

smith505, 2019-11-11:
@boelle yessir

Vertabreak, 2019-11-13:
if you wanna see my configs.
https://github.com/Vertabreak/Marlin/tree/bugfix-2.0.x/Marlin

smith505, 2019-11-13:
@Vertabreak Thanks going to look and test it out will report back <3

stefi01, 2019-11-16:
i get the same problem with skr 1.3,ubl and 2209's, i done the 100 probe points sucsessfully with the G29 P1 comand then did the offset and stored it,  but it does not matter what offset settings i use the nozzle will always be too far from the bed, also if i decide to use G29 then it doesnt matter how level the bed is is will always fail to probe, will always fail on probe point one. the probe is inductive and is in good working order.

spmgit, 2019-11-23:
I also have the same problem with skr 1.3, ubl, 2209's and bltouch. G29 P1 doesn't probe the first column ever, but I can have it fill in with G29 P3. However even with a full 10x10 grid set a G26 P10 will just sit in one spot and do a circle, then exit. If I try and add a G29 L1 and a G29 J to my start code the first probe always fails.

spmgit, 2019-11-23:
So yeah, the bltouch is to the right and in front of the nozzle which is why it never probes the first column. I still can't do a G26, it sits and spins in a circle. I manually set the 3-Point parameters, so a G29 J now works, but if I try and print the nozzle is always too far from the bed. My z offset is -1.31 so that seems reasonable. I should have enough memory on the skr to enable M111 S32 debugging so I will try and get more information.

tatusah, 2019-11-23:
Have you by any chance turned off the printer before trying to print and seeing the nozzle beeing too far? Can you @spmgit check if the z-offset value is still stored after turning off the machine and then waiting a bit before turning the printer back on?  I'm asking because I'm seeing issue where the offset value set trough LCD or M851 command is lost after powering down the machine even if it was supposed to be saved to EEPROM.

spmgit, 2019-11-23:
@tatusah I have not seen it lose my z-offset after power off. If I need to change it, I always change it with M851 and not the LCD so I don't know if that makes a difference.

boelle, 2019-11-24:
@smith505 have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

spmgit, 2019-11-25:
I really wanted to get some prints done the next couple of weeks, so I switched from UBL to Bilinear. Bilinear works for me. I will test UBL again in a couple of weeks.

Vertabreak, 2019-12-02:
Still using UBL everyday without issue could be a board thing, dunno.

boelle, 2019-12-02:
since we have not heard from @smith505 i will assume he has solved the issue

so will close this one, we can reopen if he still have an issue and others can confirm

closed 2019-12-02 by boelle