MarlinFirmware/Marlin #15814: SKR v1.3 + MKS MINI12864 + onboard SD

marvin-sinister, 2019-11-05:
### Description
Version: bugfix-2.0.x

If SD card is configured as ONBOARD for SKR v1.3, the MKS MINI1864 (and possibly others) doesn't work. I managed to get it to work by modifying the pins file for SKR as follows:
```
      #if ENABLED(FYSETC_MINI_12864)
      #define DOGLCD_CS    P1_21
      #define DOGLCD_A0    P1_22
      ...
      #elif ENABLED(FYSETC_MINI_12864_2_1)
        #define NEOPIXEL_PIN    P1_18
      #endif
```

And using:
`#define FYSETC_MINI_12864_2_1`
in `Configuration.h`.
And:
`#define SDCARD_CONNECTION ONBOARD`
In `Configuration_adv.h`.

I suspect that the core of the issue is that by setting `SDCARD_CONNECTION` to `ONBOARD` marlin uses wrong SPI for LCD.

### Steps to Reproduce

1. Connect MKS MINI12864 to SKR v1.3
2. set `SDCARD_CONNECTION` to `ONBOARD`
3. enable `MKS_MINI_12864`

**Expected behavior:** 
LCD working

**Actual behavior:**
LCD has only backlight

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3811584/Configuration.zip)

Evg33, 2019-11-05:
MKS Mini not FYSETC_MINI.
FYSETC Mini12864 2.1 (Neopixel) working at skr1.3
[image]

thisiskeithb, 2019-11-06:
I just tested an MKS Mini 12864 on an SKR 1.3 and can confirm that it works fine if you define `SDCARD_CONNECTION LCD`. If you set it to `SDCARD_CONNECTION ONBOARD`, nothing shows up on the LCD.

I'll have to confirm the MKS Mini 12864's pinout and then make sure LCD conditionals are setting the correct pins.

boelle, 2019-11-06:
so there is no issue here? if so @marvin-sinister please hit close below

thisiskeithb, 2019-11-06:
> so there is no issue here?

There is:

>  If you set it to SDCARD_CONNECTION ONBOARD, nothing shows up on the LCD.

boelle, 2019-11-06:
yes but that is not a fault with marlin, or am i mistaken`?

thisiskeithb, 2019-11-06:
> yes but that is not a fault with marlin, or am i mistaken`?

It's a bug in the SKR 1.3 pins file or possibly SPI issue since nothing shows up on the LCD.

Evg33, 2019-11-06:
I use only SDCARD_CONNECTION ONBOARD.
Tested with next screens:
FYSETC_MINI_12864_1.2 / 2.0 / 2.1
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
ULTIPANEL
I do not have MKS_MINI_12864 for testing.

Evg33, 2019-11-06:
@marvin-sinister can you test with changes in pins_BIGTREE_SKR_V1.3.h
```
      #if ENABLED(MKS_MINI_12864)
        //#define DOGLCD_CS  P1_21
        //#define DOGLCD_A0  P1_22
        #define DOGLCD_CS  P1_19
        #define DOGLCD_A0  P1_20

```

https://reprap.org/wiki/MKS_MINI_12864

Evg33, 2019-11-06:
https://github.com/MarlinFirmware/Marlin/pull/13710

marvin-sinister, 2019-11-06:
Okay, by playing around a bit i managed to get it working (and hopefuly didn't break anything else in process. The fix is as follows:
In `Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h` we need to define `DOGLED` pins that are missing:
```
      #if ENABLED(MKS_MINI_12864)
        #define DOGLCD_CS  P1_21
        #define DOGLCD_A0  P1_22
        #define DOGLCD_SCK   P0_15
        #define DOGLCD_MOSI  P0_18
        #define FORCE_SOFT_SPI
      #endif
```
And in `Marlin/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_sw_spi.cpp` we need to use same code as `FYSETC_MINI_12864`, therefore we replace:
```
  #if ENABLED(FYSETC_MINI_12864)
```
With:
```
  #if EITHER(FYSETC_MINI_12864, MKS_MINI_12864)
```
In both places, so on line number 133 and 161.

After those changes MKS MINI12864 works and SD card can be set as `ONBOARD` or `LCD`. I don't have setup to test `CUSTOM_CABLE`.

thisiskeithb, 2019-11-06:
@marvin-sinister Awesome! Wanna submit a PR?

Evg33, 2019-11-06:
@thisiskeithb If you want to create PR, then remove double FORCE_SOFT_SPI please from fysetc section.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h#L229-L234

thisiskeithb, 2019-11-06:
@Evg33: If @marvin-sinister doesn't create a PR and include, I can.

boelle, 2019-11-11:
@marvin-sinister should your PR fix this?

marvin-sinister, 2019-11-11:
Fixed by #15836

PR #15836 (2019-11-08): Fix MKS MINI12864 on SKR 1.3
### Description

Fixes #15814 - MKS MINI12864 on SKR 1.3
### Benefits
MKS MINI12864 works on SKR 1.3 with ONBOARD and LED option.

### Related Issues
#15814
closed 2019-11-11 by marvin-sinister