MarlinFirmware/Marlin #15815: [BUG] Also With Fix - Homing with Z Probe

diabl0w, 2019-11-05:
There have been a ton of issues with using a zmin probe like the BLTouch and not homing properly. I think this will fix at least some of the issues... In Conditionals_LCD.h shouldn't:
`  #define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN)`
be:
  `#define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0 && HAS_CUSTOM_PROBE_PIN)'
?
This fix allows me to NOT have to change my Z_MIN_PROBE_PIN to the same thing as Z_STOP_PIN and also not having to define the zmin_probe_uses_zmin_plug or whatever it was called as a hacky workaround.
This is on Bigtreetech E3 Mini V1.2

Sorry, I dont know how to use git properly so I cant create a pull request

shitcreek, 2019-11-06:
With adding this change to my working BLTouch on a SKR 1.3, the probe does not deploy and so the print head crashes into the bed. Perhaps this one is only for your E3 Mini 1.2?

diabl0w, 2019-11-06:
do you have your bltouch in the zmin plug?

shitcreek, 2019-11-06:
It's connected to the zmin plug. Just to clarify my setup works without this fix. I'm merely testing your fix to see if it would conflict with my setup.

diabl0w, 2019-11-06:
without even testing, can someone confirm my logic on the whole code series WITHOUT my fix as follows:
First, Z_PROBE_SERVO_NR gets set to 0 for BLTOUCH:

> #if ENABLED(BLTOUCH)
>  #ifndef Z_PROBE_SERVO_NR
>   #define Z_PROBE_SERVO_NR 0

Set HAS_Z_SERVO_PROBE because previous sets it to 0:

>  #define HAS_Z_SERVO_PROBE (defined(Z_PROBE_SERVO_NR) && Z_PROBE_SERVO_NR >= 0)

Set HAS_BED_PROBE because previous sets HAS_Z_SERVO_PROBE:

>  #define HAS_BED_PROBE (HAS_Z_SERVO_PROBE || ANY(FIX_MOUNTED_PROBE, TOUCH_MI_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, SOLENOID_PROBE, SENSORLESS_PROBING, RACK_AND_PINION_PROBE))

Set PROBE_SELECTED because previous sets HAS_BED_PROBE:

>  #define PROBE_SELECTED (HAS_BED_PROBE || EITHER(PROBE_MANUALLY, MESH_BED_LEVELING))

If HAS_BED_PROBE is true (which it is), continue. 
Then, in our configuration.h we have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN commented out, so that should make HAS_CUSTOM_PROBE_PIN true.  

In our configuration.h, we also have Z_HOME_DIR = -1 and "#define Z_MIN_PROBE_PIN    PC14".  Since we define Z_MIN_PROBE_PIN, it disables HOMING_Z_WITH_PROBE because of the "!". Isnt this counter intuitive? We would WANT HOMING_Z_WITH_PROBE to be TRUE if we have defined Z_MIN_PROBE_PIN, right?

>  #if HAS_BED_PROBE
>  #define HAS_CUSTOM_PROBE_PIN  DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
>  #define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN)

diabl0w, 2019-11-06:
> It's connected to the zmin plug. Just to clarify my setup works without this fix. I'm merely testing your fix to see if it would conflict with my setup.

I understand, I am only asking because then does that mean that maybe there is an issue in the line above it that sets HAS_CUSTOM_PROBE_PIN. Either way, can you follow the comment I set above and tell me if the logic that I based this fix on is valid?

diabl0w, 2019-11-06:
I think I understand something better now, Marlin assumes that if you don't use the Z_MIN_ENDSTOP_PIN then that means you dont want your probe being used as an endstop for homing.

My question then would be, why would Marlin assume this? What is the use case of a z-probe if it isnt used to home? Either way, we used to be able to explicitly tell marlin that we still wanted to use the probe for homing with the option Z_MIN_PROBE_ENDSTOP but for some reason that was removed.

I guess maybe the fix to this issue would be to reimplement "Z_MIN_PROBE_ENDSTOP"

thisiskeithb, 2019-11-06:
> My question then would be, why would Marlin assume this? What is the use case of a z-probe if it isnt used to home?

Some people still use an endstop for z homing & a BLTouch for ABL only.

diabl0w, 2019-11-06:
So, if you want to use a Z-probe for homing, you MUST either:
>  1. Set Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and plug your probe into the Z_MIN plug on your board OR
>  2. Set Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and redefine Z_STOP_PIN (z_min_pin) to the Z_MIN_PROBE_PIN pin and leave the probe plugged in to the PROBE plug on your board

That is fine, I just haven't seen that stated anywhere as explicitly as I just have. Maybe its just obvious for others with more knowledge than me. To me, it would make more sense to  have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP instead be called Z_MIN_PROBE_IS_Z_MIN_ENDSTOP. Then, if defined, would use Z_MIN_PROBE_PIN as the endstop trigger.

closed 2019-11-06 by diabl0w

mak0t0san, 2019-12-11:
I think HOMING_Z_WITH_PROBE should be defined in configuration.h in the probes section, and Conditionals_LCD.h can check if that is defined already.  I have the SKR Pro 1.1 board and there are dedicated pins for a BLTouch probe so I can use a those instead of the Z-min endstop pin. I would like to home with the BLTouch using the custom probe pin.  From an end-user perspective, this should be possible without having to go into Conditionals_LCD.h.

diabl0w, 2019-12-11:
> I think HOMING_Z_WITH_PROBE should be defined in configuration.h in the probes section, and Conditionals_LCD.h can check if that is defined already. I have the SKR Pro 1.1 board and there are [...]

if your using the probe as an endstop (for homing) then you should plug the probe into the endstop pin... you gain no functionality by changing the conditionals_LCD and plugging into the probe pin

mak0t0san, 2019-12-11:
It's just a matter of convenience. The probe pins on the board fit the connectors from the BLTouch better than the XH style connector on the Z min endstop. But I understand that there is no functional benefit and I did end up plugging my probe into the Z min endstop connector.  I spent way too much time trying to figure out why homing Z wasn't working with a probe and eventually found out why from diving into the code. 
I also use a Duet board and that has dedicated probe pins as well and it allows the probe for homing when used in the probe pins.