MarlinFirmware/Marlin #15818: [BUG] Nozzle drops into print at some point (let's call it 'inverted Z-Hop')

opiswahn, 2019-11-06:
### Description

I'm on a Ender-2 with a SKR E3 Mini v1.2 board. With the current Marlin version and ABL, I get some strange behavior while printing. 
On start, homing, ABL (with a 3dTouch) does it's  thing, and it begins to print. The first 1 or 2 layers look very good, but then is suddenly drops the nozzle while printing for a short period and backs up again.
Looks some kind of an inverted Z-Hop, the palces where this happens are pretty consistent through the print and through different print attempts.
I can only print 4 layers or so as the nozzle seems to drop lower every time and gets burried within the print at some point where I have to abort.

Does this sound familiar to anyone?

boelle, 2019-11-06:
maybe ask here

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

also

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

opiswahn, 2019-11-06:
@boelle uploaded the config files

opiswahn, 2019-11-06:
I'm pretty sure it has to do something with ABL... this is the current mesh it produces:
[image]

With Marlin 1.1.9 some days ago it was nearly complete flat (sadly i have no visualization of that anymore), with the same sensor and same other hardware besides the board.

opiswahn, 2019-11-08:
so... anything?

boelle, 2019-11-11:
nope i dont have the issue myself and i just updated 3 days ago

opiswahn, 2019-11-11:
It has definitly something to do with Bilinear_Subdivision... as soon as i disable it everything is ok. Activated with no matter what setting, the mesh is way off and extremly overcorrected in printing...

shitcreek, 2019-11-12:
I think @akaJes is responsible for `Bilinear_Subdivision`, maybe he/she could chime in.

boelle, 2019-11-21:
@akaJes ??

tpruvot, 2019-11-21:
overcompensating ? i see the opposite here... its like if the mesh levels were 10x more flat than before... the xyze struct change, made in the end of september

opiswahn, 2019-11-21:
Well, maybe it's manifesting differently on different setups... mine is definitly overcompensating, or better the mesh is overexaggerated...

boelle, 2019-11-26:
@opiswahn btw what version of marlin are you using?

opiswahn, 2019-11-26:
@boelle the latest one 2.0.x. Just updated from repo, faulty behavior remains...

boelle, 2019-12-07:
@Roxy-3D any clue what could be wrong here?

boelle, 2019-12-07:
@opiswahn since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

opiswahn, 2019-12-26:
I'm still having the problems... nothing has changed here sadly

boelle, 2020-01-08:
@opiswahn still no good news with the latest bugfix 2.0.x?

boelle, 2020-01-15:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-15 by boelle