MarlinFirmware/Marlin #15821: TMC Connection Error when using TMC2208 with DUE.

bob808, 2019-11-06:
### Description

Trying to test TMC2208 with DUE and I'm getting TMC Connection Error. 
The TMC2208 drivers are good for sure as they work standalone, and I can connect to them via serial using a ftdi adapter + Script Communicator. If I move the rx/tx lines from the ftdi board to the DUE I get the connection error message on the display.
I tried with two different TMC2208, and I also tried using any serial1/serial2/serial3 ports but nothing.
The serial ports are working for sure as I tested with the serial console.

M122 reports this:
`		X	Y
Enabled		false	false
Set current	800	800
RMS current	1436	1436
MAX current	2025	2025
Run current	25/31	25/31
Hold current	12/31	12/31
CS actual	0/31	0/31
PWM scale	0	0
vsense		0=.325	0=.325
stealthChop	false	false
msteps		256	256
tstep		0	0
pwm
threshold		
[mm/s]			
OT prewarn	false	false
OT prewarn has
been triggered	false	false
off time	0	0
blank time	16	16
hysteresis
-end		-3	-3
-start		1	1
Stallguard thrs		
DRVSTATUS	X	Y
stst			
olb			
ola			
s2gb			
s2ga			
otpw			
ot			
157C			
150C			
143C			
120C			
s2vsa			
s2vsb			
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!

Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
ok`

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3814045/config.zip)

bob808, 2019-11-09:
Can anyone else with a DUE and TMC2208 confirm this?

boelle, 2019-11-16:
@bob808 i assume this is still an issue?

bob808, 2019-11-17:
Yes, at least for me.

bob808, 2019-11-18:
This is what's on RX/TX when talking to tmc2208 via ftdi with scriptcommunicator:
[image]

And this is what's on RX/TX when DUE is trying to talk to tmc2208:
[image]

sjasonsmith, 2019-11-18:
@bob808, those look so different because the top picture is a read, and the bottom picture is a write.

Try capturing while executing an M122 command, to capture some reads through Marlin.

bob808, 2019-11-18:
Here's how it looks when issuing M122 X
[image]

The two consecutive commands are constant retries to the tmc2208, they happen constantly. In the middle is the M122 command.
This is the first info in the M122 command:
[image]

boelle, 2019-11-24:
@sjasonsmith do the new captures make any bells ring?

sjasonsmith, 2019-11-24:
I just realized that the bottom picture shows a slave address of 1 being sent, when for a 2208 this should always be zero. The CRC is correct for those bytes, which tells me that address was intentionally sent, and this isn't just a timing error.

My first guess was that Marlin was misconfigured for a TMC2209 with a slave address set, but that is not the case according to the posted configs.

@bob808, are you sure the configs you posted are what you are currently running?

Can you also make sure you are using the latest version of bugfix-2.0.x? If Marlin or TMCStepper is actually causing a slave address to be added that is not in your configuration file, this issue would be much more widespread.

bob808, 2019-11-24:
I'll try again, I did tinker with the settings since and I want to test again fresh, just to make sure.

bob808, 2019-11-25:
I used a fresh version from a couple of days ago, only modified Configuration.h, Configuration_adv.h and the pins file, all added in the archive.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3889058/config.zip)

Still doesn't work, TMC connection error on display, serial says:
`Testing X connection... Error: All LOW`
And logic analyzer shows this as the first command after issuing M122 X
[image]

sjasonsmith, 2019-11-25:
OK, that looks better, it has a '0' for the slave address, as it should.

Two questions:

1. Where are your probes attached? Their position relative to any resistor(s) in the line is important.

2. Where is the resistor attached? There should be a resistor between the TMC UART pin and your TX pin. There should be a direct connection between the TMC UART pin and the RX pin (no resistor).

When I see a valid read command going out with no response, it makes me think that the resistor is connected to RX instead of TX.

Also, can you verify that you are powering your stepper drivers? If you are only powering with USB they will not respond.

Right now it seems like a possible wiring problem. This is what I would expect to see if the resistor it attached to the RX pin instead of the TX pin.

bob808, 2019-11-25:
I don't have HV, I feed it 5V on it, and seems to be able to talk with scriptcommunicator. But I can add the required higher voltage.
I have a 1K resistor between RX/TX. 
Ah, ok, had RX/TX swapped. Just notice, I wanted to be sure it's ok before finishing my reply :)
Thank you for your help!
I think I tried swapping RX/TX before but apparently it was that extra bit that was messing things up then.

closed 2019-11-25 by bob808

sjasonsmith, 2019-11-25:
Glad to hear it worked! I have a Due here, but I don't have a shield I can actually use with it, so I wasn't able to try it myself!

bob808, 2019-11-25:
I finished designing mine and apparently everything should work.
I have 3xUART capable driver slots (hardware UART), all 5 capable of SPI. One with UART has two connectors for motors, for Z axis.
8 thermistor inputs, one max31865 sensor.
6 PWM fans, 3 of which can be powered by 24V/12V/5V
2 servos and one bltouch connector
3 endstops
two extruder heater connections, car fuses for VMOT/extruders/bed. poly fuses for lower current rails. bigger mosfets are properly driven, most comms are also buffered, and I managed to get it pretty decent on two layers:
[image]
Everything should work. I'll see when I get it.

boelle, 2019-11-26:
@bob808 nice looking board are the files for that published anywhere? i assume its a shield for a due

i tried to make a Ramps-FD v2.2 but i gave up on it as it was to expensive and to much surface mount

bob808, 2019-11-26:
not yet.
I tried remaking a ramps-fd but I figured I should start from scratch since I have a different configuration.
Basically all pins are used/broken out or both. I think there's one pin left (for filament runout sensor if using software SPI for tmc drivers/max31865. I have a jumpered setup for that. if using hardware then there's 3 extra pins left.
there's also a precise voltage reference for the analog stuff, and I tried to keep it somewhat isolated. I don't know how much that matters, hope I find the time to take a look at the performance between with and stock.
ground looks like this:
[image]
there are a few unconnected tracks that show but I was lazy in correctly labeling the different 5V supplies etc.

bob808, 2019-11-26:
I am really curious if I will be able to directly control a lower powered 405nm laser diode with the 5V regulator. I won't be needing bltouch or any servo so I can adjust that converter to something comfortable for a 100mA laser diode. I got one that's rated for 190mA or so, running it at 100mA would be more than enough for presensitized pcb work. as it happens this diode runs at about 5V for 100mA. sure there might be some drift but there's plenty room both ways for presensitized pcb. I'd be practically using my fun 405nm pointer for this.

bob808, 2019-11-28:
Managed to squeeze in a laser driver made up by a LM317 controlled by a MCP4018 digipot as I've seen there's already support for it in Marlin. That way I could control the laser beam intensity via I2C, then also be able to PWM the diode using one of the FAN connectors. Also has optional classic FAN use as well, just that I can connect it via jumper to the LM317 output.
That should give 1A of current to the laser diode. And I can adjust the voltage that the LM317 receives from the cheap dc/dc adjustable converters. That way I generate the least amount of heat in the LM317, and can squeeze current out of it.
Just ordered the board, I can hardly wait to get it!
[image]
Yeah I know it's red....

boelle, 2019-11-28:
what will it cost in components ?

bob808, 2019-11-28:
Really not that much. I mean, the max31865 is about 5$ depending where you get it from. 3x 74hc125 at 0.35$ each, one 74hc02 at 0.38$. you can use the cheapest i2c eeprom. AOD240 bed mosfet I found at 0.80$, the two extruder mosfets AO4430 at about 0.85$. the six fan mosfets are between 0.15$-0.45$ in 5s quantities. six ES1F diodes at about 0.1$ each. lm317 is about 0.60$ and the MCP4018 digital pot is around 0.50$.
there's no special requirement for passives, and almost all are 0805 or a few 1206.
the only thing that I'm not sure of is the voltage reference. I'm not sure where the point of diminishing returns is for it's precision vs cost. I used a MAX6070B 3V reference which costs about 2$.
Around 13-15$ without passives and connectors. Most of the parts are found in singles or fives for the smaller ones. They are just about the cheapest/best for the application. I guess any part that fits the footprint could be used as all are fairly standard for that use.
For example for lm317 you could use a lm338 or lt1083/4/5 etc.
the dc/dc converter modules are 1$-1.5$ each. and it uses the esp12e connector for esp stuff. those are fairly ubiquitous.
I don't remember much about the fuse connectors. I need to look for their code.

bob808, 2019-11-28:
AO3404A are the fan mosfets. angry little fellas. 5.8A rated, and the lowest one, that can be used for laser, is being driven by the 74hc02 driver at 5V (had a free connection left). So it for sure can push the 1A for laser, without a sweat.
Also basically the two DC/DC converters are ultimately fully adjustable, so the one used for VIN for DUE can be anywhere between 6-12V, even a bit higher I think. Then the 5V one can be set to anything as well, as long as you are not using the bltouch probe socket and the two servo connectors as they are 5V use only. apart from that, you can go wild on any of the two DC/DC converters. lm317 has access to both of them with a solderable jumper, and separately the three upper fan headers are connectable to either 24V either any of the two dc/dc converters. as long as you don't go overboard with current (I'd say about 1A each), you can power anything you want with PWM.

bob808, 2019-11-28:
for the car fuse socket I've used Keystone 3522:
https://www.digikey.com/product-detail/en/keystone-electronics/3522/36-3522-ND/228600
or like this:
https://www.digikey.com/product-detail/en/keystone-electronics/3522-2/36-3522-2-ND/151571
0.20$ each and you need 6, so that's another 1.2$, and I guess the sd-card connector is about 2$ or something. I'm recovering mine from e cheap sd-adapter, used same footprint. the eeprom I'm scavenging from another cheap adapter as well. that has a smd footprint inside the DIP one, just in case it burns I can find something to shove in there.

bob808, 2019-11-28:
If you like it and want to make one I could send you a board, after I get them and they check out. I'll also give you the schematic/details etc.

boelle, 2019-11-28:
will have to figure it, i do have 2 due's i had left over from when i gave up building a ramps-fd

but it will be after new year before i make up my mind, are you on discord also? if so you should be able to find me (same pic and name bo herrmannsen)

bob808, 2019-11-28:
don't have Discord but might get it. I need to wait for the boards anyway so around new year I'll see what's what.
never played with digital potentiometers before. I also added a series resistor footprint, just for that extra granularity. hope the code would be simple to use.