MarlinFirmware/Marlin #15822: [BUG] [bugfix-2.0.x] TMC 2130 Sensorless Homing not triggering

Thoralf690, 2019-11-06:
### Description

Sensorless Auto Homing does not trigger when homing both X and Y

### Steps to Reproduce
[new_Config_nov2019.zip](https://github.com/MarlinFirmware/Marlin/files/3816189/new_Config_nov2019.zip)

Configuration:
SKR 1.3 with Bigtreetech TMC 2130 V3.0 SPI on all axes, using the build-in jumpers off the board.
1. Using G28 (or G28 X0 Y0) causes printer to smash into frame without endstop being recognized.
2. Just homing X or Y separately, triggers sensorless homing without a problem

For now, I deactivated QUICK_HOME, which suppresses the problem.
The described behaviour was not present in my old Marlin 2.0 setup (August 2019
[old_Config_aug2019.zip](https://github.com/MarlinFirmware/Marlin/files/3816190/old_Config_aug2019.zip)
)

#### Additional Information

Firmware was flashed through Octoprint

masdisaster, 2019-11-06:
Same here;

SKR 1.3 with Bigtreetech TMC 2130 V3.0 SPI on all axes , Deltaboot (AnyCubic Kossel) ,homing XYZ smash to max position - stall_guard not triggering .
Before working OK. Now even if i recompile with old setup and configs make no difference , seams some updated framework broke sensorless homing.

thisiskeithb, 2019-11-06:
Strange. I'm running an SKR 1.3 with TMC5160s on all axis and and sensorless homing works fine.

boelle, 2019-11-11:
@Thoralf690 still an issue?

Thoralf690, 2019-11-11:
@boelle I haven't tried with the most recent state, but will do as soon as I get to my printer

masdisaster, 2019-11-11:
I tried recent build …. still nothing. If i check with osciloskope DIAG pins … there is still no stall guard triggering to see.

kiryam, 2019-11-17:
Have same issue. with latest 2bd6dc02d677d2b1f19795c7cceffaa12b0d5e40

boelle, 2019-11-24:
@Thoralf690 gotten to the printer yet?

maybe ask in these place's too:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

Thoralf690, 2019-11-26:
@boelle I had but unfortunately some hardware problems got in the way that I wanted to resolve beforehand.

Today I reconfigured Marlin, downloaded yesterday, 2019-11-25 (from scratch) and the problem persists. 

Parameters I tried to vary:
- `SQUARE_WAVE_STEPPING` + `MINIMUM_STEPPER_PULSE 2`
- `SQUARE_WAVE_STEPPING` alone -> current config
- `MINIMUM_STEPPER_PULSE 2` alone
- enabled/disabled `HOME_USING_SPREADCYCLE` -> no changes

I have to admit, for now I'm happy without `QUICK_HOME`.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

Thoralf690, 2019-12-07:
@boelle I didn't update yet, will happen on Monday

boelle, 2019-12-11:
@Thoralf690 did the update help?

Thoralf690, 2019-12-11:
@boelle no, no change in behavior at all.

Allarabie, 2019-12-15:
Same problem here.. SKR 1.3 with Bigtreetech TMC2130 V3.0 SPI on Delta.
Today, I tried the lates 2.0.x Bugfix and still Sensorless homing does not trigger.
I've tried stall_sensitivity values from -64 to 63 with no change in behavior.
Anyone figure this issue out?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

masdisaster, 2019-12-24:
I can confirm for delta printer ,skr1.3 and TMC2130 V3 ,,,, issue persist.
BTW
In Sumer edition bugfix 2.0.0 worked all flawlessly (don´t remember exact date on which it was still working) some later editions broke it completely !

boelle, 2019-12-24:
@masdisaster did you use the config that @Thoralf690 posted?

Thoralf690, 2019-12-24:
@boelle  the originally posted config isn't valid anymore anyway, to many changes in syntax since my initial post. I will be able to post my latest version earliest in mid January, no way to access my files till then.

boelle, 2019-12-24:
no problemo we just want to keep the ball slowly rolling :-D

masdisaster, 2019-12-24:
@boelle yes i did… only changes i made ….adopted for delta style printer.
The frustrating thing is this all worked perfectly in some early release …. after i recompiled for some minor changes (extruder steps and direction) suddenly stopped working at all (like some updated framework broke it). :-(

boelle, 2020-01-08:
@Thoralf690 @masdisaster still no good news with the latest bugfix 2.0.x?

masdisaster, 2020-01-08:
@boelle i didn`t try latest build jet due to lack of time …. i would try this in couple of days and i hope that this is fixed .because its really frustrating that this dont work like before :-(

boelle, 2020-01-15:
@Thoralf690 @masdisaster tried yet?

Evovita, 2020-01-19:
I experienced what looks like the same issue and logged it @ [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/29](url) before I found the bug here . Using BigTreeTech SKR Pro 1.1, BigTreeTech TMC2130 SPI Drivers, and Marlin 2.0.1 configured for CoreXY. Unable to trigger sensor-less homing at all on X or Y Axis.

boelle, 2020-01-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-23 by boelle