MarlinFirmware/Marlin #15824: [BUG][BUGFIX 2.0.x] Doesn't compile with Delta config for SKR 1.3

Ohmarinus, 2019-11-06:
### Description

So the latest Marlin bugfix 2.0.x doesn't want to compile through PlatformIO. I am using the Delta example config files and have put in my own values.

The board I'm using is the SKR v1.3

### Steps to Reproduce

Zipfile included
[nocompile_marinus.zip](https://github.com/MarlinFirmware/Marlin/files/3816899/nocompile_marinus.zip)

What I did:
1. Copy and paste Delta example files
2. Add in personal values (enabled TMC2208 drivers etc)
3. Hit the 'build' button, so not the upload, only the build button in Atom

**Expected behavior:**
For Marlin to compile without errors

**Actual behavior:**
This:
`Compiling .pio/build/LPC1768/src/src/module/endstops.cpp.o
Compiling .pio/build/LPC1768/src/src/module/motion.cpp.o
Compiling .pio/build/LPC1768/src/src/module/planner.cpp.o
Compiling .pio/build/LPC1768/src/src/module/planner_bezier.cpp.o
In file included from Marlin/src/module/../inc/MarlinConfig.h:35,
                 from Marlin/src/module/configuration_store.h:24,
                 from Marlin/src/module/configuration_store.cpp:47:
Marlin/src/module/configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
**Marlin/src/module/../inc/Conditionals_post.h:1478:75: error: 'nozzle_to_probe_offset' was not declared in this scope**
       #define DELTA_CALIBRATION_RADIUS (DELTA_PRINTABLE_RADIUS - _MAX(ABS(nozzle_to_probe_offset.x), ABS(nozzle_to_probe_offset.y), ABS(MIN_PROBE_EDG
E)))
                                                                           ^~~~~~~~~~~~~~~~~~~~~~
Marlin/src/module/configuration_store.cpp:2350:32: note: in expansion of macro 'DELTA_CALIBRATION_RADIUS'
     delta_calibration_radius = DELTA_CALIBRATION_RADIUS;
                                ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/LPC1768/src/src/module/printcounter.cpp.o
Marlin/src/module/../inc/Conditionals_post.h:1478:75: note: suggested alternative: 'probe_offset'
       #define DELTA_CALIBRATION_RADIUS (DELTA_PRINTABLE_RADIUS - _MAX(ABS(nozzle_to_probe_offset.x), ABS(nozzle_to_probe_offset.y), ABS(MIN_PROBE_EDG
E)))
                                                                           ^~~~~~~~~~~~~~~~~~~~~~
Marlin/src/module/configuration_store.cpp:2350:32: note: in expansion of macro 'DELTA_CALIBRATION_RADIUS'
     delta_calibration_radius = DELTA_CALIBRATION_RADIUS;
                                ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/LPC1768/src/src/module/probe.cpp.o
***** [.pio/build/LPC1768/src/src/module/configuration_store.cpp.o] Error 1**
========================= [FAILED] Took 132.84 seconds =========================`

#### Additional Information

If there is anything missing in my description I'd gladly hear it. I'm new to using github for these kind of things.

ellensp, 2019-11-06:
You don't seem to have any probe type defined. But you do have NOZZLE_TO_PROBE_OFFSET defined..  this is causing the issue.

Ohmarinus, 2019-11-07:
Thanks, you're right, that fixed it. However, I didn't set the NOZZLE_TO_PROBE_OFFSET..

Shouldn't this be commented out then in the Delta example config files?

boelle, 2019-11-11:
will close this one as fixed

> Thanks, you're right, that fixed it.

closed 2019-11-11 by boelle

sjasonsmith, 2019-11-13:
@Ohmarinus is right, it doesn't make sense to have examples that cannot be compiled in their default state. I've fixed the build issue in PR #15887, and discovered a lot of other delta probing issues along the way.

Ohmarinus, 2019-11-13:
> @Ohmarinus is right, it doesn't make sense to have examples that cannot be compiled in their default state. I've fixed the build issue in PR #15887, and discovered a lot of other delta probing issues along the way.

Thanks Jason! Indeed, this is what I meant. I'm having a hard time getting Marlin to work with my inductive probe. When using G33 and G29 the probe also doesn't respect the offsets I've given it, and more. I wonder what the other Delta probing issues were in this build. When will your fixes be available in a new bugfix?

sjasonsmith, 2019-11-13:
> When using G33 and G29 the probe also doesn't respect the offsets I've given it, and more.

You can follow the link to the PR for more information on what I encountered, my description is pretty detailed. Thinkyhead reworked some of the code after my initial posting, so I can't guarantee it is functional right at this moment. I'll be testing his changes tonight.

I don't have an exact timeline, I would expect a few days of churn, although you can always pull from my branch referenced in the PR to try it out before it is merged.

If your "and more" includes other probe-related issues I didn't already mention in that PR's description, feel free to add a comment to that PR with details we might had missed.

Guoshengchao, 2020-06-11:
Hello, I'm having the same problem.How did you solve the problem?