MarlinFirmware/Marlin #15825: [BUG] (short description)

el-quique, 2019-11-06:
### Description

The last marlin generates an error on the LCD Full Graphics. I try Kimbra and it works well.
I don't want to use Kimbra it was just a test

### Steps to Reproduce

compiling with arduino ide 1.6.13 and 1.8.10

THX
Errors LCD
[image]

No error LCD
[image]

el-quique, 2019-11-06:
[image]

el-quique, 2019-11-06:
Maybe errors in pins_runba.h?

ellensp, 2019-11-07:
This isn't a bug, its the slow LCD supplied with your printer.  Marlin tries to push the LCD as fast as it can so it doesn't have to waste cpu cycles updating the LCD vs doing other important tasks.
add #define ST7920_DELAY_3 DELAY_NS(63) to your configuration.h (this is the default value for a 16mhz atmega based controller)
Try larger values than 63, many report 125 works on horridly slow LCD's

boelle, 2019-11-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

3Dangel, 2019-11-07:
same as I suggested for another reported issue

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER /
#define ST7920_DELAY_1 DELAY_NS(125) //these help keep the display from being weird
#define ST7920_DELAY_2 DELAY_NS(125)
#define ST7920_DELAY_3 DELAY_NS(125)
#endif

hope this helps

el-quique, 2019-11-07:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[Marlin-bugfix-2.0.x-11-10-19-USB-FIXMOUNT-Desmontable.zip](https://github.com/MarlinFirmware/Marlin/files/3821957/Marlin-bugfix-2.0.x-11-10-19-USB-FIXMOUNT-Desmontable.zip)

el-quique, 2019-11-07:
> same as I suggested for another reported issue
> 
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER /
> #define ST7920_DELAY_1 DELAY_NS(125) //these help keep the display from being weird
> #define ST7920_DELAY_2 DELAY_NS(125)
> #define ST7920_DELAY_3 DELAY_NS(125)
> #endif
> 
> hope this helps

Solved. It's strange because i use Marlin 1.1.x until 2.0 (September) I never put those options and never problems. Thank you very much for your time

ruggb, 2019-11-08:
see #15745 and #15833

el-quique, 2019-11-08:
> see #15745 and #15833

Thx. Solution with 3Dangel comment.

closed 2019-11-08 by el-quique