MarlinFirmware/Marlin #15826: [BUG] SKR mini E3 v1.2 compile error

mpguerreiro, 2019-11-06:
### Description

I just received my SKR mini E3 v1.2 and I want to change some parameters in the firmware for a Ender 5, but when compiling it always gives the same error! (error log below) I've spent hours and got nothing!!

### Steps to Reproduce
 
Fleshing a new Firmware

**Expected behavior:**

A new firmware to be flashed.

**Actual behavior:**
In the end: STM32F103RC_bigtree      FAILED 

#### Additional Information

[image]

thisiskeithb, 2019-11-06:
Read the error message. It’s telling you to install git.

mpguerreiro, 2019-11-07:
> Read the error message. It’s telling you to install git.

Thank you a tone, problem solved!!

closed 2019-11-07 by mpguerreiro