MarlinFirmware/Marlin #15827: [FR] Clear known positions axis

hydrogen18, 2019-11-07:
### Description

I have a printer that uses the X-Y-Z-E axis from a stock Creality Ender 3. I'm running Marlin 1.x on a MKS GenL . My Z-Axis falls under its own weight under normal circumstances. I've added a BLTouch & use bilinear ABL. I use cura. Basically what happens is 

1. Run a print 
2. Cura finishes,
3. Print head is parked
4. disables X, Y, & E to save power
5. Z axis times out
6. Z axis hits beds
7. Start another print
8. During the G28 homing cycle, Marlin runs the Z axis as hard as possible into the bed, then drags it across the bed 

Obviously step 8 kills the nozzle of the extruder. You can fix this by manually holding the Z-Axis near Z-Max before G28 starts.

# Possible solutions
I poked around in the code and there is in `Marlin_main.cpp` a global variable `axis_known_position` . It looks to be a bitmask and the initial value must be 0? Once G28 is complete is the X,Y,Z bits are set. I can't tell if I can just clear bits at any time without updating any other state? That's is mainly what is holding me back on trying to fix this. I don't see anywhere in the code where this is ever cleared, just has bits added to it.

## Solutions?
1. Add to G28 a flag of "forget all axis"
2. Add another G-Code for "forget this axis"
3. Add a define like "Z_AXIS_HOMING_AMNESIA" and if it is set we clear Z bit for `axis_known_position` anytime G28 is homing the Z-axis

Basically if someone can answer if it is safe to just clear bits in `axis_known_position` I think I can implement this. I'm unsure what solution of the above is the best, someone might have a better idea anyhow.

thisiskeithb, 2019-11-07:
Why wouldn’t you set `DISABLE_INACTIVE_Z` to false to keep your z stepper active instead?

https://github.com/MarlinFirmware/Marlin/blob/a84e3d1b8049653345d694c1171d6d3491ebf618/Marlin/Configuration_adv.h#L655

hydrogen18, 2019-11-07:
That is one valid solution to this problem. However, my printer often sits idle for long periods of time after a print. I'd rather not waste the power and I don't know about the impact on driver & motor life.

sjasonsmith, 2019-11-07:
@hydrogen18, have you tried Z_SAFE_HOMING? This lifts Z before homing X/Y, then homes Z in the center of the bed using the probe.

hydrogen18, 2019-11-07:
My `Configuration.h` has this line 

```
#define Z_SAFE_HOMING
```

It makes no difference, I can get this behavior with and without.

sjasonsmith, 2019-11-07:
How about `HOME_AFTER_DEACTIVATE`?

sjasonsmith, 2019-11-07:
I think that does exactly what you want, but for all your motors, not just Z.

hydrogen18, 2019-11-07:
The implementation for that is as follows

```
  bool axis_unhomed_error(const bool x/*=true*/, const bool y/*=true*/, const bool z/*=true*/) {
    #if ENABLED(HOME_AFTER_DEACTIVATE)
      const bool xx = x && !TEST(axis_known_position, X_AXIS),
                 yy = y && !TEST(axis_known_position, Y_AXIS),
                 zz = z && !TEST(axis_known_position, Z_AXIS);
    #else
      const bool xx = x && !TEST(axis_homed, X_AXIS),
                 yy = y && !TEST(axis_homed, Y_AXIS),
                 zz = z && !TEST(axis_homed, Z_AXIS);
    #endif
```

In other words axis_homed is swapped for axis_known_position.

These values are identical unless an axis has been disabled via one of the `disable_` macros.

The implementation of `gcode_g28` doesn't seem to invoke `axis_unhomed_error`, so changing that define isn't going to do anything from what I can tell.

The section in `gcode_g28` that needs to change is 

```
    const float z_homing_height = (
      #if ENABLED(UNKNOWN_Z_NO_RAISE)
        !TEST(axis_known_position, Z_AXIS) ? 0 :
      #endif
          (parser.seenval('R') ? parser.value_linear_units() : Z_HOMING_HEIGHT)
    );

    if (z_homing_height && (home_all || homeX || homeY)) {
      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      destination[Z_AXIS] = z_homing_height;
      if (destination[Z_AXIS] > current_position[Z_AXIS]) {

        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(LEVELING))
            SERIAL_ECHOLNPAIR("Raise Z (before homing) to ", destination[Z_AXIS]);
        #endif

        do_blocking_move_to_z(destination[Z_AXIS]);
      }
    }
```
My guess is that  `do_blocking_move_to_z` after startup winds up being a relative move to the **actual** Z. After step 6 in my example the move becomes an absolute move based on the **believed** Z. 

If the current believed Z position is numerically less than `z_homing_height` then this movement winds up being downwards into the bed. Keep in mind this actually occurs _before_ X,Y homing and not during Z homing.

Interestingly if Cura parks the head at say, 20mm above the bed. It appears presently it will always run `G1 F3000 Z20 ` after a successful print. Then its possible running `G28 R25` during startup will insure that I get a minimum of 5mm of raise, up to 25mm. 

However, this assumes that Cura always successfully finishes a print. Which is not always the case. Still it might be better than nothing.

thisiskeithb, 2019-11-07:
This may actually be a bug with how homing is handled.

I have a BLTouch config with `Z_SAFE_HOMING` enabled and that works fine/lifts the nozzle as it should if you run G28 after bootup.

If I disable steppers & send G28 again, Z does not raise.

boelle, 2019-11-07:
oki title and label corrected until we hear from @hydrogen18

hydrogen18, 2019-11-07:
What are you waiting to hear from me about?

boelle, 2019-11-07:
@hydrogen18 read what @thisiskeithb wrote above ^^^^

hydrogen18, 2019-11-07:
His description of the problem accurately describes it to me.

thisiskeithb, 2019-11-07:
I was just confirming the issue. The Z axis should raise and it’s not.

sjasonsmith, 2019-11-09:
> I just ran a test and this may actually be a bug with how homing is handled.

I just reproduced the same behavior.

> My guess is that do_blocking_move_to_z after startup winds up being a relative move to the actual Z. After step 6 in my example the move becomes an absolute move based on the believed Z.

When I read that code it looks like it wouldn't move the nozzle down, it would just fail to raise it. I think you could fix by changing the single line that  sets destination.z:

```
      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      destination.z = TEST(axis_known_position, Z_AXIS) ? z_homing_height : current_position.z + z_homing_height;
```

@hydrogen18 or @thisiskeithb, are either of you intending to post a PR to fix this? If not, I could try the change I suggested above and put something together.

hydrogen18, 2019-11-09:
@sjasonsmith If the "believed" position is below it, it fails to move.  If the "believed" position is above it, it runs the nozzle into the bed.

In both cases, the "believed" position is the position from the last move G-code _before_ the Z-axis falls under its own weight.

The code you've shared must be from a newer version of Marlin, I'm using the latest 1.x release.

sjasonsmith, 2019-11-09:
> I'm using the latest 1.x release.

OK, I'm looking at bugfix-2.0.x. I've never worked in 1.x.

sjasonsmith, 2019-11-10:
I just posted a PR to fix this for bugfix-2.0.x.

As for fixing it in 1.x, my first recommendation is that @hydrogen18 try switching over to 2.x, where there is going to be much better support for fixing anything moving forward.

Perhaps the maintainers would accept a similar change in bugfix-1.1.x (since it does seem like a bug, not a feature), but I won't be doing it myself.

thisiskeithb, 2019-11-10:
> I just posted a PR to fix this for bugfix-2.0.x.

@hydrogen18: This patch works and will now raise the hotend (or lower the bed depending on the config) if steppers are disabled and not just on first boot.

hydrogen18, 2019-11-10:
Can I run 8-bit hardware on 2.x.x? My understanding is it was 32-bin only?

thisiskeithb, 2019-11-10:
Marlin 2.0 is “for all the boards” and works on 8bit just fine 🙂

From [the `bugfix-2.0.x` readme](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/README.md):

> Marlin 2.0 introduces a layer of abstraction so that all the existing high-level code can be built for 32-bit platforms while still retaining full 8-bit AVR compatibility. Retaining AVR compatibility and a single code-base is important to us, because we want to make sure that features and patches get as much testing and attention as possible, and that all platforms always benefit from the latest improvements.

sjasonsmith, 2019-11-10:
@hydrogen18, I actually wrote and tested that fix on an 8-bit MKS Gen L. That said, I highly  recommend using 32-bit boards, I only had this board connected to try to reproduce a different issue.

I mostly use 32-bit boards, but I've also used 2.0.x on 4 different 8-bit boards without issues.

boelle, 2019-11-16:
@hydrogen18 did you give 2.0.x a try?

hydrogen18, 2019-11-16:
I appreciate the attempt at providing a solution for this problem, but I simply don't have the time to attempt upgrading several different machines to a 2.0 version. At some point I will attempt to fix this on the 1.x branch.

sjasonsmith, 2019-11-24:
@boelle, this is fixed in 2.0, but is still an issue in 1.1.9, which OP is used.
I can see the argument for it being an actual bug that could be patched in 1.1.9, if someone were so inclined. Is there a label such as "1.x" that can be added, so that it is clear to people working on 2.x issues that this doesn't apply?

boelle, 2019-11-24:
not a label as such but i have edited the title to include 1.1.9

but people should really use 2.0.x, i have also voiced more than once that 2.0.x becomes the default not as people from time to time download 1.0.x as that is the default

boelle, 2019-12-01:
@hydrogen18 i had a look at your last post and there is a kind of dilemma here, 1.1.x is past EOL but 2.0.x is not quite ready yet

the issue list is getting shorter but there are still ~80 open issue that are not confirmed yet and then another ~19 that is confirmed

i assume that once the major part of that is done 2.0.x will be close to a 2.0 release

as harsh as it sounds it does not help to have issues based on 1.1.x open as that will never get fixed anyway

and there is only one maintaner on marlin so it will take some time

the very best and helpfull thing is to test just one machine on 2.0.x and let us know

boelle, 2019-12-03:
17 days since @hydrogen18 replied so assume he had moved on and dont have an issue anymore

closing and we can always reopen later

closed 2019-12-03 by boelle