MarlinFirmware/Marlin #15828: [BUG] Speaker Configuration E3 Mini V1.2

diabl0w, 2019-11-07:
When setting #define SPEAKER, the speaker doesn't work
when setting //#define SPEAKER, the speaker does work

Using Marlin 2.0.x bugfix branch and without changing any other config
Board: BTT_SKR_MINI_E3_V1_2

ellensp, 2019-11-07:
Does it have a speaker? I suspect not, most displays have a buzzer, a very different thing. You don't define SPEAKER when you have a buzzer.

diabl0w, 2019-11-07:
> Does it have a speaker? I suspect not, most displays have a buzzer, a very different thing. You don't define SPEAKER when you have a buzzer.

Thanks for the explanation

diabl0w, 2019-11-07:
@ellensp according to https://github.com/MarlinFirmware/Marlin/pull/15781#issuecomment-549172924 , my board does have a speaker... So, I am not sure why I am not getting tones when enabling SPEAKER.
Reopening issue and correcting original post to be more clear

qwewer0, 2020-03-19:
In the Creality example file for the Ender 3, the SPEAKER is defined, and with the original Creality board the piezo speaker/buzzer (LCD 12864 display) was able to produce tones, like [here](https://www.youtube.com/watch?v=eDqj7CBiZzs).

But this isn't the case with the SKR Mini E3 v1.2 board:
- If the SPEAKER is defined, the printer won't make any sound.
- If the BUZZER if defined, the printer makes short beeps on clicking in the printer menu, and a longer beep when saving/loading setting, but It only makes a single tone/frequency with the M300 command.

So, as far as I know, the printer - Ender 3 in my case - with the SKR Mini E3 v1.2, should be able to play different tones/frequencies, but it doesn't.

Should @diabl0w reopen this, or should I open a new issue for this? @ellensp, @thinkyhead, @jbrazio

diabl0w, 2020-03-19:
I will reopen this because, after I closed this issue, I have since seen a youtube video of a stock ender 3 playing multiple tones. 

I am thinking there may be confusion about a speaker vs a monotone buzzer vs a multitone buzzer, but I am not really sure if there is such things

diabl0w, 2020-03-20:
reopenning, see above new comments

skipfie, 2020-03-26:
Can reproduce on SKR DIP 1.0 with STM32F103RE_btt environment.

Buzzer produces single tone when #define SPEAKER is commented, but when #define SPEAKER is uncommented, the buzzer produces no noise.

thisiskeithb, 2020-03-28:
The SKR Mini E3 series need to have `SPEAKER` disabled/commented out and should probably automatically undefine it in the pins file to prevent further confusion.

qwewer0, 2020-03-29:
Just want to make things clear in my head.
Enabling SPEAKER in marlin for SKR Mini E3, can't work because the board has hardware limitations for 12864 display beeper, or is it because the 12864 displays has beeder and a beeper "can't" make different sounds with M300 or is there a marlin limitation for beeper to work with M300?

Chronicmemecrafter, 2020-04-13:
> Just want to make things clear in my head.
> Enabling SPEAKER in marlin for SKR Mini E3, can't work because the board has hardware limitations for 12864 display beeper, or is it because the 12864 displays has beeper and a beeper "can't" make different sounds with M300 or is there a marlin limitation for the beeper to work with M300?

I believe that you are correct from what I have been reading in this thread but I am going to reformat the question to give a solid answer.

On the stock ender 3 (not pro) 1.13 mainboard [which right now runs Marlin 1.16.2 (I think, someone please confirm, I don't have an original only a pro) pretty outdated] the beeper can play different tones with stock firmware using M300. On my stock (electronics wise) ender 3 pro with the 1.14 mainboard, I can only play one tone. I can still control the length but it ignores any pitch instructions when I send them via octoprint terminal. Qwewer you're saying that on your SKR mini E3 V1.2 you get the same results as me on my ender 3 pro stock mainboard. As far as I know, the Ender 3 and Ender 3 pro use the same firmware right now with probably a few slight changes for compatibility.

What we are trying to figure out is what of the three things is causing the problem. The mainboard (i.e. Melzi 1.13, Melzi 1.14, SKR mini E3 V1.2), the firmware (i.e. custom 2.0.x Marlin, 1.16.2 Creality), or possibly the buzzer on the LCD between the regular and the pro.

What I need to know is whether you are using an Ender 3 or a Ender 3 pro.
Because if you are on an Ender 3 then the problem could be either on the mainboard side or the custom firmware side. Sadly there is no way of testing whether it is the firmware or not because 1.16.2 is not compatible with 32-bit boards.
If you are using an Ender 3 Pro then it could be any one of the 3.

What we need is someone who owns both machines to swap out the mainboards and LCDs to the other machine and test whether it is a mainboard or LCD issue

If the person testing it moves the normal ender 3 LCD to the pro with 1.14 board stock firmware and it works, then the LCD is the problem.

If the person testing changes the motherboards and the pro works with its original LCD and firmware, then it is a problem with the newer mainboard not communicating the same way or something.

If they try both of the possible swaps and it still doesn't work with either then it is a problem with the firmware communicating to the mainboard and they will have to check the firmware.

Of course, this goes much more in-depth but I'm pretty sure anyone with a good amount of experience can see where I'm going by this point.
The other thing that is a possible (though not very likely) is that that video was made using slightly older firmware as it was published a few months before 1.16.2 was the newest download available on Creality's website.

Obviously, there are many flaws with this method of testing but I will leave those to get figured out by someone who has way more experience than me on this. Also, I've been typing this for long and it's starting to hurt.

qwewer0, 2020-04-13:
@Chronicmemecrafter I have a non pro Ender 3.

Chronicmemecrafter, 2020-04-13:
> @Chronicmemecrafter I have a non pro Ender 3.

This means that the issue is most likely a communication error somewhere in the process of firmware to the mainboard to buzzer. This issue now is most likely the cause of the 1.13 board being different from the 1.14 board on the pro. What we need is someone with both to swap just the boards and if the pro works with the 1.13 then it is because of the motherboard.

Chronicmemecrafter, 2020-04-13:
As a follow up to my last comment because I keep noticing new things we could try, why not try setting //#define LCD_FEEDBACK_FREQUENCY_HZ 5000 (from the default marlin config or in particular with the mini E3 board example file) to a different frequency, flash and test if it actually can do it. I don't have a bootloader flashed board yet so I can't do it until my new board arrives but when it does ill let you know.

Chronicmemecrafter, 2020-04-14:
Ok so I just tested the speaker on the tft35 E3 V3.0 using the really helpful gcode menu on the lcd and it only made 1 sound no matter what I tried. So it Is probably not the buzzer that is causing issues. Gonna check the feedback beep frequency and flash that to see if it works that way.

(Btw, omg the board is so fricking silent now, I mean I expected it to be silent but not this quiet. Upgrade well worth it!)

qwewer0, 2020-04-14:
Can't change the frequency with LCD_FEEDBACK_FREQUENCY_HZ.

diabl0w, 2020-04-14:
> Can't change the frequency with `LCD_FEEDBACK_FREQUENCY_HZ` only the duration with `LCD_FEEDBACK_FREQUENCY_DURATION_MS`.

does it still play a sound when changing the frequency or no sound?

qwewer0, 2020-04-14:
LCD_FEEDBACK_FREQUENCY_HZ doesn't change the frequency but it still plays the beep.

diabl0w, 2020-04-14:
> LCD_FEEDBACK_FREQUENCY_HZ doesn't change the frequency but it still plays the beep.

you tested by issuing m300 or by pressing buttons on the LCD? im guessing probably m300, can you try by pressing buttons on LCD?

qwewer0, 2020-04-14:
Tested by pressing buttons on the LCD.

diabl0w, 2020-04-14:
> Tested by pressing the button on the LCD.

okay, thanks for testing... unfortunately, it looks like changing that variable doesnt matter because it is only used for buzzers controlled thru I2c. so it doesnt tell us anything... our buzzer is controlled through buzzer.h

closed 2020-04-14 by diabl0w

qwewer0, 2020-04-21:
So, this is what I was able to get with some [help](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/fnghjxe/):

The SKR Mini E3 v1.2 board can make a BTT TFT35-E3 screen to produce M300 tones with Steaker defined in the configs, but only in the touchscreen mode, and not in the Marlin LCD mode. So, it seems that the board is capable to utilize the M300 command, but not through the LCD connection.