MarlinFirmware/Marlin #15831: [FR] is it possible to send folder to save at sd card using usb cable before print?

scarrrr316, 2019-11-07:
is it possible to send the file to save at sd card using USB cable before print?

I saw this function first time on tiertime's  3d printer, and I think this function is great. let me describe this function. when I click the print button in the software, it will send the .tsk file (just like gcode file )to the printer and the printer will save it in its own memory before it starts to print.

1.slice the 3d model 
2.click the print button
3.send file to the printer
4.printer will save at its own memory
5.start print

I knew that there is a similar function in Marlin firmware, but using gcode to send the .gcode file to the sd card of the printer is very slow,  so I hope that you will add the function that is said, or tell me how to achieve this function, thank you!

ellensp, 2019-11-08:
Trying to understand this request I found https://www.tiertime.com/introducing-the-Tiertime-Task/

reloxx13, 2019-11-08:
the speed is slow cause its not a usb connection, its a serial connection.   
there is no other way to transfer a file to sd card as over the serial connection.   
additional this is a FR to your slicer software and not for marlin.   
marlin supports coping files to sd card already.

ellensp, 2019-11-08:
The upload speed is related to your hardware. If the controller uses a usb/serial chip its really a serial device and is limited to the serial baud rate. More modern hardware has virtual UART's. These ignore the baud rate and transfer as fast as it can (up to USB speed - overheads)  Most 32 bit controllers have this, as do a few 8 bit controllers based on AT90USB1286. But your still limited by the speed of the SD card interface, all but Duet use a slow SPI interface...   No current controllers have enough ram to store an entire object.

scarrrr316, 2019-11-11:
> the speed is slow cause its not a usb connection, its a serial connection.
> there is no other way to transfer a file to sd card as over the given (mostly serial) connection of your printer.
> additional this is a FR to your slicer software and not for marlin.
> marlin supports coping files to sd card already.

oh,thankyou

tvixen, 2019-11-13:
I would suggest you to use a raspberry Pi, if you want to send the file to the SDcard/Pi.
The transfer to a Pi, is approximate 50 times faster that copy to the SDcard on the motherboard. 
You will also gain the wireless feature with a Pi. 

Have a look at it .... It's not that difficult to do your self:  https://www.youtube.com/watch?v=mnN4HVmjafs

Grogyan, 2019-11-13:
The host essentially needs to unmount the SD card, and remount it as a disk drive, then send the files over, as a drive, unmount it, then remount it ready for the printer board to use the SD card to print from.

shitcreek, 2019-11-13:
Closing because this is cannot be done on Marlin's end.

closed 2019-11-13 by shitcreek

Grogyan, 2019-11-13:
In fact Marlin itself already allows for this via M22, Release SD, and M21, Initialize SD