MarlinFirmware/Marlin #15833: [BUG] (FULL_GRAPHIC_SMART_CONTROLLER problem)

TreeTwo27, 2019-11-07:
I installed Mariln 2.0 version on the bigtree skr1.1 board using PlatformIO IDE, but the 12864lcd screen failed I searched for this problem, but I didn't see anyone with a similar problem.

### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [First Step] Set the value of delta and 12864lcd on the skr1.1 board before installing Marilyn 2.0.
2. [Second Step] I checked there was a problem with the LCD screen. And when I turned the knob, there was a reaction. After that, I manipulated the printer with printrun and it worked normally.
3. [and so on...] I searched Google for similar symptoms, but I couldn't find them. When I connected 12864lcd to another printer, I found that it reacted normally. Is it possible that the skr1.1 board had an initial defect?

**Expected behavior:** [What you expect to happen] I tried to safely apply the firmware, turn on the printer, and operate it.

**Actual behavior:** [What actually happens]Lcd 12864 is a strange screen showed.
If you move the knob, it will react. There's nothing wrong with the print run. Applying the lcd to another printer did not show the same symptoms
[image]
[image]
[image]
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3820189/Marlin.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

3Dangel, 2019-11-07:
This is what I use .
it is from another forum's member who's coding ..
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
#define ST7920_DELAY_1 DELAY_NS(125) //these help keep the display from being weird
#define ST7920_DELAY_2 DELAY_NS(125)
#define ST7920_DELAY_3 DELAY_NS(125)
#endif

try and see if any help

thisiskeithb, 2019-11-07:
I have an SKR 1.1 I can test this issue on, but can’t get to it until later tonight.

Ping me if I forget 😅

ruggb, 2019-11-08:
I already opened this bug on #15745

thisiskeithb, 2019-11-08:
> I already opened this bug on #15745

It's not clear they were the same issue.

ellensp, 2019-11-08:
I tried this on a SKR 1.3 and a geeetech clone of the RRD GLCD  (Closest I have, same MPU) 

Using the latest marlin bugfix 2.0,  changed #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3 and #define SERIAL_PORT -1
Then enabled the lcd #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. Updated .platformio with default_envs = LPC1768
Made no other changes 
It works fine. 
NB with this clone display you do need to flip the cables on the back of this display 180 degrees to work on the SKR 1.3, but you get nothing if you don't do this, not a scrambled screen.  

So this may be SKR 1.1 specific...

TreeTwo27, 2019-11-08:
> 이것이 내가 사용하는 것입니다.
> 그것은 .. 코딩있어 다른 포럼의 멤버 출신
> #DEFINE REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> #DEFINE ST7920_DELAY_1 (125) //이 도움이되는 것을 표시를 유지 DELAY_NS 이상한
> #DEFINE ST7920_DELAY_2 DELAY_NS (125)
> #DEFINE ST7920_DELAY_3 DELAY_NS (125)
> #endif 다음을
> 
> 시도하고 도움이 있는지 확인하십시오

I tried to add it, but it didn't work out. There was no change in the symptoms.

TreeTwo27, 2019-11-08:
> 이것이 내가 사용하는 것입니다.
> 그것은 .. 코딩있어 다른 포럼의 멤버 출신
> #DEFINE REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> #DEFINE ST7920_DELAY_1 (125) //이 도움이되는 것을 표시를 유지 DELAY_NS 이상한
> #DEFINE ST7920_DELAY_2 DELAY_NS (125)
> #DEFINE ST7920_DELAY_3 DELAY_NS (125)
> #endif 다음을
> 
> 시도하고 도움이 있는지 확인하십시오

> 이것이 내가 사용하는 것입니다.
> 그것은 .. 코딩있어 다른 포럼의 멤버 출신
> #DEFINE REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> #DEFINE ST7920_DELAY_1 (125) //이 도움이되는 것을 표시를 유지 DELAY_NS 이상한
> #DEFINE ST7920_DELAY_2 DELAY_NS (125)
> #DEFINE ST7920_DELAY_3 DELAY_NS (125)
> #endif 다음을
> 
> 시도하고 도움이 있는지 확인하십시오

I tried to add it, but it didn't work out. There was no change in the symptoms.

ruggb, 2019-11-08:
> > I already opened this bug on #15745
> 
> It's not clear they were the same issue.

How do we clear that up? Multiple people, different configs, same display, very similar symptoms == scrambled display. The only diff is they are not all scrambled the same way. Yeah, I guess that makes them diff issues. But for one, they all corrected with the same patch. And the patch was posted on a diff forum. Maybe you need to get that person on your team.

3Dangel, 2019-11-08:
Treetwo27
Have you set your Marlin to work on another language? Or it is “en”?
Also, there were different set ups as for the serial port. 
Try to modify the serial ports and see if that helps.

thisiskeithb, 2019-11-08:
> How do we clear that up? 

By including configs when opening an issue like the issue template asked 🙂

Otherwise, we’re left trying to guess which hardware config this fix applies to.

AnHardt, 2019-11-08:
This is clearly a timing problem.
If the already mentiond delays do not work even when doubled, tripled you can try to define

#define DOGM_SPI_DELAY_US 5 (Configuration_Adv.h)

with increasing values up to about 10.

Also improving the hardware may help. 
Shorter, better shielded cables.
Large capacitor between Vcc and GND on the display.
...

If you can see, what you see, your display is connected correctly. Maybe check at the display if Vcc is 5V.

ruggb, 2019-11-08:
> > How do we clear that up?
> 
> By including configs when opening an issue like the issue template asked 🙂
> 
> Otherwise, we’re left trying to guess which hardware config this fix applies to.

why was this confirmed as a BUG if the info was insufficient?

@AnHardt
ALSO, why is this now a hardware issue, when previous versions of Marlin worked fine? = FOR MANY PEOPLE.... I know, it is because you are a software person, right?

AnHardt, 2019-11-08:
Because it was always a hardware issue. The ST7920 is slow and picky about signal and power quality.

In theory at 5V a 16MHz AVR needs half a NOP delay in ST7920_DELAY_3 and nothing else.
In practice all of my various ST7920 based displays (7) do work completely without delays (On MEGA + RAMPS 1.4 + smart adapter + 200 mm standard flat cables.)

At the 20MHz AVRs one NOP is required in ST7920_DELAY_3 else you'll see this typical noise on shreen.
If i remember correctly next was adding more delays for MELZI boards. We never properly investigated why these boards (16MHz AVR) behave worse than Arduino MEGA with RAMPS. But we also could not find Melzis working without delays. 
Later we added more delays for other boards. 
For the faster 32bit processors the reason is clear. The additional increased delays for 3.3V systems are also obvious (see [ST7920 datasheet](http://www.hpinfotech.ro/ST7920.pdf)).

Others reported to see noise on long cables and fixing this with more delay.
Others reported to fix noise by shielding the cables and being able to reduce the delays again.
Others reported stabilizing Vcc by a capacitor helped to reduce the delays.  

My conclusion:
You can fight a wide range of noise (signal/power quality) issues with additional delays.
By improving the signal/power quality you can reduce the delays down to the values from the datasheet (and sometimes below of that).

3Dangel, 2019-11-08:
@AnHardt  thank you very much for explaining in here with such detail. 
I have benefit from your info!!!
Just so you know.

ruggb, 2019-11-09:
@AnHardt
Thanks - but what changed in the recent commits? Mine was working fine with one about 1-2 months old. No hardware change. I guess I will get out my new scope and take a look. It has been looking for something to do.

boelle, 2019-11-11:
so since this is a hardware issue and not marlin related maybe we should close this one?

ruggb, 2019-11-11:
Since things were working up to recent releases of Marlin, and adding delays to the s/w fixes things, I'm not sure I would dismiss this so quickly. I'd like to know why new releases show the problem.

swilkens, 2019-11-12:
> Since things were working up to recent releases of Marlin, and adding delays to the s/w fixes things, I'm not sure I would dismiss this so quickly. I'd like to know why new releases show the problem.

You can find out with a git bisect, stepping through the changes untill you find the offending commit

boelle, 2019-11-21:
@TreeTwo27 still no change with copy of 2.0.x from today?

sjasonsmith, 2019-11-24:
I finally pulled my display out of its box so I could try to reproduce this.

It is working fine for me, with both an SKR 1.3 and an Atmega2560/RAMPS.

boelle, 2019-11-24:
and its working fine for me on re-arm

@sjasonsmith you think we can close this one? or should we let it stay a few more days?

rflulling, 2019-11-29:
I observed an issue like this and generated a warning to others on facebook. the source of the issue was an update to Arduino IDE. No other changes to defines or updates effected the bug. But rolling back Arduino itself fixed the issue. 

However that I have been on SKR PRO since then, and there is a push to deny development of anything for Arduino. I don't expect to be back on Arduino IDE any time soon. 

My assumption is that those who compile on Arduino see this and those who compile on Visual Studio, do not.

boelle, 2019-12-02:
since we have not heard from @TreeTwo27 in 25 days i will close this one and assume he have solved his issue

closed 2019-12-02 by boelle