MarlinFirmware/Marlin #15834: [BUG] (MAXBEDTEMP HALTED after reflashing)

407s, 2019-11-07:
Hello. Today i take last version Marlin 2.x form github (Latest commit [47805e1])(https://github.com/MarlinFirmware/Marlin/commit/47805e18f85d7220332a84cfdbe7174cd44d9760)

Do config files for my printer: Ender-3, SKR 1.3, TMC2209
Immediately after booting i see MAXBEDTEMP, Printer halted.
If i using 
#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000

Printer starting okay, and temp show correctly. 23-24 celsium. Not show another values 50-60-70 how it was in previous bug reports.

Patag, 2019-11-07:
I confirm the bug... and the fix !
BTW, fix or workaround ?

p3p, 2019-11-07:
Updating the framework should install `0.2.1`, that hopefully fixes the issue you shouldnt need a grace period if you didn't previously, https://github.com/MarlinFirmware/Marlin/pull/15719 increases how many bits of the adc reading Marlin can use and in some cases its causing problems.

I'm still looking into if it causes anything else weird, if you have a heart beat led you will see it flashing a lot faster for instance, because the temperature ISR is running fully more often.

407s, 2019-11-07:
> 0.2.1

i Don't unsderstand what the framework?
installed
Vusial Studio 1.3.92
NXP Arduino LPC176x 0.1.0
TMCstepper 0.5.2

ManuelMcLure, 2019-11-07:
Just delete the .pio folder and rebuild. That will pull the new framework.

407s, 2019-11-07:
> Just delete the .pio folder and rebuild. That will pull the new framework.
Done. After this i start build.

`PACKAGES: framework-arduino-lpc176x 0.2.0, toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)`

p3p, 2019-11-07:
> NXP Arduino LPC176x 0.1.0

That is the platform that builds Marlin for LPC176x, it installs the appropriate Arduino framework, how to update platformio depends on how you are using it, 

`platformio update` from the terminal, clicking on the alien symbol in VSCode and clicking `Update All` in the quick access menu, deleting the platformio cache folders.

PlatformIO should keep things upto date itself but I'm not sure if it checks every build.

407s, 2019-11-07:
@p3p , i found your release
`0.1.0 Release released 4 days ago`
`framework-arduino-lpc176x 0.2.0 (installed)`

ManuelMcLure, 2019-11-07:
@p3p Do you need to make a 0.1.1 release that fetches 0.2.1? Or change platformio.ini to fetch from master?

p3p, 2019-11-07:
No the platform doesn't need a new release, It should be pulling down the updated framework as I've updated the remote(hosted on github) package manifest.

407s, 2019-11-07:
1. Deleted .pio
2. Deactivated BOGUS_TEMPERATURE_GRACE_PERIOD 
3. Compiled
4. Flashing
5. MAXTEMP: Bed PRINTER HALTED.

11 oct 2019 i taken source code and compiled for same board only with TMC2208. Error with max temp not was present.

p3p, 2019-11-07:
@407s what is the output from `platformio update`

ManuelMcLure, 2019-11-07:
I just downloaded https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.0.zip as specified in platformio.ini and the manifest has:
```
{
    "framework-arduino-lpc176x": [
        {
            "url": "https://github.com/p3p/pio-framework-arduino-lpc176x/archive/0.2.0.zip",
            "version": "0.2.0"
        }
    ]
}
```
so there's something wrong here. Maybe a Github problem?

ManuelMcLure, 2019-11-07:
Looks like platformio.ini should list
https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/release.zip
instead...

p3p, 2019-11-07:
The platform uses the live (online) release branch manifest file, not the downloaded version one.

No the platformio.ini is correct.

407s, 2019-11-07:
> @407s what is the output from `platformio update`

```
PS C:\Marlin\1.3-2209-BL\Marlin> platformio update
Updating tool-scons                      @ 2.20501.190826 [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.191017    [Up-to-date]
Updating contrib-piohome                 @ 2.3.3          [Up-to-date]

Platform Manager  
================  
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.15.0         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]
Updating framework-arduinoavr            @ 4.1.2          [Up-to-date]

Platform NXP Arduino LPC176x
--------
Updating nxplpc-arduino-lpc176x          @ 0.1.0          [Detached]
Updating toolchain-gccarmnoneeabi        @ 1.80201.190214 [Up-to-date]
Updating framework-arduino-lpc176x       @ 0.2.0          [0.2.1]
Uninstalling framework-arduino-lpc176x @ 0.2.0:         [OK]
PackageManager: Installing framework-arduino-lpc176x @ 0.2.1
Downloading  [####################################]  100%
framework-arduino-lpc176x @ 0.2.1 has been successfully installed!
Uninstalling framework-arduino-lpc176x @ 0.1.3:         [OK]

Platform NXP Arduino LPC176x
--------
Updating nxplpc-arduino-lpc176x          @ 0.0.2          [Detached]
Updating toolchain-gccarmnoneeabi        @ 1.80201.190214 [Up-to-date]
Updating framework-arduino-lpc176x       @ 0.2.1          [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 5.7.0          [Up-to-date]
Updating tool-stm32duino                 @ 1.0.2          [Up-to-date]
Updating framework-arduinoststm32-maple  @ 1.10000.190819 [Up-to-date]

Library Manager
===============
Library Storage: C:\Users\Yuriy\.platformio\lib
Updating TMCStepper                      @ 0.5.2          [Up-to-date]
```

p3p, 2019-11-07:
> Uninstalling framework-arduino-lpc176x @ 0.2.0:         [OK]
PackageManager: Installing framework-arduino-lpc176x @ 0.2.1
Downloading  [####################################]  100%

That's all you needed to do ^^

407s, 2019-11-07:
Oooooohhh yes. Problem fixed. 

Sry for my english. He stay on level "Edinburg the capital of Scotland".

closed 2019-11-07 by 407s

sbaeder, 2019-11-08:
Not really intuitive to find this information when trying to compile and use the 11/7 build...Even after removing the .pio folder, platformio re-used the 0.2.0 version of the framework.

Eventually, removing all the platformio extensions related to the LPC176x caused it to be proplerly updated...

J-charles-C, 2019-11-08:
I,

I have this problem and I did not understand how to solve it.

Imac-5K-de-Jean-Charles:Marlin-bugfix-08-11-19 Godjc$ platformio update
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating tool-pioplus                    @ 2.5.8          [Up-to-date]
Updating contrib-pysite                  @ 2.27.191017    [Up-to-date]
Updating contrib-piohome                 @ 3.0.0          [Up-to-date]
Updating tool-scons                      @ 2.20501.190826 [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.15.0         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]
Updating framework-arduinoavr            @ 4.1.2          [Up-to-date]

Platform NXP LPC
--------
Updating nxplpc                          @ 4.5.0          [Up-to-date]

Platform NXP Arduino LPC176x
--------
Updating nxplpc-arduino-lpc176x          @ 0.1.0          [Detached]
Updating toolchain-gccarmnoneeabi        @ 1.80201.181220 [Up-to-date]
Updating framework-arduino-lpc176x       @ 0.2.1          [Up-to-date]

Library Manager
===============
Library Storage: /Users/Godjc/.platformio/lib
Updating TMC2130Stepper                  @ 2.5.1          [Up-to-date]
Imac-5K-de-Jean-Charles:Marlin-bugfix-08-11-19 Godjc$ 

thanks

407s, 2019-11-08:
Del folder with name .pio and then

[image]

407s, 2019-11-08:
> find this information

i just do `platformio update`, and got `0.2.1`

J-charles-C, 2019-11-08:
I,

I have already update.

shitcreek, 2019-11-08:
@J-charles-C make sure you also deleted the .pio folder then relaunch vscode. Had this very issue and these steps resolved it.