MarlinFirmware/Marlin #15835: [BUG] Tool change breaking auto leveling (G29)

DreadPirateMiko, 2019-11-07:
### Description

When using a multi tool with singlenozzle (Y adapter/Diamond) Z value was changing after the 1st tool change (T1)

I noticed the Z height was changing from 0.24 (1st layer height) to 1.53 when changing to Tool 1 (during the wipe)

enabling debug for leveling displayed the following:

> Recv: >>> T(1)
> Recv:   current_position= X177.21 Y172.76 Z0.24 : BEFORE
> Recv:   current_position= X177.20 Y172.76 Z-1.29 : sync_plan_position
> Recv: Axis X min:-2.00 max:200.00
> Recv: Axis Y min:-12.00 max:200.00
> Recv: Axis Z min:0.00 max:200.00
> Recv: Offset Tool XY by { 0.00, 0.00, 0.00 }
> Recv:   current_position= X177.20 Y172.76 Z-1.29 : sync_plan_position
> Recv:   destination= X177.20 Y172.76 Z0.00 : Move back
> Recv: >>> do_blocking_move_to X177.20 Y172.76 Z0.00
> Recv: <<< do_blocking_move_to
> [...]
> Recv: echo:Active Extruder: 1
> Recv:   current_position= X177.18 Y172.76 Z1.53 : sync_plan_position
> Recv:   current_position= X177.18 Y172.76 Z1.53 : AFTER

seems that tool change is disabling leveling.

commenting out the line in tool_change.cpp solves the issue.

>     #if HAS_LEVELING
>       // Set current position to the physical position
>       //      TEMPORARY_BED_LEVELING_STATE(false);
>     #endif

boelle, 2019-11-08:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

DreadPirateMiko, 2019-11-08:
Sorry, attached now
pins files are also attached. my board is MKS_GEN_L with https://reprap.org/wiki/Stepper_Expander
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3823039/Marlin.zip)

shitcreek, 2019-11-13:
I suspect not having line above might cause the the print head to misalign and crash when switching tools for non single nozzle setups. Perhaps something like this instead:
```
#if HAS_LEVELING && !ENABLED(SINGLENOZZLE)
  // Set current position to the physical position
  TEMPORARY_BED_LEVELING_STATE(false);
#endif
```

Or perhaps reinstating it afterwards (line 959)
 ```
// Return to position and lower again
      if (safe_to_move && !no_move && IsRunning()) {

        #if HAS_LEVELING
          // Set current position to the leveled position
          TEMPORARY_BED_LEVELING_STATE(true);
        #endif
```

Would you mind testing this out for me? Thanks.

DreadPirateMiko, 2019-11-14:
> I suspect not having the line above might cause the the print head to misalign and crash when switching tools for non single nozzle setups. Perhaps something like this instead:
> 
> ```
> #if HAS_LEVELING && !ENABLED(SINGLENOZZLE)
>   // Set current position to the physical position
>   TEMPORARY_BED_LEVELING_STATE(false);
> #endif
> ```
> 
> Or perhaps reinstating it afterwards (line 959)
> 
> ```
> // Return to position and lower again
>      if (safe_to_move && !no_move && IsRunning()) {
> 
>        #if HAS_LEVELING
>          // Set current position to the leveled position
>          TEMPORARY_BED_LEVELING_STATE(true);
>        #endif
> ```
> 
> Would you mind testing this out for me? Thanks.

1st solution is working
see debug output when changing the tool:
```
Recv: >>> T(1)
Recv:   current_position= X184.16 Y170.54 Z0.24 : BEFORE
Recv: Axis X min:-2.00 max:200.00
Recv: Axis Y min:-12.00 max:200.00
Recv: Axis Z min:0.00 max:200.00
Recv: Offset Tool XY by { 0.00, 0.00, 0.00 }
Recv:   current_position= X184.16 Y170.54 Z0.24 : sync_plan_position
Recv:   destination= X184.16 Y170.54 Z0.24 : Move back
Recv: >>> do_blocking_move_to X184.16 Y170.54 Z0.24
Recv: <<< do_blocking_move_to
Recv: echo:Active Extruder: 1
Recv:   current_position= X184.16 Y170.54 Z0.24 : AFTER
Recv: <<< T()
```
2nd solution is not working
see debug output when changing the tool:
```
Recv: >>> T(1)
Recv:   current_position= X184.16 Y170.54 Z0.24 : BEFORE
Recv:   current_position= X184.15 Y170.54 Z-1.36 : sync_plan_position
Recv: Axis X min:-2.00 max:200.00
Recv: Axis Y min:-12.00 max:200.00
Recv: Axis Z min:0.00 max:200.00
Recv: Offset Tool XY by { 0.00, 0.00, 0.00 }
Recv:   current_position= X184.15 Y170.54 Z-1.36 : sync_plan_position
Recv:   current_position= X184.16 Y170.54 Z0.24 : sync_plan_position
Recv:   destination= X184.15 Y170.54 Z0.00 : Move back
Recv: >>> do_blocking_move_to X184.15 Y170.54 Z0.00
Recv: <<< do_blocking_move_to
Recv:   current_position= X184.13 Y170.54 Z-1.59 : sync_plan_position
Recv: echo:Active Extruder: 1
Recv:   current_position= X184.15 Y170.54 Z-0.00 : sync_plan_position
Recv:   current_position= X184.15 Y170.54 Z-0.00 : AFTER
Recv: <<< T()
```
after tool change height is now 0 (instead of 0.24)

InsanityAutomation, 2019-11-14:
Fyi, this is the cause of the open pr that's been sitting rolling back the scope dependent methods for explicit calls.

shitcreek, 2019-11-14:
@InsanityAutomation should we then hold off on creating a new pr for this?

InsanityAutomation, 2019-11-14:
@shitcreek all the open pr does is roll it back to explicit remove / reapply leveling so it shows it's not dropping out of scope correctly. The problem reported affects non single nozzle as well, though it is the only variable on single nozzle. Problematic tool offsets may be blamed otherwise.

That said, once the root cause was found the exemption for single nozzle is something I've done in my branches and planned to send here afterwards.

Kurairaito, 2019-11-24:
I have a similar issue with a similar configuration (except that z goes down in my case).

Each time I change tool (with T0 ou T1) the z raise and lower and come back at it initial state. but on the next movement z lower again, so it crash on the print.

for exemple, doing

```
G28;
T1;
G0 F1500 X125
T0;
G0 F1500 X125
T1;
G0 F1500 X125
T0;
G0 F1500 X125
T1;
G0 F1500 X125
T0;
G0 F1500 X125
[....]
```

Leads to a crash of the nozzle in the bed after enough iteration. (it's an interpolation of me seeing the coupler nut not in the exact same position as before and seing it going slowly to a crash) Effectively, when doing multicolor parts, the nozzle crash in the printed part.

I tried @shitcreek fix without success and also tried PR https://github.com/MarlinFirmware/Marlin/pull/15203 without success (I had to adapt a header file for it to apply)

depending on the fiw I tried, comportment of the command changed, but result was always the same.

InsanityAutomation, 2019-11-25:
@Kurairaito I rebased the PR above so its current and should function better. Are you using UBL or bilinear? if UBL, is youre park point outside the mesh?

Kurairaito, 2019-11-25:
I'm using UBL and yes my park point is outside the mesh

InsanityAutomation, 2019-11-25:
@Kurairaito then that is a separate known issue. UBL does not apply correctly crossing mesh boundary. Extend the mesh size to the soft limits instead of the bed size, then let G29P3 fill in the points it cant reach.

Kurairaito, 2019-11-25:
But the issue does appear also if I change tool inside the mesh zone (at the center of the plate) for example)

Kurairaito, 2019-11-25:
Your new PR solved my issue (the previous didn't), it now works at the middle of the plate AND also when I change tool after a G27, which is outside the mesh, and I didn't extend it.

InsanityAutomation, 2019-11-25:
Sounds good. Are you on a single nozzle setup? That avoids disabling and reapplying leveling altogether.

Kurairaito, 2019-11-25:
Yes I'm on singlenozzle, but I wrote to fast, it is still broke if I do a G29 J before.
Yet it is not exactly the same result, Doing T0;T1;T0;T1;T0 etc make the nozzle go down even on the middle on the bed.

boelle, 2019-12-01:
@DreadPirateMiko @InsanityAutomation what can we do to clear this one if any thing can be done?

DreadPirateMiko, 2019-12-01:
1st suggestion (changing "if HAS_LEVELING" ) is solving the issue.

boelle, 2019-12-01:
oki so closing is ok?

DreadPirateMiko, 2019-12-02:
Can confirm. 
Will this be merge to master?

boelle, 2019-12-02:
at some point yes, i guess the PR from @InsanityAutomation needs a tiny bit more polishing

boelle, 2019-12-10:
@DreadPirateMiko the PR  from @InsanityAutomation will get merged when @thinkyhead does the merge

boelle, 2019-12-24:
@thinkyhead there is a PR from @InsanityAutomation that fixes this one, please consider a merge even thou it will revert some changes you did

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle