MarlinFirmware/Marlin #15838: [BUG] x-y motor clicking during diangonal movement

emaayan, 2019-11-07:
### Description
[Marlin-bug-fix-1-1-9.zip](https://github.com/MarlinFirmware/Marlin/files/3822111/Marlin-bug-fix-1-1-9.zip)

during rapid diagonal motions both x-y and motors begin a faint sound of clicking, this only happens when bed leveling is enabled 

### Steps to Reproduce

[level.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/3822116/level.gcode.txt)
[Marlin-bugfix-1-1-9.zip](https://github.com/MarlinFirmware/Marlin/files/3822119/Marlin-bugfix-1-1-9.zip)

1. flash the firmware 
2. use the enclosed gcode file (bed leveling must be done) 

**Expected behavior:**
smooth motion with no clicking 

**Actual behavior:**
random faint clicking during motion. 

#### Additional Information

https://www.youtube.com/watch?v=X4dnCctrQUA
you can hear faint clicking during the motion, this is NOT a belt slip, 
i've tried replacing everything, mks gen 1.4 , drivers, motors, flashed different firmware version, even tried using sdcard. still clicking. 
here's another vid of just the motor (although a different type, but same clicking) 
the clicking will only happen when BOTH x-y are moving and bed leveling is enabled 
https://www.youtube.com/watch?v=8EyYAeQp6aY

using MKS GEN 1.4 with TMC2100 vref at 1.5

boelle, 2019-11-08:
tried with bugfix 2.0.x?

emaayan, 2019-11-08:
no, because i have no knowledge of it, or the parameters needing to be changed (or converted from 1.1) , the build environment needed for it, or how stable it is..

emaayan, 2019-11-08:
> tried with bugfix 2.0.x?

just tried it. still same issue

blueeagle69, 2019-11-08:
> > tried with bugfix 2.0.x?
> 
> just tried it. still same issue

Did you clear the EEPROM after flashing 2.0x?

emaayan, 2019-11-09:
> > > tried with bugfix 2.0.x?
> > 
> > 
> > just tried it. still same issue
> 
> Did you clear the EEPROM after flashing 2.0x?
M502 and M500 ? yes

emaayan, 2019-11-11:
> > > tried with bugfix 2.0.x?
> > 
> > 
> > just tried it. still same issue
> 
> Did you clear the EEPROM after flashing 2.0x?

just a small update, i tried replacing x driver with 2208, i'm guessing the defintions aren't correct, because i now hear clicking when i just move the x-axis, regadlesss of printing or other motors.
BUT
the interesting thing, is that i hear the SAME clicking, when i move y or z axis as well, even though they are still with TMC2100 driver (and i didn't change the definition about them in the firmware)

boelle, 2019-11-20:
and you are using 24V for the TMC's right?

emaayan, 2019-11-20:
> and you are using 24V for the TMC's right?

yes

Hubertusfan, 2019-11-22:
Same Issue. I use Ramps 1.4 with Marlin 2.0, 12V. Also Toc Toc Sound on X and Y if movement is longer than ca 4 cm. No Toc Sound on Short movement.

emaayan, 2019-11-23:
> Same Issue. I use Ramps 1.4 with Marlin 2.0, 12V. Also Toc Toc Sound on X and Y if movement is longer than ca 4 cm. No Toc Sound on Short movement.

that's interesting, btw in marlin 2.0 i've experienced cases where there's a toc toc sound even when you travel https://www.reddit.com/r/3Dprinting/comments/ce9hkx/clicking_motor_noises_after_marlin_20_update/
like so
(it's not the same ticks as i have)

boelle, 2019-11-24:
@emaayan  have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

Hubertusfan, 2019-11-24:
> that's interesting, btw in marlin 2.0 i've experienced cases where there's a toc toc sound even when you travel [...]

But it's exactly the same sound as i have on X, Y and Z.

boelle, 2019-11-24:
@Hubertusfan and just curious, are you using TMC at 12V ? active cooling?

emaayan, 2019-11-24:
> @emaayan have you tried to ask for help any of these places?
>
> RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
> [Tom's 3D Forums](https://discuss.toms3d.org/)
> Facebook Group ["Marlin [...]

i tried asking in reddit, not much help, as well as in black widow facebook group 
my request to join marlin for 3d printers is pending for like weeks, don't know why 

i'm trying to acheive a state with the least bit of adapation is needed, i litreally have a bare board with mechanical swiches . i dont  even use real bed leveling data, just the debug one.

i suspect the issue occurs with very rapid movement (like when starting an infill from a corner of a rectangle) , and just lingers on . the clicks also don't happen randomly i'm fairly certain they are always at the same time. even disconnecting the z motor mid print would have those ticks happen. 
the fact that with recent marlin 2.0 bug-fix you major clicks in motor during travel also lead me to suspect it's marlin related

boelle, 2019-11-24:
you dont have to wait if using discord

Hubertusfan, 2019-11-24:
As statet before, I use TMC 2209 v2 with 12 Volt and, yes, active cooling. The Toc Toc sound starts directly, not after some time. Hard to believe a thermal problem.

Hubertusfan, 2019-11-24:
> i suspect the issue occurs with very rapid movement (like when starting an infill from a corner of a rectangle) , and just lingers on . the clicks also don't happen randomly i'm fairly certain they [...]

Yes, the Toc sound occurs in a constant frequent, may be ca every 5 cm movement. I've tested a lot of various changes on velocity and accelereration adjustements, no changes of the sound.

emaayan, 2019-11-24:
> Yes, the Toc sound occurs in a constant frequent, may be ca every 5 cm movement. I've tested a lot of various changes on velocity and accelereration adjustements, no changes of the sound.

are you talking about this ?https://www.reddit.com/r/3Dprinting/comments/ce9hkx/clicking_motor_noises_after_marlin_20_update/ 
cause that goes away if you comment MONITOR_DRIVER_STATUS

AnHardt, 2019-11-24:
> the Toc sound occurs in a constant frequent, may be ca every 5 cm movement.

Constant frequency or constant way? Can you describe the frequency in time? (Like - every second, or every half second, ...)
If indeed depending on way - at what speed?

Hubertusfan, 2019-11-24:
> > the Toc sound occurs in a constant frequent, may be ca every 5 cm movement.
> 
> Constant frequency or constant way? Can you describe the frequency in time? (Like - every second, or every half second, ...)
> If indeed depending on way - at what speed?

It seems to be time dependent. Low speed 5 Toc per 100 mm, higher speed only 3 Toc per 100 mm.
Low speed gentle Toc sound, with higher speed louder.
Estimated a Toc every 0.5 sec.

Hubertusfan, 2019-11-24:
> 
> are you talking about this ?https://www.reddit.com/r/3Dprinting/comments/ce9hkx/clicking_motor_noises_after_marlin_20_update/
> cause that goes away if you comment MONITOR_DRIVER_STATUS

Great! I've deactivated MONITOR_DRIVER_STATUS and the Toc sound is away! 

Thanks emaayan and AnHardt

emaayan, 2019-11-24:
> > are you talking about this ?https://www.reddit.com/r/3Dprinting/comments/ce9hkx/clicking_motor_noises_after_marlin_20_update/
> > cause that goes away if you comment MONITOR_DRIVER_STATUS
> 
> Great! I've deactivated MONITOR_DRIVER_STATUS and the Toc sound is away!
> 
> Thanks emaayan and AnHardt

great now i'm left with my messed up problem :(

boelle, 2019-12-01:
@emaayan still a messed up problem?

emaayan, 2019-12-01:
> @emaayan still a messed up problem?

definitely.. 
https://www.facebook.com/groups/makerbot/permalink/2535589233225232/
i've reverted back to 1.1.9 stable (not the bug-fix) because 2.0 is currently unusable due to the OTHER clicking sound originating from the MONITOR_DRIVERS setting 
 
i'm not sure if it's also effecting bed leveling first layer extrusion, but i'm almost sure those clicks always come in exactly the same time (meaning if i restart the print they'll sound off in the same time they sounded earlier)

boelle, 2019-12-07:
@emaayan since 2.0 was just released a few days ago has this changed this issue at all?

emaayan, 2019-12-07:
> @emaayan since 2.0 was just released a few days ago has this changed this issue at all?

i don't think i'll be able to use 2.0 at all, due to the other clicking sounds that are coming from MONITOR_DRIVER_STATUS bug

boelle, 2019-12-07:
i have not checked but it could have been fixed, and there will be a version 2.0.1 before end of weekend

emaayan, 2019-12-07:
i understood it's very problematic to resolve because the way the motors
work, but i'll try again when 2.0.1 comes out.  actually i can't even
compile marlin 2 because of these errors
(it didn't use to be that way)

sketch\src\feature\bltouch.cpp: In static member function 'static void
BLTouch::init(bool)':

sketch\src\feature\bltouch.cpp:67:7: error: 'DEBUG_ECHOLNPGM' was not
declared in this scope

       DEBUG_ECHOLNPGM("config mode - "

       ^~~~~~~~~~~~~~~

ManuelMcLure, 2019-12-07:
Just turn ``MONITOR_DRIVER_STATUS`` off. You don't need it except for debugging. It should not be turned on in a normal build.

emaayan, 2019-12-07:
i thought the main strength point is to change status on runtime?

ManuelMcLure, 2019-12-07:
You don't need ``MONITOR_DRIVER_STATUS`` enabled to be able to change settings or query settings at runtime. It just turns off the continuous polling of the status which is what is causing the pauses/clicking.
Without ``MONITOR_DRIVER_STATUS`` you can still query the driver status with ``M122`` as long as you have ``TMC_DEBUG`` enabled.

emaayan, 2019-12-07:
sadli i'm still getting the compile error though

sjasonsmith, 2019-12-07:
@emaayan you probably have to use the bug fix-2.0.x branch. I think there is supposed to be a 2.0.1 update this weekend which would pull those into a release branch.

boelle, 2019-12-11:
will close this one, 1.1.x is not worked on active anymore and hence the error will prob stay

use bugfix 2.0.x and if the error is there then open a new issue

closed 2019-12-11 by boelle

emaayan, 2019-12-17:
ok, tried with marlin 2.0 bug-fix, even went with platformio
here you go: https://github.com/MarlinFirmware/Marlin/issues/16248