MarlinFirmware/Marlin #15839: G29 P1 Probes one point in the middle of the bed, no other points probed

danandmary, 2019-11-08:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3822623/Marlin.zip)

Problem : G29 P1 only probes and records one point (in the middle of the bed) and does not continue with the rest of the mesh.

Expected : G29 P1 probes a 3x3 grid

Output of G29 P1 T

12:05:52.716 : N26 G29T0 P1*3
12:05:52.731 : Mesh invalidated. Probing mesh.
12:05:54.195 : Bed Topography Report:
12:05:54.207 : (  1,533)      (929,533)
12:05:54.220 : 0       1       2
12:05:54.236 : 2 |   .       .       .
12:05:54.236 : |
12:05:54.257 : 1 |   .       .       .
12:05:54.257 : |
12:05:54.272 : 0 |[  .   ]   .       .
12:05:54.275 : 0       1       2
12:05:54.290 : (  1,  1)      (929,  1)
12:06:02.380 : Bed Topography Report:
12:06:02.396 : (  1,533)      (929,533)
12:06:02.411 : 0       1       2
12:06:02.427 : 2 |   .       .       .
12:06:02.427 : |
12:06:02.434 : 1 |   .    [-0.788]   .
12:06:02.434 : |
12:06:02.449 : 0 |   .       .       .
12:06:02.465 : 0       1       2
12:06:02.480 : (  1,  1)      (929,  1)
12:06:09.450 : X:20.00 Y:30.00 Z:10.00 E:0.00 Count X:1600 Y:2400 Z:2400
12:06:09.457 : Bed Topography Report:
12:06:09.473 : (  1,533)      (929,533)
12:06:09.473 : 0       1       2
12:06:09.488 : 2 |   .       .       .
12:06:09.488 : |
12:06:09.504 : 1 |   .     -0.788    .
12:06:09.519 : |
12:06:09.535 : 0 |[  .   ]   .       .
12:06:09.551 : 0       1       2
12:06:09.557 : (  1,  1)      (929,  1)

Have run M502, M500, M501 the G29 P1

Can't get this to work at all.

RudyBenner, 2019-11-08:
[image]

danandmary, 2019-11-08:
Isnt this essentially manually leveling the bed?  Should it not automatically probe the various mesh points?

danandmary, 2019-11-08:
This is the output of G29 S0, S1 and S2
`13:15:20.387 : N15 G29 S0*101
13:15:20.403 : Done.
13:15:31.311 : N16 G29 S1*103
13:15:31.326 : Done.
13:16:49.532 : N17 G29 S2*101
13:16:49.654 : Done.
`
At no point during the above commands am I given the option to adjust height.  LCD makes clicking sound and nothing happens with each step, no movement of X Y axis at all.

danandmary, 2019-11-08:
I should point out that manual probing appears to work as expected, empty points are moved to, with option on LCD to adjust Z height.  Clicking encoder stores the settings and moves gantry to next position.  My issue is that **_automated probing_** only probes the center of the bed, then appears to exit the routine.  No error messages are displayed when this occurs.

Roxy-3D, 2019-11-08:
This problem was seen 3 weeks ago, and fixed 2 weeks ago:  

https://github.com/MarlinFirmware/Marlin/commit/3642a12fc170ffdee8e5db5d78684f82914edc41

It is very strange that it showed up again...

hackebike, 2019-11-08:
I have this issue with U-ABL, Marlin 2.0.x from Nov 5th. I can make a video tomorrow and post configs.

danandmary, 2019-11-08:
@Roxy-3D I'll give it a go and report back in a few days.  I'm assuming this means absolutely no more development occurring with 1.1.9?  I did read in 1.1.9 was finished, just wasn't sure if that meant no more patches too.  Thanks for your help.

Roxy-3D, 2019-11-08:
Something like this can be patched.   There is no more new development on v1.1.9.    But the whole purpose of a BugFix branch is to fix bugs.    So if somebody steps up with a fix, we can put into the branch.

boelle, 2019-11-11:
@hackebike config and video might be a good start
maybe @danandmary can confirm if what happens in video is the same

shitcreek, 2019-11-11:
SInce there is no new development on v1.1.9 and this is not an issue on v2.0 (no problem testing G29 P1 with toady's build), I will say this is not a bug. If you feel inclined, please create a PR, but otherwise I'd recommend upgrading to v2.0.

closed 2019-11-11 by shitcreek