MarlinFirmware/Marlin #15840: [BUG] TMC CONNECTION ERROR

kramer04, 2019-11-08:
DESCRIPTION

ENVOYÉS : M122
LUS : Testing X connection... Error: All LOW
LUS : Testing Y connection... OK
LUS : Testing Z connection... OK
LUS : Testing E connection... OK
LUS : ok

Ender 3 + BTT SKR MINI E3 + HOTEND E3D V6

**Expected behavior:** 

moving X axis

**Actual behavior:

X axis doesn't move

[marlin_bug_fix_021119.zip](https://github.com/MarlinFirmware/Marlin/files/3823226/marlin_bug_fix_021119.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2019-11-08:
Which version of the SKR E3 Mini you have?

On the board it will say either "BIGTREETECH SKR-mini-E3", or it will say "BIGTREETECH SKR-mini-E3 1.2".

kramer04, 2019-11-08:
on my board it's written BIGTREETECH SKR-mini-E3
i'm sure that's not 1.2 because i bought it before they launch the v1.2

sjasonsmith, 2019-11-08:
You may have a damaged driver on your board. Usually your axis will move even with the TMC error, but may move at the wrong speed.

I have the same board, and the connection works to all drivers with your configuration.

kramer04, 2019-11-08:
So i'll try to put my old motherboard

boelle, 2019-11-08:
please close this one if it turns out to be hardware issues

closed 2019-11-08 by kramer04

zmp2000, 2019-11-27:
Hi, I have the same issue but mine works, and with old Firmware (1.9) it works fine without errors
I've already checked the pins file (pin_ramps.h) and the pins are the same.
I'm currently using tmc2130 for the X and Y axis, and only the Y axis is throwing the error but still works