MarlinFirmware/Marlin #15842: [BUG] (short description)

hussainsail2002, 2019-11-08:
### Description

I am trying to compile marlin 2.0 for MKS Sbase 32 bit controller which uses LPC1768 
which has a mks mini 12864 LCD display along with the  SD card reader.On compiling I get the following error:

_In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:42:_
Marlin\src\lcd\dogm\ultralcd_DOGM.h:195:23: error: 'DOGLCD_CS' was not declared in this scope
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                       ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:195:23: note: suggested alternative: 'DOGLCD_SCK'
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                       ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:195:34: error: 'DOGLCD_A0' was not declared in this scope
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:195:34: note: suggested alternative: 'DOGLCD_SCK'
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~_
*** [.pio\build\LPC1768\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1_

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. Download marlin 2.0 from marlinfw.org
2. Setup platform.io in Vs code
3. import the marlin folder to Platform.io
4. Make changes in the configuration.h and configuration.adv files
5. In the platform.ini file change the default_envs to LPC1678
6. In configuration.h change board to BOARD_MKS_SBASE
7. Other changes are normal as with arduino IDE.
8. Uncomment your required board ( mks mini 12864)
9. Compile the program.

**Expected behavior:**Expected the comiplation to complete

**Actual behavior:** Throws and error, this error points to a line of code in the ultralcd_DOGM.h file
which says the following:

_// Use HW-SPI if no other option is specified
#ifndef U8G_PARAM
  #if ENABLED(FORCE_SOFT_SPI)
    #define U8G_PARAM DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0 // SW-SPI
  #else
    #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
  #endif
#15686 

#### Additional Information

From my understanding, the LCD is creating some problem.

hussainsail2002, 2019-11-08:
From my understanding, I dont think the MKS Sbase supports the mks mini panel.

I looked through the pins.h file for this board and only found the fysetc mini 12864 was defined. I am not sure if I am right though.

thisiskeithb, 2019-11-08:
The SBase config doesn’t have support for the MKS mini LCD currently. Someone will need to verify then pins and submit a PR to get it working.

boelle, 2019-11-08:
So this could be renamed to be a FR instead?

thisiskeithb, 2019-11-08:
Yes. Configs won’t help here.

boelle, 2019-11-08:
configs are just a standard i hit when i see nothing attached
there will at some point be a robot that will do the same thing

so if @hussainsail2002 agree we can rename this a FR and clean it up a bit :-D

hussainsail2002, 2019-11-08:
@thisiskeithb I also tried connecting to pronterface and the board would just not connect. 

as mentioned by @boelle I will attach the config files

Thanks

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3824326/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3824327/Configuration_adv.zip)

hussainsail2002, 2019-11-08:
@thisiskeithb 
How hard / easy would it be to configure the pins ?
 just wondering if I could do them myself.

boelle, 2019-11-08:
remember that if you end up with working pins give us a copy so that we have a chance of feeding it back in to marlin 2.0.x

i can do a PR, i just need to know precise what file to add or change

Evg33, 2019-11-08:
https://github.com/MarlinFirmware/Marlin/pull/15836

marvin-sinister, 2019-11-08:
It seems to me the pins on EXP1 of MKS SBASE don't have pins connected for MKS MINI12864 (NC on board schematics), you might need to create custom cable. I think it's because MKS MINI12864 has mirrored connectors.

Adding:
```
      #elif ENABLED(MKS_MINI_12864)
        #define DOGLCD_CS    P1_21
        #define DOGLCD_A0    P1_22
        #define DOGLCD_SCK   P0_15
        #define DOGLCD_MOSI  P0_18
        #define FORCE_SOFT_SPI
```
to `Marlin/src/pins/lpc1768/pins_MKS_SBASE.h` let's me compile the firmware (the pins are bogus for that board). Also, FYSTEC doesn't have "FORCE_SOFT_SPI" on MKS SBASE, so i don't know if you need it for MKS DISPLAY.

qyh1991qyh, 2019-11-09:
You can refer to:
https://github.com/makerbase-mks/MKS-SBASE/issues/101

thisiskeithb, 2019-11-09:
> You can refer to:
> [makerbase-mks/MKS-SBASE#101](https://github.com/makerbase-mks/MKS-SBASE/issues/101)

Since you have to physically modify the motherboard, does it make sense to add native Marlin support for this controller?

marvin-sinister, 2019-11-09:
Maybe put an `#error` message explaining? Perhaps put that link, and the pin definitions after that.

3Dangel, 2019-11-09:
Why I have a feeling that the issue is more linked to settings and not the hardware as Makerbase has done with the main board......
I have the Sbase and loaded with MArlin 2 bug fix and working Reprap full discount grapic LCD and the SD card on the LCD pannel works as well......

Sorry that I'm not contributing anything here, but just sharing my thinking.....

thisiskeithb, 2019-11-09:
> Why I have a feeling that the issue is more linked to settings and not the hardware as Makerbase has done with the main board......

This is the issue:

[image]

hussainsail2002, 2019-11-10:
Is there a way to fix this at the firmware ?
without having to do too much hardware changes to the MKS Sbase. 

Will probably try the soldering method for now. Thanks  @qyh1991qyh for the guide

swilkens, 2019-11-10:
> Is there a way to fix this at the firmware ?
> without having to do too much hardware changes to the MKS Sbase.
> 
> Will probably try the soldering method for now. Thanks @qyh1991qyh for the guide

This can not be fixed at the firmware level, it requires physical modifications to the board.

hussainsail2002, 2019-11-11:
I tried the soldering method that was recommended along with the changes in the firmware.

When I connect board to the MKS mini 12864, the LCD turns on for few seconds and then blanks out

Not sure what the issue is though.

hussainsail2002, 2019-11-11:
I have managed to get it working,

The mistake I made was I activated Minipanel in the lcd section instead of mks_mini_12864.

The reason why I prefer the mks mini 12864 is because of the front loading LCD

It now works after soldering and adding few lines of code as mentioned in the link above.
We could maybe add those lines to the main marlin, also the read me could be updated for future users

Thanks for the help !

Hussain

closed 2019-11-11 by hussainsail2002

magpern, 2020-06-07:
@hussainsail2002 Could you specify closer what you did?
I also have a MKS SBASE V1.3 and a MKD MINI12864 V2.0
trying to get Marlin 2.0 (or bug fix) to show on the display.

I did the solder thing. Soler two pins together, and the other two pins together.
This https://github.com/makerbase-mks/MKS-SBASE/issues/101

but LCD is just lit up. Nothing is on it