MarlinFirmware/Marlin #15851: [BUG] G29 Probes off bed with BLTOUCH on Ender 3

OughRay, 2019-11-09:
### Description

When I send G29 to my printer the probe tries to probe off to the left of the bed. I cannot find any X_PROBE_OFFSET_FROM_EXTRUDER options in marlin, so there is no way that Marlin knows where my BLTOUCH is.

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [Power printer on]
2. [Send G28 then G29]
3. [Printer goes to X 0 Y 0 and goes down until it rams into the bed]

**Expected behavior:** [BLTOUCH Probe is on the bed to probe when I send G28 + G29]

**Actual behavior:** [Bed goes to X0 Y0 and crashed down with the probe to the left side of the bed.]

#### Additional Information

Probably I am just missing something?
I have Z_Safe_Homing enabled.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3827251/Marlin.zip)

thisiskeithb, 2019-11-09:
> I cannot find any `X_PROBE_OFFSET_FROM_EXTRUDER` options in marlin

You’ll use `NOZZLE_TO_PROBE_OFFSET` to define that now.

You should also enable `Z_SAFE_HOMING` which will home Z in the center of the bed to ensure your probe has a surface to touch.

OughRay, 2019-11-09:
> > I cannot find any `X_PROBE_OFFSET_FROM_EXTRUDER` options in marlin
> 
> You’ll use `NOZZLE_TO_PROBE_OFFSET` to define that now.
> 
> > Bed goes to X0 Y0 and crashed down with the probe to the left side of the bed.
> 
> Enable `Z_SAFE_HOMING` to home Z in the center of the bed.

So I defined the offsets `#define NOZZLE_TO_PROBE_OFFSET { -44, -7, -3 }` like this, but when i G28 and then G29 the BLTOUCH is still off the bed, and the Probe still only checks on the front left and then does nothing.

OughRay, 2019-11-09:
So I was able to find the settings to change everything, but now I send g28 and it homes, then i set the x offset to -1.5 and save it with m500. after i start a print it goes and calibrates with g29 but when it starts printing it doesnt use the offset of Z-1.5 I provided.

What could be the problem here?

407s, 2019-11-10:
> here?
Problem in situation when you not read manual.
[image]

OughRay, 2019-11-10:
> > here?
> > Problem in situation when you not read manual.
> > [image]

I did read the manual, I found these settings and adjusted them accordingly. But my grid where the BLTOUCH probes is off center. It probes more to the left side of the bed than to the right.

tonymills, 2019-11-11:
did anyone get anywhere with this? 

ive got a ramps 1.4 and a bltouch 3.1 on an overhauled wanhao i3 2.1

according to bltouch i need to use marlin 2, and their documentation says to do the following

//============================== Endstop Settings ===========================
#define USE_ZMIN_PLUG // a Z probe
#define ENDSTOPPULLUPS
#define ENDSTOP_INTERRUPTS_FEATURE

//============================= Z Probe Options =============================
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define BLTOUCH
#if ENABLED(BLTOUCH)
//#define BLTOUCH_DELAY 500 // *option
//#define BLTOUCH_FORCE_5V_MODE
#endif
#define PROBING_HEATERS_OFF // *option
#define PROBING_FANS_OFF // *option
#define X_PROBE_OFFSET_FROM_EXTRUDER 0 //Depend on your BLTouch installation value
#define Y_PROBE_OFFSET_FROM_EXTRUDER -22 //Depend on your BLTouch installation value
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.35 //Depend on your BLTouch installation value
//It must be greater than or equal to the higher of the X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER.
#define MIN_PROBE_EDGE 22
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z / 5
#define Z_CLEARANCE_DEPLOY_PROBE 15 // set up at least 10
#define Z_CLEARANCE_BETWEEN_PROBES 5 // set up at least 5
//=============================== Bed Leveling ==============================
#define AUTO_BED_LEVELING_BILINEAR // *option

so you do that and it wont compile as we need to use NOZZLE_TO_PROBE_OFFSET

that doesnt make any difference at all and the nozzle hits the bed

this was mentioned in #15725 but that didnt get an answer either. 

so im a bit stuck with a bltouch 3.1 and no marlin that appears to work

Any help greatly appreciated 

Thanks

tonymills, 2019-11-11:
ok ftr, in case someone else google's it the documentation from BLtouch is out of date and the 3.1 version need marlin 2 (so i read)

The following settings are deprecated 

X_PROBE_OFFSET_FROM_EXTRUDER
Y_PROBE_OFFSET_FROM_EXTRUDER
Z_PROBE_OFFSET_FROM_EXTRUDER

and you need to set NOZZLE_TO_PROBE_OFFSET instead

in my case

#define NOZZLE_TO_PROBE_OFFSET { -25, -45, 0 }

then you need to enable Z_SAFE_HOMING 

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

the x / 2 and the y / 2 sets where the middle of the bed is so its important that X_BED_SIZE and Y_BED_SIZE are correct, in my case 200 and 200

Now working like a charm

yudi-jpn, 2019-11-12:
Good morning.
In my environment, the probe moves down after the head moves to the X 0, Y 0 position.

If you touch the BL-toutch with your hand, it will be leveled at the next position.

However, the defined positional relationship is not effective, and the probe descends outside the bed, so the head hits the head and the stepping motor slips.

AmiSMB, 2019-11-18:
> Good morning.
> In my environment, the probe moves down after the head moves to the X 0, Y 0 position.
>
> If you touch the BL-toutch with your hand, it will be leveled at the next position.
>
> However, [...]

As is states above enable Z_SAFE_HOMING and then the homing of x.y and z will be done in the centre of the bed on a G28. Also make sure that the NOZZLE_TO_PROBE_OFFSET is set correctly otherwise the probe will not be in the correct position. You could also try increasing MIN_PROBE_EDGE to make sure that the probe is inside the bed.

yudi-jpn, 2019-11-20:
AmiSMB Thank you.
Sorry for the late reply.

Now, I am commenting out the definition below and defining it.
`#define NOZZLE_TO_PROBE_OFFSET {-43, -5, 0}`
`#define Z_SAFE_HOMING`

When the auto home operation is performed, the heat nozzle goes to the center.

But when doing ABL,
Even if the values of the XY axes are assigned in reverse, the probe descends at the X0 and Y0 positions.
`#define NOZZLE_TO_PROBE_OFFSET {43, 5, 0}`

Because the definition is not reflected, the cause was investigated.
I feel that it is caused by other unknown definitions so far.

If it was a simple mistake, it would be nice.
Thank you carefully.

damvcoool, 2019-11-20:
@OughRay 
How did you wire your BLTouch? are you using the Z_Endstop or are you using the Probe_Specific pin? would also be good to include your PIN config file from Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h.

If you are not using the Z_Endstop i would suggest to disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

I believe you already fixed MIN_PROBE_EDGE and NOZZLE_TO_PROBE_OFFSET.

I know I do not have the same board, but take a look at my configuration, and compare it may help you find an answer (if you do find the answer please share it here)
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

AmiSMB, 2019-11-20:
> AmiSMB Thank you.
> Sorry for the late reply.
>
> Now, I am commenting out the definition below and defining it.
> `#define NOZZLE_TO_PROBE_OFFSET {-43, -5, 0}`
> `#define Z_SAFE_HOMING`
>
> When the auto home [...]

I have a Hero Me on the Ender-3 and I used the offsets given in the design which are:-

#define NOZZLE_TO_PROBE_OFFSET { -42, -16, 0 }

Which are working fine with a BLTOUCH. I then use the Z offset tuning parameter during a print to get the extruder in the best place possible which currently is -1.925.

Without knowing what holder you are using for the BLTOUCH or being in front of your actual machine and measuring with digital callipers where your BLTOUCH is in relation to the nozzle would be it would be hard to know what the correct offsets are for your setup.

Here are my configuration files which are working and maybe you can just change the NOZZLE_TO_PROBE_OFFSET values for your probe setup to get you working as well as the drivers and axis directions that you are using as I am using TMC2130.

[Marlin_Ender3.zip](https://github.com/MarlinFirmware/Marlin/files/3869775/Marlin_Ender3.zip)

yudi-jpn, 2019-11-22:
@AmiSMB
  Thank you.

The definition by #define confirmed in this thread seems to be the same, but I will flush it with the archive setting and verify it.

Thank you carefully.

sjasonsmith, 2019-11-24:
It is unclear to me whether anyone is still waiting for responses in this issue. If there is any reason for this issue to remain open, I would appreciate a very brief description of the remaining issue, and I am more than willing to try to help. This is core functionality which I have tested a lot in the past week, so I would be surprised if it is not functional.

One thing I did not see mentioned is that probe offsets are stored in the EEPROM. Make sure to reset the EEPROM contents using M502 (to reset) and M500 (to save), to rule out bad values loading from the EEPROM.

draregerreip, 2019-11-24:
@OughRay I  had the exact same issue (ender 3/skr 1.3/marlin 2.0) and I finally solved it by just realizing that my NOZZLE_TO_PROBE_OFFSET was stuck at (0.0.0) in the EEPROM even after reloading a new firmware. After that, i just had to send a M502 to clear the EEPROM and stop using the LCD to send the autoleveling command.
Instead I use through octoprint bed visualizer (or pronterface) and type :
G28
G29 T1 V4   (the 4 is just to get as much info as possible in the the terminal).
Hope that solves your problem (I just lost about 10 hours on this issue on my ender 3).

boelle, 2019-11-25:
@OughRay did it help to clear the eeprom?

M502 followed by M500

boelle, 2019-11-26:
@OughRay btw what version of marlin are you using?

AmiSMB, 2019-11-26:
I finally moved my Re-ARM and RAMPS board on a P3Steel away from Smoothieware so used the same Marlin 2 codebase as my Ender 3 setup and I did find the same issue as this but I found the issue to be with the EEPROM settings. I did an M502 followed by an M500 and then made sure I ran a M503 and double checked what the settings were and then found that the G29 command ran fine. HTH

boelle, 2019-11-26:
i bet @OughRay will have his problems solved once he do a M502 and M500

RedOctobyr, 2019-11-28:
> It is unclear to me whether anyone is still waiting for responses in this issue. If there is any reason for this issue to remain open, I would appreciate a very brief description of the remaining [...]

I am having the issue with the probing grid not being centered on the bed. I'm using 2.0.x bugfix, downloaded 11/23/19. I'm using Bilinear leveling, on an Ender 3, with an SKR Mini E3 1.2. Bed size is 235x235mm. Soft Endstops are enabled. 

I have Z_SAFE_HOMING enabled, using BED_SIZE/2. 

I have set my NOZZLE_TO_PROBE_OFFSET as { -42, -13, -1.7 }. 
I have cleared the EEPROM with M502, then M500. 
G28 probes looks to be correctly in the middle of the bed. It probes at 159, 130mm, X and Y. Factoring in my offsets, that would be an "equivalent" nozzle location of 117, 117, which is correct for a 235x235 bed. 

But the G29 grid is shifted offcenter. I wrote down the locations it uses while probing. The probing locations seem to vary as I change MIN_PROBE_EDGE. 

With it set to 20 (safe value, easily avoids my bed clips), the center probe location is 149, 131. It should be 159, 130. The extreme corner probe locations are 62, 33, and 235, 228. 
With it set to 15 (just misses clips), the center location is 146, 131. The extreme corner probe locations are 57, 28, and 235, 233. 
With it set to 10 (had to remove my clips), the center location is 144, 129. The extreme corner probe locations are 52, 23, and 235, 235. 

It looks as though it's incorporating the MIN_PROBE_EDGE, and the max size dimensions of 235mm. 

The grid appears to always center itself halfway between the minimum probing location for an axis, and the maximum probing location. The trouble seems to be that in one direction it is maxing out at the size limit (235mm), but in the other direction it's still based on the MIN_PROBE_EDGE. This shifts the grid off-center. 

I haven't tried probing with Soft Endstops disabled, to see if letting the nozzle move beyond 235mm would let the grid center itself. But that doesn't seem like the "proper" way to work around this.

sjasonsmith, 2019-11-28:
This is expected. It probes as much of the bed as it can reach, which gives an off-center grid. If you want a grid covering the entire bed, you can use UBL, which allows you to manually probe points your probe cannot reach.

RedOctobyr, 2019-11-28:
Interesting, thank you. When I was using TH3D firmware recently (based on Marlin 1.1.9, I think), that G29 process was centered in the bed, as I recall, so this seemed odd to me. 

Do you know if it at least "understands" that it's probing off-center, and takes that into account when making Z adjustments during a print? Or does it treat that grid data as if it's centered on the bed? 

If your bed had a high spot in the center of the probing grid (which would no longer be the center of the actual bed). Would it raise the Z axis when printing over that off-center location? Or would it raise Z when printing over the center of the bed? 

Maybe I'll try using the L R F B parameters to see if I can keep it to a smaller, centered grid. Or if it doesn't matter (that is, if it understands that the grid is off-center, and adjusts properly while printing), then I'm less concerned. 

I actually briefly tried UBL for the first time this morning, but it was probing 100 points, and I didn't quickly figure out how to change that to something quicker like 25 points. Since I sometimes change build surfaces (so the bed surface profile might change), it seemed like I don't necessarily want to use a really-in-depth probing map which might need to be redone if I changed the bed. 

I found some info to read through later on UBL, to learn more about how it works. For now, I went back to bilinear.

sjasonsmith, 2019-11-28:
I don’t personally know, but I assume the probe position is accounted for during correction. I have only looked into the probing process, not the actual correction process.

yudi-jpn, 2019-11-30:
@boelle
Thank you.
For now, my previous problem has been solved.

It was confirmed that the following definitions are valid by the modification of "bugfix-2.0.x / 2019-11-18".
`#define NOZZLE_TO_PROBE_OFFSET {-43, -5, -0.4}`
(I am adjusting the offset from time to time. Don't worry.)

I have now flashed again from "1.1.9" to "bugfix-2.0.x".

I verified the process without resetting the EEPROM.
and, even if the `#define NOZZLE_TO_PROBE_OFFSET` constant is changed, it is correctly stored in the EEPROM.

The problem that the probe descends when changing the parameter from the rotary encoder was also fixed.

The problem of ABL breaking when changing parameters from the rotary encoder during ABL operation is working perfectly as far as we have confirmed today.
(I confirmed that the problem of the Z axis going up forever was solved.)

There is a new desired menu for easy operation for maintenance.
Should I launch a new thread?

Thank you carefully.

thisiskeithb, 2019-11-30:
> Should I launch a new thread?

Yes. Please close this one if your issue is resolved and open a new Feature Request.

yudi-jpn, 2019-11-30:
@thisiskeithb
Thank you.

Since there may be other reports, I will save this thread.
(Because my carelessly closing the thread suspects the action)

For feature requests, I will also launch a thread.
thanks so much.

thisiskeithb, 2019-11-30:
@boelle This can be closed.

closed 2019-11-30 by boelle

Johnday29, 2020-02-25:
> It is unclear to me whether anyone is still waiting for responses in this issue. If there is any reason for this issue to remain open, I would appreciate a very brief description of the remaining [...]

This was my fix!!!!!!! Thanks