MarlinFirmware/Marlin #15852: [BUG] Pullrequest 15722 unable to compile LCD_I2C_routines.c with LPC1768

atoomnetmarc, 2019-11-09:
### Description

Pull request https://github.com/MarlinFirmware/Marlin/pull/15722 breaks compilation.

### Steps to Reproduce
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3827261/Configuration.zip)

1. #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
2. #define ULTI_CONTROLLER
3. Compile for env:LPC1768

**Expected behavior:** 
Successful compilation.

**Actual behavior:** 
Compilation gives error.
e:\latest\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.c:159: undefined reference to `millis'

#### Additional Information

When reverting the changes from https://github.com/MarlinFirmware/Marlin/pull/15722/commits/2bc580bb5de33519e724aa2e932c39fcab0013e4#diff-e5d3fa0289cecacc5a3e41e06e410004 Marlin compiles.

RudyBenner, 2019-11-09:
I get the same error message.

psavva, 2019-11-11:
@p3p could you have a look

p3p, 2019-11-12:
It does indeed look like I broke it,

atoomnetmarc, 2019-11-15:
Verified fix with latest marlin source. 

LPC1768                         SUCCESS

Thanks.

closed 2019-11-15 by atoomnetmarc