MarlinFirmware/Marlin #15855: [BUG] The first screen of Marlin 2.0 is showing some horizontal lines, appearing to be broken

dvdsouza, 2019-11-09:
### Description

We are using the branch dev-2.1.x and the first screen of Marlin is showing some horizontal lines, appearing to be broken.

[image]

After 3 seconds, appears the boot screen and everything works normally.

[image]

### Steps to Reproduce
ZIP file containing the `Configuration.h` and `Configuration_adv.h` files. 
[marlin_config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3827441/marlin_config_files.zip)

GMagician, 2019-11-09:
Better to try 2.0 bugfix it more updated. AFAIK the 2.1 is not yet "active" its just prepared to be used when 2.0 will be released

AnHardt, 2019-11-09:
Disable `#define SHOW_CUSTOM_BOOTSCREEN`.

sl1pkn07, 2019-11-10:
> Disable `#define SHOW_CUSTOM_BOOTSCREEN`.

same here with you tip

407s, 2019-11-10:
These squares appear during the firmware process. Apparently the firmware is loaded from the memory card each time.

ellensp, 2019-11-11:
This is not SHOW_CUSTOM_BOOTSCREEN. 
I also get this on a skr1.3 with SHOW_CUSTOM_BOOTSCREEN disabled
I just presumed this is what you get pre screen initialization.  
It takes a while to setup everything and read the sd card looking for a new firmware.bin 
I deliberately enabled SHOW_CUSTOM_BOOTSCREEN as a test, and it did as I suspected it would It displays the two bars of junk data, then after a short time it displays the Custom boot screen, then it displays the Marlin screen.

dvdsouza, 2019-11-11:
We have tried with:
`//#define SHOW_CUSTOM_BOOTSCREEN`
it did not solve the problem.

We are going to try the branch 2.0 bugfix and we let you guys know the result.

dvdsouza, 2019-11-12:
> Better to try 2.0 bugfix it's more updated. AFAIK the 2.1 is not yet "active" its just prepared to be used when 2.0 will be released

We tried bugfix 2.0, but it gives the same error. Even a totally new code, just configuring the card and the screen, the error is the same.

3dtechbrasil, 2019-11-13:
I am having the same problem with the bugfix 2.0 and the board BIGTREE_SKR_V1_3

LinFor, 2019-11-14:
I have the same issue on each power-up (not only when flashing new firmware).
No SHOW_CUSTOM_BOOTSCREEN defined.
I'm not uncomfortable with the problem, but if someone will undertake to solve it...

PS: LPC1768, SKR1.3, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

dvdsouza, 2019-11-15:
The first time the firmware is loading the "firmware.bin", those horizontal lines do not appear, but when it is loading the "FIRMWARE.CUR" it does appear on each power-up.

3dtechbrasil, 2019-11-15:
We have checked the power supply.
We have wrapped the "flat cable" in aluminum foil and ground everything correctly.
We have deleted the "firmware.bin" and it did not solve the problem.

ellensp, 2019-11-15:
@dvdsouza It doesn't ever load FIRMWARE.CUR. It looks for firmware.bin, if it is on the sdcard it loads firmware.bin and copies it to the flash and then renames the file to fimware.cur. It never looks at firmware.cur again.  Boot loader firmware https://github.com/triffid/LPC17xx-DFU-Bootloader look in main.c   NB if remove the sd card, you still get the bars.

dvdsouza, 2019-11-15:
@ellensp you are right, even without the sd card, we still get the bars.
Currently, we are looking for the error in the firmware boot section.

ellensp, 2019-11-15:
I think the bug is in the bootloader. the bootloader was designed for a smootieboard witch has 5 LEDs. LED1 = P1_18  and this line is activated by the bootloader to indicate status on of the bootloader. Now along comes the skr1.3 and they dropped the leds and connected  1.18 to LCD_EN pin. Once this line is active any activity on the SPI bus is read into the LCD as display data.   I think we need a modified boot-loader that doesn't try and set the non existing LED pins.

shitcreek, 2019-11-15:
I think I might have stumbled upon the solution but I cannot be sure what exactly it was that resolved this issue for me. I think it was when I updated the library.
I deleted this folder: **.platformio\packages\framework-arduino-lpc176x** when I was getting the missing **header:const_functions.h** error, then I deleted the **.pio** folder and launched vscode and updated all platformIO libraries and cleared the cache. Recompiled and flashed the machine. The scrambled bars appear on first boot up when it is flashing the new firmware, but no longer does any time afterwards. It does appear for a split second though. Could someone test this out?

blueeagle69, 2019-11-16:
@shitcreek 
I just tried what you said.
If I hit the reset button, I get a faint horizontal line on my screen for about a second or so and then the main Marlin logo appears.
If I power-cycle my printer then those two horizontal bars appear.
Can't really tell if the time they are displayed is any shorter.

(SKR 1.3 with custom_bootscreen disabled.)

dvdsouza, 2019-11-19:
@shitcreek 
We have tried it out, but did not solve the problem.
Now we are debugging the "bootloader" and "idle" function, to see whether the boot timing has any problem.

dvdsouza, 2019-11-22:
Hey guys, does anyone have anything new about this? We tried a few things, but that didn't solve the problem.

boelle, 2019-11-24:
@dvdsouza have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

ellensp, 2019-11-27:
After a long fight with my J-Link I managed to get debugging working. Now I can only do soft restart with this so you don't get the full screen of garbage, you just get a few lines (same as when you press reset vs full power off)  But it is already on screen before main() is called.

dvdsouza, 2019-11-27:
@ellensp yeah, it is already on screen before main() is called.
We were debugging the idle() function, but we were unsuccessful.

ellensp, 2019-11-28:
Making some headway.  Ill make some notes here in case it helps others quicker than me to find the issue. How I got this far. Firstly disable USE_WATCHDOG in configuration_adv.h, you also have to comment out watchdog_init(); in DebugMonitor.cpp to stop the watchdog rebooting the controller during debugging.  using a j-link set debug_init_break = tbreak Reset_Handler in platformio.ini start debugging. Replug the GLCD. (this puts garbage in its display buffer)  Display is clean until the line     SysTick_Config(SystemCoreClock / 1000); // Start millisecond global counter in framework-arduino-lpc176x/cores/arduino/main.cpp  after this line you have garbage on the LCD until the real first screen is displayed.  This looks like a  @p3p thing to me.

p3p, 2019-11-28:
Not sure why starting the Systick could cause noise on the display .. its just an internal timer interrupt

rflulling, 2019-11-29:
If you compile with Arduino. Can you try to roll back to version of Arduino that you know worked and try to compile again? I have seen a similar issue in certain newer versions of Arduino IDE when compiled with.

3dtechbrasil, 2019-11-29:
@ellensp We will test these settings here.

ellensp, 2019-11-30:
I replaced the boot loader on my SKR 1.3 and the bars are gone.  Can someone else confirm? 
The source for the modified boot loader can be found here https://github.com/ellensp/LPC17xx-DFU-Bootloader  The only change is that I removed all references to the LED's and the setup of those IO pins. 
The already compiled files can be found here https://github.com/ellensp/LPC17xx-DFU-Bootloader-build    
To replace the boot loader follow this guide https://reprap.org/forum/read.php?13,863336
NB this will wipe your current firmware.  So you have to put a firmware.bin on the sd card directly as the USB port on the SKR 1.3 isn't active yet.

boelle, 2019-12-02:
@dvdsouza tried what @ellensp suggested ?

dvdsouza, 2019-12-03:
The test will be held tomorrow or Wednesday night, we look forward to trying it, we just don't have the correct board at this time.

boelle, 2019-12-07:
@dvdsouza what did the test say?

boelle, 2019-12-07:
@dvdsouza since 2.0 was just released a few days ago has this changed this issue at all?

dvdsouza, 2019-12-08:
@ boelle We have tried the 2.0 release and nothing has changed.

@ellensp Unfortunately, right now we don't have an extra board available to change the boot loader, the ones we have are already on the sold machines and we don't want to risk changing the boot loader. If it was inside Marlin, there would be no problem.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

blueeagle69, 2019-12-24:
Yep I have the same issue.
For now I have been ignoring it and carrying on.
Merry Christmas!

boelle, 2019-12-24:
what happens if you wait like 30 sec when booting?

blueeagle69, 2019-12-24:
The board boots up fine after showing the white lines and carries on as normal.  At least it does for me.  Dont know what others are experiencing  :)

sl1pkn07, 2019-12-24:
> The board boots up fine after showing the white lines and carries on as normal.

this lines is the problem

rado79, 2019-12-26:
same problem here - broken white stripes - since coming to 2.0 from 1.1.9 - skr 1.3 - all versions/bugfixes
with or without custom bootscreen
reset after firmware flash = clear bootscreen with a little 1mm "-" on screen
on power up = broken lines

dvdsouza, 2019-12-26:
> is the issue still the same with all the updates in the last 16 days?

Yes, the issue still the same with all the updates!

boelle, 2020-01-08:
@dvdsouza  still no good news with the latest bugfix 2.0.x?

dvdsouza, 2020-01-10:
@boelle  We have no BIGTREE_SKR_V1_3 board at this moment to test the bugfix 2.0.x. We'll have a board on Monday and I'll let you know.

rado79, 2020-01-10:
@boelle no good news, SKR 1.3

Dracrius, 2020-01-11:
Oh I see the same thing on an SKR mini e3 1.2 at bootup!

dvdsouza, 2020-01-13:
@boelle  , no good news!

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

dvdsouza, 2020-02-10:
@boelle , repo updated and the problem still there.

thinkyhead, 2020-02-11:
This is not something that can be addressed in the firmware. Case closed.

closed 2020-02-11 by thinkyhead