MarlinFirmware/Marlin #15857: [BUG] Can't set EXTRUDER_AUTO_FAN_SPEED != 255

BastR, 2019-11-09:
### Description

SKR PRO is a board that is capable of hosting 3 extruders, so it has 3 PWM outputs for extruders fans (FAN0, FAN1, FAN2). I only have one extruder but I use the 3 FAN outputs this way:

-FAN0 is the extruder fan
-FAN1 is the hotend fan that is declared with E0_AUTO_FAN_PIN
-FAN2 is the board fan that is declared with USE_CONTROLLER_FAN

I want to set FAN1 (E0_AUTO_FAN_PIN)  with a value other than 255 for EXTRUDER_AUTO_FAN_SPEED but Sanitycheck.h does not allow it:

**Configuration_adv.h**
```
// @section extruder

/**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 127   // 255 == full speed
//#define CHAMBER_AUTO_FAN_TEMPERATURE 30
//#define CHAMBER_AUTO_FAN_SPEED 255
```

**SanityCheck.h**
```
/**
 * Auto Fan check for PWM pins
 */
#if HAS_AUTO_FAN && EXTRUDER_AUTO_FAN_SPEED != 255
  #define AF_ERR_SUFF "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 255."
  #if HAS_AUTO_FAN_0
    static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
  #elif HAS_AUTO_FAN_1
    static_assert(PWM_PIN(E1_AUTO_FAN_PIN), "E1" AF_ERR_SUFF);
  #elif HAS_AUTO_FAN_2
    static_assert(PWM_PIN(E2_AUTO_FAN_PIN), "E2" AF_ERR_SUFF);
  #elif HAS_AUTO_FAN_3
    static_assert(PWM_PIN(E3_AUTO_FAN_PIN), "E3" AF_ERR_SUFF);
  #endif
#endif
```

I've commented out all the lines above and I can compile, I've tried 127 and 63 instead of 255 and all values works.

I don't understand why it is allowed to set a different fan speed if SanityCheck will systematically reject it. I don't have the coding knowledge but it should be possible to set another fan speed value if the board has PWM outputs or if the user makes some tweaks (with Ramps fan extender or similar)

English is not my first language, I hope to make myself understood

sjasonsmith, 2019-11-09:
Could you try this using the branch from the following open PR? There were some PWM issues on the SKR Pro that are resolved as part of these changes.
#15655

LinoBarreca, 2019-11-09:
The PR mentioned above by @sjasonsmith correctly sets the PWM pins in the board variant. so please try it. iwhatever the result is let us know.

BastR, 2019-11-09:
Same error with #define EXTRUDER_AUTO_FAN_SPEED 106, if I let 255 I compile without error, juste warnings :

`In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32/timers.h:25,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:99,
                 from Marlin/src/HAL/HAL_STM32/HAL.cpp:26:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../shared/Marduino.h:36,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:28,
                 from Marlin/src/HAL/HAL_STM32/HAL.cpp:26:
/Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:254:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/fastio.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32/HAL_SPI.cpp:25:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/../shared/Marduino.h:36,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/HAL.h:28,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL.h:26,
                 from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/HAL_STM32/HAL_SPI.cpp:25:
/Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:254:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/fastio.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32/Servo.cpp:25:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/../shared/Marduino.h:36,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/HAL.h:28,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL.h:26,
                 from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/HAL_STM32/Servo.cpp:25:
/Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:254:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/fastio.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32/fastio.cpp:25:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /Users/bastien/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/../shared/Marduino.h:36,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/./HAL_STM32/HAL.h:28,
[... 58 lines cut ...]
 static EEPROMClass EEPROM;
[... cut ...]
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/fastio.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/timers.cpp.o] Error 1`

LinoBarreca, 2019-11-09:
please attach config files (standard and advanced). I guess you found another bug.

LinoBarreca, 2019-11-09:
no wait. I found the problem. config files aren't needed.

BastR, 2019-11-09:
Ok

LinoBarreca, 2019-11-09:
@sjasonsmith Problem here is that STM32 don't have real "PWM pins". every digital pin can be defined as PWM. to check if a pin is actually defined as PWM you gotta make a loop. so a STATIC ASSERT (at compile time) can't be done. I'm opening a proper issue about this.

LinoBarreca, 2019-11-09:
@BastR I am sorry to say that you gotta wait for that.
We first need to merge #15655 then I'll take care of fixing your problem with the bug #15858

BastR, 2019-11-09:
> @BastR I am sorry to say that you gotta wait for that.
> We first need to merge #15655 then I'll take care of fixing your problem with the bug #15858

No problem, I can wait. Thanks for your work ;)

BastR, 2019-11-10:
@LinoBarreca I'm running your branch now, I've made more test today. It seems I've an issues with the extruder fan too (PLA cooler). If I set fan to 50% from the LCD, he clearly not spins at 50%, he stay very close to 100%. I have measured the supplied voltage at various speed values:

100% - 12V
50%   - 11,2V
20%   - 10,5V 
   1%   -  5,2V

It's not related to your branch, I've put a previous build and this problem is already present.

BlackMelt, 2019-11-10:
Hi there, 
same issue here. 

SKR Pro with 1 Extruder set on Pin PB1
1 Coldend Fan set on Pin PC8
1 Part Colling Fan on Pin PE5

I try'd to compile with #define Extruder_Auto_Fan_Speed 127

and this error shows up https://pastebin.com/xuEYkyEX

This was just to verify the issue from BastR

BastR, 2019-11-10:
@BlackMelt can you try something for me please, try to set your part cooling fan to 50% with your LCD and tell me if you hear clearly your fan slowing. As explained above mine doesn’t low his speed, it’s just to know if it’s hardware or software.

Thanks you

BlackMelt, 2019-11-10:
@BastR So my stepps ware fallowing:
1: Set the Nozzle to <50 °C (to start the Coldend Fan)
2: Set the Part Cooling fan to 50
3: nothing changed even if i hold manualy the Fan Rotors to ensure the same noise Lvl.

Hope this Bug gets Fixed, my ColdendFan is to Powerfull. 

Greez

LinoBarreca, 2019-11-10:
as regards the build error I already opened a bug with technical details which will help someone solve the problem.
Has regards the "uncorrect fan speed", to check if it's a software or hardware problem, you need to:
- use https://github.com/MarlinFirmware/Marlin/pull/15655
- set fan to 50%
- connect an oscilloscope to the base of the mosfet which runs the output.
- measure the wave on it.

if you have a square, 1kHz, 50% duty cycle, wave the software is ok.
Measure it again on the fan connector. if you see the same wave (but at a higher voltage) the hardware is ok.

salacpavel, 2019-11-11:
https://github.com/MarlinFirmware/Marlin/issues/15362

LinoBarreca, 2019-11-13:
this should be closed as duplicate.

closed 2019-11-13 by BastR