MarlinFirmware/Marlin #15858: [BUG] Static Assert shouldn't be used ON STM32

LinoBarreca, 2019-11-09:
### Description

On STM32 Platform every PIN can be "PWM" because they are handled through a timer if they are defined in the variant.
The macro `PWM_PIN` expands to `digitalPinHasPWM(P)` (which is correct)
Problem here is that you can't check it at compile time, (with a static assert)

### Steps to Reproduce
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);

**Expected behavior:** compile fine

**Actual behavior:** 
```
In file included
from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:38:0,
from Marlin/src/HAL/HAL_STM32/timers.h:25,
from Marlin/src/HAL/HAL_STM32/HAL.h:99,
from Marlin/src/HAL/HAL_STM32/HAL.cpp:26: Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1854:5: error: non-constant condition for static assertion static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
```

#### Additional Information
this shouldn't be a static assert on STM32.

LinoBarreca, 2019-11-09:
this was found here
https://github.com/MarlinFirmware/Marlin/issues/15857

shitcreek, 2019-11-13:
Can we close this one?

LinoBarreca, 2019-11-13:
Close the issue? yes. 'cause it's a duplicate of this https://github.com/MarlinFirmware/Marlin/issues/15362

closed 2019-11-13 by LinoBarreca