MarlinFirmware/Marlin #15860: [BUG] Multiple definition of eeprom functions when compiling Marlin 2 with framework-arduino-lpc176x@0.2.1

atoomnetmarc, 2019-11-10:
### Description

When compiling for a board with a LPC1768 you get a multiple definition error on the eeprom functions at the linking stage. The empty eeprom functions in arduino.cpp (from framework-arduino-lpc176x) conflict with eeprom_i2c.cpp from Marlin 2.

I entered an issue (https://github.com/p3p/pio-framework-arduino-lpc176x/issues/26) with pio-framework-arduino-lpc176x but got redirected here.

### Steps to Reproduce

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3828306/Configuration.zip)

1. #define I2C_EEPROM
#define E2END 0x7FFF
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
2. Compile for env:LPC1768

**Expected behavior:** 
Compilation success.

**Actual behavior:** 
Multiple 'multiple definition' errors at linking stage'.

Linking .pio\build\LPC1768\firmware.elf
c:/users/marc/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1768\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_write_byte(unsigned char*, unsigned char)':
C:\Users\Marc\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:111: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'; .pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:e:\eeprom-test\Marlin-bugfix-2.0.x/Marlin\src\HAL\shared/eeprom_i2c.cpp:49: first defined here

#### Additional Information

When I comment out the empty eeprom functions in arduino.cpp (from framework-arduino-lpc176x) the compilation succeeds.

/*
void eeprom_write_byte(uint8_t *pos, unsigned char value) { }
uint8_t eeprom_read_byte(uint8_t * pos) { return '\0'; }
void eeprom_read_block(void *__dst, const void *__src, size_t __n) { }
void eeprom_update_block(const void *__src, void *__dst, size_t __n) { }
*/

Chris https://github.com/p3p/pio-framework-arduino-lpc176x/issues/26#issuecomment-552142902 suggested that this issue may be something to be fixed in Marlin.

How to continue?

boelle, 2019-11-11:
maybe @p3p knows?

p3p, 2019-11-12:
Will have to move the I2C EEPROM implementation over to the PersistentStore api

boelle, 2019-11-21:
@p3p i guess life etc comes first and this will have to wait for a while?

p3p, 2019-11-21:
Tis on the list, but open to anyone who wants to fix it, may just remove them from the framework, but the i2c and spi eeprom implementations do need moved over to the persistent store api.

boelle, 2019-12-07:
@atoomnetmarc since 2.0 was just released a few days ago has this changed this issue at all?

atoomnetmarc, 2019-12-07:
Tested. No change.

Linking .pio\build\LPC1768\firmware.elf
c:/users/marc/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1768\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_write_byte(unsigned char*, unsigned char)':
C:\users\marc\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:111: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'; .pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:c:\Temp\Marlin-2.0.x/Marlin\src\HAL\shared/eeprom_i2c.cpp:49: first defined here

As long as both https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/shared/eeprom_i2c.cpp and https://github.com/p3p/pio-framework-arduino-lpc176x/blob/master/cores/arduino/arduino.cpp define eeprom_write_byte() you will get a compile error.

Manual workaround is still the same: comment out the empty eeprom functions in https://github.com/p3p/pio-framework-arduino-lpc176x/blob/master/cores/arduino/arduino.cpp

void eeprom_write_byte(uint8_t *pos, unsigned char value) { }
uint8_t eeprom_read_byte(uint8_t * pos) { return '\0'; }
void eeprom_read_block(void *__dst, const void *__src, size_t __n) { }
void eeprom_update_block(const void *__src, void *__dst, size_t __n) { }

boelle, 2019-12-11:
tried to delete these 2 folders?

[image]

or hit the trashcan icon?

[image]

atoomnetmarc, 2019-12-11:
No and no. Error from last build was from a fresh .zip file in a new directory.

boelle, 2019-12-11:
try and try

atoomnetmarc, 2019-12-11:
I tried... and failed in deleted the 2 folders. They simply are not there when unzipping a fresh .zip file.

[image]

I cleanly build Marlin just now again from the fresh download.

Here is the build output:
[build log.txt](https://github.com/MarlinFirmware/Marlin/files/3952282/build.log.txt)

With the first error being:

> Linking .pio\build\LPC1768\firmware.elf
> c:/users/marc/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1768\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_write_byte(unsigned char*, unsigned char)':
> C:\Users\Marc\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:111: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'; .pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:d:\Temp\try-and-try\Marlin-bugfix-2.0.x/Marlin\src\HAL\shared/eeprom_i2c.cpp:49: first defined here

atoomnetmarc, 2019-12-18:
I noticed that when erasing the LPC1768 chip via SWD with a Segger J-link that the runtime stats were reset. 
That means that the I2C connected EEPROM is not used. Could it be that the LPC1768 HAL forces FLASH_EEPROM_EMULATION?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

atoomnetmarc, 2019-12-24:
@boelle Yes, issue unchanged.

But if https://github.com/MarlinFirmware/Marlin/issues/15860#issuecomment-567200848 means that i2c EEPROM reading and writing is not implemented on the LPC1768 HAL, then maybe this bug report makes less sense to keep open.

boelle, 2019-12-24:
strange thing is i use a re-arm which is also LPC1768 and i can save to eeprom just fine

atoomnetmarc, 2019-12-24:
Hmm. That is strange.
I know i2c works because the connected oled screen from a Ultimaker 2 works. This would mean that I configured something wrong.
Let me research this. To be continued.

boelle, 2019-12-24:
i dont have an eeprom connected, they use the internal flash for that

atoomnetmarc, 2019-12-24:
EEPROM emulation works, but not saving to an external i2c EEPROM.
Last week I studied the LPC1768 HAL (and other HALs), I could not discover i2c eeprom support.

The one thing I could find is the FLASH_EEPROM_EMULATION define in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/persistent_store_api.h 
I suspect that this define forces saving settings in flash.

atoomnetmarc, 2019-12-26:
It makes more sense to close this issue and wait until i2c EEPROM support has been build into the LPC1768 HAL.

closed 2019-12-26 by atoomnetmarc