MarlinFirmware/Marlin #15863: [BUG] Chamber falsely errors with MAXTEMP on power up SKR 1.3

ellensp, 2019-11-10:
#### Description
With Chamber enabled,  see #15861 to get this to compile, on power on the SKR1.3 immediately goes into an error condition:  MAXTEMP Chamber, PRINTER HALTED, Please Reset   

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3828402/configs.zip)
It is easier with  a LCD attached, I used a RRD GLCD clone.

1. Enable chamber by setting TEMP_CHAMBER_PIN, HEATER_CHAMBER_PIN and TEMP_SENSOR_CHAMBER 
2. Change #define CHAMBER_MAXTEMP    to 600   (in Configuration_adv.h) 
3. Compile and Upload the firmware and reset the controller board
4. Put a 200k variable resistor on the Chamber thermistor pins
5. Set the variable resistor so the Chamber temperature reads something sane eg 35 degrees.
6. Edit define CHAMBER_MAXTEMP  back to 60 
7. Compile and Upload the new firmware and reset the controller board
8. The board will now constantly error MAXTEMP Chamber even though the variable resistor is still set to 35 degrees.

**Expected behavior:** [What you expect to happen]

This error should happen.

**Actual behavior:** [What actually happens]

The error displays and the buzzer sounds.

#### Additional Information
Image of test environment, Chamber temperature set to 35c
[image]
Image after MAXTEMP Chamber lowered to 60c and firmware uploaded.
[image]

p3p, 2019-11-10:
I thought I'd managed to work around all the start up temperature issues at this point, but the 12bit ADC PR may have brought some back.. does `#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000` solve the problem?

stefi01, 2019-11-10:
since when adc was changed a few days ago i get this same error with skr 1.3 but i have no chamber enabled, mine is with the bed. start up printer and mine says err: max temp: bed

p3p, 2019-11-10:
Make sure the `framework-arduino-lpc176x` package is up to date (0.2.1) using platformios update all (`platformio update` from the terminal or the appropriate button in the IDE you use)

AnHardt, 2019-11-10:
Please report:
Number of temperature table
MAXTEMP value
MINTEMP value
Guessed temperature of the measured object when the error happens

AnHardt, 2019-11-10:
@p3p 
Midpoint initialization is much better then endpoint initialization. At least there is a chance the first value is between min- and max-temp. However. If the maxtemp is low enough there is still a chance the init is outside the allowed range.
Could be improved by the midpoint between mintemp and maxtemp, or ideally init by the first real measured value.

ellensp, 2019-11-10:
@AnHardt 
Requested information is in the configuration files provided. 
For the Heated chamber:
Number of temperature table  1
MAXTEMP value 60 
MINTEMP value 5 
Guessed temperature of the measured object when the error happens. 35 as is originally stated, is a variable resistor set to 35c for testing!

AnHardt, 2019-11-10:
```cpp
int filter(int adc, int len) {
  static int akku = 0;
  if (!akku) akku = adc;
  else         akku = akku * (len-1)/len + adc; 
  return akku;
}
``` 
Something like that.

ellensp, 2019-11-10:
@p3p  was already 0.2.1 "Updating framework-arduino-lpc176x       @ 0.2.1          [Up-to-date]"

AnHardt, 2019-11-11:
```cpp
  { OV( 480), 115 },
  { OV( 516), 110 }, // !!!
  { OV( 553), 105 },
  { OV( 591), 100 },
  { OV( 628),  95 },
  { OV( 665),  90 },
  { OV( 702),  85 },
  { OV( 737),  80 },
  { OV( 770),  75 },
  { OV( 801),  70 },
  { OV( 830),  65 },
  { OV( 857),  60 }, // !!!
  { OV( 881),  55 },
  { OV( 903),  50 },
  { OV( 922),  45 },
  { OV( 939),  40 },
  { OV( 954),  35 }, // !!!
  { OV( 966),  30 },
  { OV( 977),  25 },
```

p3p, 2019-11-11:
I know what you mean AnHardt I didn't want to have to keep track of extra state, with the lowpass set to 2 as it it is now it would only take a few readings to initialise.

The issue is with 12 bit PR the 16x multisample is no longer present so the first few readings are being used as is from the lowpass filter.

I'm pretty sure there is a pid Dt issue with that PR so until that is looked into it may be best to set the LPC HAL back to using Marlins filtering at 10bit.

p3p, 2019-11-11:
If it is the initial reading `#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000` should work around it .. unless it doesnt work with case temp.

ellensp, 2019-11-11:
Yes this works 
#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000

ellensp, 2019-11-11:
Perhaps this should be enabled in src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h by default?

p3p, 2019-11-11:
No it does need fixed one way or another, without Marlins 16x multisample any MAX temp set below the midpoint of the active temperature table may cause a max temp error even with the current minimal lowpass filtering.

I'l update the framework to use the first reading read to initialise the filters, .. if people have noisy wiring it could very well make the situation worse for them but for most it should be better.

p3p, 2019-11-11:
@ellensp can you update Marlin and tell me if the update solves the issue. Platformio should be forced to update the packages .. hopefully.

ellensp, 2019-11-11:
Removed BOGUS_TEMPERATURE_GRACE_PERIOD from Configuration.h. Updated framework-arduino-lpc176x to 0.2.2. Compiled and uploaded. no error on startup. Tried various Chamber temperatures. Seems to be sorted.

closed 2019-11-11 by ellensp