MarlinFirmware/Marlin #15864: [FR] (feature request title)

hitchhiker85, 2019-11-10:
Hi!

I am absolute new in this 3d printing game :)

I own a Ender2

After looking some Youtube videos i print the FreeABL mounts and flash bootloader and Marlin 2.0 via PlatformIO on VScode.

now i got some problems!

My K-Fan cant adjust now via Display an knob

My Bedtemp. for ABS can only adjust to 65.

can someone please help?

I have the sanguino mcu on it.

I dont know what i must change to get higher Bedtemp and K Fan adjustment direct on the Ender2.

Thanks

swilkens, 2019-11-10:
Since you used PlatformIO to compile a new firmware for your printer, you must have made some changed to the `configuration.h` and `configuration_adv.h` files.

Please find those files, add them to a .zip file and post that here so we can have a look what you did :)

thisiskeithb, 2019-11-10:
> Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md
> 
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.
> 
> Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

hitchhiker85, 2019-11-10:
sorry!  i go to the forum! facebook is not mine!

thanks for help!

swilkens, 2019-11-10:
Good luck :) !

Please remember to close issue reports when your question has been answered.

closed 2019-11-11 by boelle