MarlinFirmware/Marlin #15865: [BUG] (short description) SKR 1.3 with TMC2208 UART and TMC5160 SPI both freezes board on allmost any print with longer lines (bigger dimensions).

Charly333, 2019-11-10:
### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-11-11:
StealthChop does not work / work well with Linear Advance. Some have reported rattling noises and in my case, it causes my X and Y to stutter. Try either turning off stealthChop for the extruder or not using LA. https://github.com/MarlinFirmware/Marlin/issues/14634 https://github.com/MarlinFirmware/Marlin/issues/10272

ManuelMcLure, 2019-11-11:
The rattling is not caused by StealthChop - I’ve seen extruder rattling even with A4988 drivers on the extruder.

lavstudia, 2019-11-11:
ManuelMcLure - I confirm. Same problem. Tested on TMC2208 and AA4988.

Charly333, 2019-11-11:
Just had another try few seconds ago - Spread Cycles hangs up the board much quicker than Stealth Chop - but both do hang up the board now already on small benchys.... I´ve tried to go back to version from 10.10 that did work fine but it doesn´t compile any more. Some missing includes in pio arduino folder - that are not known to platformio.io ... any idea to get it compiled again?

shitcreek, 2019-11-11:
Have you tried deleting the .pio folder, relaunch vscode, then update all (platformio) libraries?

Charly333, 2019-11-13:
@shitcreek 
did so, but didn´t help. Here´s the error i get...

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/./HAL_LPC1768/../shared/Marduino.h:36,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/./HAL_LPC1768/HAL.h:39,

                 from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
C:\users\john5\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35:10: fatal error: const_functions.h: No such file or directory
 
*************************************************************************
* Looking for const_functions.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:const_functions.h"
* Web  > https://platformio.org/lib/search?query=header:const_functions.h
*
*************************************************************************
 
 #include <const_functions.h>
          ^~~~~~~~~~~~~~~~~~~

shitcreek, 2019-11-13:
I can't reproduce this error. Are you using the recent build of Marlin?
Check line 36 of your Marduino.h file and see if it's `#include <Arduino.h> `

thisiskeithb, 2019-11-13:
Moving Marlin to the root of your hard drive might help if you're on Windows since it doesn't handle long file paths well.

Charly333, 2019-11-14:
@shitcreek - checked - include arduino.h is in line 36
This only happens with some older versions from before November. They had been compiling well - I now have to backdrop cause new versions freeze skr board. But all versions on HD and from external backup don´t compile any more. Fresh versions loaded this week do compile, but do crash skr board...

@thisiskeithb 
thanks, did with folder directly under root - same problem

Charly333, 2019-11-14:
Maybe worth to mention - having atmega in platformio.io and configuration.h in those old versions it does compile - as soon I switch back to SKR 1.3 board missing link is reported as lined up above.

p3p, 2019-11-14:
Others use the new framework version on SKR 1.3 so you will have to be more specific with what you mean by freezing? and make sure you keep no files from the old version, not even the pins file.

If you do upgrade to the new framework version to roll back you need to delete all platformio cache before building the old version of Marlin.

shitcreek, 2019-11-14:
Ok, I was able to reproduce this issue with the missing **header:const_functions.h** this morning and my solution was to simple delete the folder **.platformio\packages\framework-arduino-lpc176x**

shitcreek, 2019-11-14:
I also had to delete .pio folder again and relaunched vscode.

Charly333, 2019-11-14:
@shitcreek 
You´ve saved my day!!!! Can compile the older versions now. Will step back to the last working  october version and wait until SKR1.3 will be supported by marlin error free again.

thisiskeithb, 2019-11-14:
> Will step back to the last working october version and wait until SKR1.3 will be supported by marlin error free again.

The SKR 1.3 is supported "error free", but sometimes PlatformIO needs a little manual help when it gets hung up.

karabas2011, 2019-11-20:
> > Will step back to the last working october version and wait until SKR1.3 will be supported by marlin error free again.
> 
> The SKR 1.3 is supported "error free", but sometimes PlatformIO needs a little manual help when it gets hung up.
You mean compiling old marlin?

p3p, 2019-11-20:
There is no known problem with SKR 1.3 using the current bugfix 2.0.x, This issue is probably caused by `MONITOR_DRIVER_STATUS` and `STOP_ON_ERROR` but that is not the SKR 1.3 "freezing" that is what those options do, I don't know why the drivers would report an problem I don't use TMC drivers.

In general `MONITOR_DRIVER_STATUS` is not a good idea after you get things working as it causes overhead during the print because of how the communications work.

karabas2011, 2019-11-20:
Ok, I disabled these and will check.

karabas2011, 2019-11-21:
Black screen again. Does not work.

karabas2011, 2019-11-22:
No success, black screen on october snaphot

MitchelBerberich, 2019-11-22:
Hi. I don't know if this is related: Using current bugfix 2.0.x, my SKR 1.1 freezes after a short time, too. 

Drivers are drv8825 on x and y (dual stepper drivers each) and tmc2208_standalone on z. So no problem directly related to advanced stepper functions (uart, spi). Have seen someone reporting the problem using a4988 above, which seems to confirm, that advanced stepper functions are not the problem.

No problems when using the lcd screen to move axis. No problems, when homing. 
The problem occurs when running gcode from sdcard (onboard and lcd) and when trying to run gcode from remote (cncjs on raspberry connected via usb).
Running the gcode-script the startup code (homing and waiting for user to press button) is ok, then some more gcode commands are executed and then the board freezes (no stepper movements anymore, no reaction on lcd, usb-serial-connection to raspbery simply disappears / disconnects). The time it takes for the board to freeze varies (one second up to some seconds). Someone said above the problem occurs on long movements: Well, in this case there are only small movements, but slow movements, as I'm using the board for a cnc milling machine.

Disabling ADAPTIVE_STEP_SMOOTHING, HOST_KEEPALIVE_FEATURE and setting MM_PER_ARC_SEGMENT to .3 instead of 1 (differences from my config to the config in the bigtree-tech-repository) somewhat helped ... no freeze anymore! BUT: after some time, the Z stepper doesn't move anymore. Z-Movements instead take place on the Y-Axis! This persists until reboot and also affects moving the axis via lcd menu. And in the first line of the lcd display there are chinese chars ... 

all of this => memory / stack / concurrency-problem?

i'll try to investigate a bit further later. but while having experience in programming and software-engineering, I'm pretty new to marlin, so not sure how much I can help on this (besides testing).

p3p, 2019-11-22:
If you have gcode that can reproduce the issue, please supply it along with the configs that are used.

MitchelBerberich, 2019-11-22:
@p3p as requested my configs and gcode (the version that causes the freeze after a few seconds of moving)

if there's anything else you need or which I should try, please say so.

**IMPORTANT** As said, GCode is for CNC, that means: it will home on all axis, move to Z5 and wait for the user to push the button on the lcd. Then it will move to Z14.5 and from that height it should go down again to Z0. Doing so it should "draw" a filled star with height 3mm in the center of a rectangle which goes down all the way. GCode is for xyz-movement only, no spindle control (and of course no extrusion), so it should also be ok to execute on a printer, if you stop it before reaching Z0 (or better replace Z0.* by a safer value)

[Issue15865-CfgAndGCodeByMB.zip](https://github.com/MarlinFirmware/Marlin/files/3880828/Issue15865-CfgAndGCodeByMB.zip)

p3p, 2019-11-24:
@MitchelBerberich I can reproduce your freeze issue, so moving forward on that, seems to be directly related to `ADAPTIVE_STEP_SMOOTHING`. Don't have the slightest idea how moves could suddenly be executing on the wrong axis though ..

boelle, 2019-11-24:
@Charly333 did you solve your issue?

p3p, 2019-11-25:
@MitchelBerberich could you try #15996 and see if that solves your issue.

MitchelBerberich, 2019-11-25:
@p3p My issue seams to be solved with your patch. Everything ok now. Thank you very much for your fast help.

(Just_ to remember: The original issue reprted by @Charly333 most likely is an other problem, as my problem turned out to be related with my special setup for cnc without any extruder. So the original issue reported by Charlie333 probably has to be kept open for now.)

boelle, 2019-11-26:
@Charly333 btw what version of marlin are you using?

boelle, 2019-12-01:
since  #15996 is merged i assume that this one is solved and will close

we can always reopen if the OP is still having issues

closed 2019-12-01 by boelle