MarlinFirmware/Marlin #15866: E1 Err: MINTEMP / E1 THERMAL RUNAWAY (PRINTER HALTED Please reset)

Primmson, 2019-11-10:
Hi, I have a problem with temperature swinging and with error ''E1 Err: MINTEMP / E1 THERMAL RUNAWAY (PRINTER HALTED Please reset)''. I have SKR 1.3 board and 2208 UART driver on Ender 3. When turn on printer temperature works like normal. Then heated to working temperature  (in this time works well) when starts printing (via USB or card) temperature swings a lot (like from 150°C to 210°C in range with less than second. When printing around one hour or less, printer beeps and I got error ''E1 Err: MINTEMP PRINTER HALTED Please reset'' or ''E1 THERMAL RUNAWAY PRINTER HALTED Please reset''. And I have a second problem with speed of printings. I print around 50mm/s and one straight line prints like 50mm/s, then one random line print like 100mm/s. Can I setup this in configuration.h?
What could be a problem for both situations?

I record video from this situation (link below)
https://drive.google.com/open?id=13DBjCPzpazk-YW4d8C1iu6ngvgbPMwuV

p3p, 2019-11-10:
It looks like you have a wiring issue with the thermistor, it's unlikely to be just electrical noise with that much deviation caused by movement so check thoroughly that all the connections are sound and that there are no breaks in the wiring.

Primmson, 2019-11-11:
I replace termistor one month ago and then works okay. Yesterday I do PID autotune and check resistance of termistor.

Primmson, 2019-11-11:
> It looks like you have a wiring issue with the thermistor, it's unlikely to be just electrical noise with that much deviation caused by movement so check thoroughly that all the connections are sound and that there are no breaks in the wiring.

I replace termistor one month ago and then works okay. Yesterday I do PID autotune and check resistance of termistor.

boelle, 2019-11-11:
did  you try and replace the thermistor again?

boelle, 2019-11-11:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Gerggg, 2019-11-11:
I have a similar problem, the "THERMAL RUNAWAY" does not appear but actually the nozzle temperature continues to go from 215 ° to 209 ° as if the stabilization of the temperature was stable.

Gerggg, 2019-11-11:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3832133/Marlin.zip)

p3p, 2019-11-11:
@Gerggg What you describe sounds like a smooth curve on the temperature graphs, thats caused by PIDs not being tuned properly and is different from the instantaneous dodgy readings shown in this issue.

A resent PR changed the PID dT, but when PIDs are loaded from EEPROM this is not taken into account, so you need to retune or reload the defaults.

Primmson, 2019-11-12:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3836592/Marlin.zip)
Hi, this is my configuration. I installed SKR 1.3 board with 2208 stepper drivers (UART) and BL touch.

p3p, 2019-11-12:
Sorry @Primmson but at this point I have to assume it's a hardware issue with your printer as the temperature readings jump around instantly and by large amounts showing huge amounts of noise, I've asked a few others with SKR 1.3 boards and I can't find anyone else with a similar problem.

I'm hesitant to offer you a software solution because it would just hide the underlying hardware issue not fix it and that could cause more issues.

Primmson, 2019-11-12:
> Sorry @Primmson but at this point I have to assume it's a hardware issue with your printer as the temperature readings jump around instantly and by large amounts showing huge amounts of noise, I've asked a few others with SKR 1.3 boards and I can't find anyone else with a similar problem.
> 
> I'm hesitant to offer you a software solution because it would just hide the underlying hardware issue not fix it and that could cause more issues.
> 
> (If I do find anything in software that could be causing this, I'l let you know .. but it's a lot more likely its hardware.)

Hi, I connect new thermistor and now is reads ok. But I have problem with acceleration when printing. I reduce speed and acceleration on ender (not on configuration.h). When printing with speed 40mm/s works great, and then just one straight line prints with abnormal speed and acceleration. Where can I set up this? Thanks for all answers!

boelle, 2019-11-12:
will close this one as the issue is fixed

for questions go here

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-12 by boelle