MarlinFirmware/Marlin #15867: [BUG] MAXTEMP : bed error

tvm2487, 2019-11-10:
I just flashed the latest marlin 2.0 bugfix on my cr-10s with an skr v1.3 and tmc2208 v3 drivers and now all the sudden I'm getting a maxbed temp error as soon as I turn my printer on everything was fine before I flashed the latest firmware so I'm not sure what happened, I also tried 2 new thermistors for the hotend and bed with the same result

thisiskeithb, 2019-11-10:
Clean your libraries or delete your .PIO folder and try updating again.

AnHardt, 2019-11-10:
Please report:
Number of temperature table
MAXTEMP value
MINTEMP value
Guessed temperature of the measured object when the error happens

shitcreek, 2019-11-10:
Do what @thisiskeithb said. I'm 100% certain that is the issue and solution as myself and others have confirmed.

closed 2019-11-10 by shitcreek

Heimdoulila, 2020-06-08:
Hi,
I have the same problème, I did what @thisiskeithb  said but is not working for me.
I have an SKR 1.4 with TMC2208 and TFT24. My printer is CR10s and I use Marlin 2.0.5.3.

without connecting the bed, I start the card via the computer, it beeps then the message "**MAXTEMP : bed error**" appears.

Thanks you for your help, I've been looking for the solution since this morning