MarlinFirmware/Marlin #15873: [BUG] STM32F103VET6 and software serial

0Jmaster0, 2019-11-11:
After successful building of Marlin 2.0 bugfix for MKS Robin (bases on STM32F103VET6), I tried to add TMC2208 UART control. Pins_MKS_Robin don't have uart description, so I copied corresponding part from SKR V1.3 config. As sample I tried to make X stepper working, first with hardware UART, to do it I disabled Wifi serial in configuration.h
` #define SERIAL_PORT_2 -1`
And uncommented line in pins
` #define X_HARDWARE_SERIAL  Serial1`
Everything worked, so I tried to use software UART. Robin have unused pins in XS1 (PC3 and PA1) and in XS2(PF9 and PF8), choosed XS2 for X. Commented them as servos, disabled hardware serial and enabled software serial
`  #define X_SERIAL_TX_PIN  PF9
  #define X_SERIAL_RX_PIN  PF8`
It compiled, flashed to controller and all I receive "TMC communication error". Tried listening port with usb2com - nothing, measured pins voltage with tester(TX should be pulled up to 1) - both zero. Tried to bind LED to PF9 and PF8 - both working.
Is there something I have to enable first? Or software serial is unavailable for STM32F103?

Attached my Configuration.h(taken with Configuration_adv.h from examples, only change is tmc2208 choose for X and disabled second serial, no changes to Configuration_adv.h ), working pins with hardware serial and not working pins with software ones.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3832572/Configuration.zip)
[pins_MKS_ROBIN_software_serial.zip](https://github.com/MarlinFirmware/Marlin/files/3832576/pins_MKS_ROBIN_software_serial.zip)
[pins_MKS_ROBIN_hardware_serial.zip](https://github.com/MarlinFirmware/Marlin/files/3832577/pins_MKS_ROBIN_hardware_serial.zip)

Second question, if first one can be done, is following: in some boards like "pins_BIGTREE_SKR_E3_DIP.h" pins for rx and tx are the same, but in skr v1.3 these ones are different. Can I combine this pins for Robin?

```
#define X_SERIAL_TX_PIN  PC10
#define X_SERIAL_RX_PIN  PC10
```

0Jmaster0, 2019-11-11:
So, I found answer to both questions. For first one, in lastest Marlin version softwareserial.cpp is dummy,  inside it there is a following text:
```
/**
 * Empty class for Software Serial implementation (Custom RX/TX pins)
 *
 * TODO: Optionally use https://github.com/FYSETC/SoftwareSerialM if TMC UART is wanted
 */
```
And really, downloading Fysec library and placing it into Marlin\src\HAL\HAL_STM32F1\ does everything I wanted.

And second answer is also "yes". It is possible to bind tx and rx of virtual port to single output. In this case you don't need 1k resistor between tx and rx, simply connect pdn to chosen output and everything will work as it must

closed 2019-11-11 by 0Jmaster0

daniyo27, 2020-01-18:
Hey, really glad you posted this. I have the same board and am considering the TMC2208 upgrade as well. Could you upload your new pins files?  Would be great help! Thanks in advance!

sjasonsmith, 2020-01-18:
> And really, downloading Fysetc library and placing it into Marlin\src\HAL\HAL_STM32F1\ does everything I wanted.

Several boards specify SoftwareSerialM dependencies in the platformio.ini file. I didn't add it for MKS boards because I didn't have anything to test with. If you add it and it works, I'm sure a pull request would be welcome to help other users who encounter the same problem.

0Jmaster0, 2020-01-18:
**daniyo27,** I can provide file, but it is very tricky to solder wires to controller and 99.9% you'll need to have microscope. Stm32F103 have pins with only 3.3v level and 5v tolerant. All of the pins, that are free and already soldered on board are 3.3v olny. I tried to add protection circuits with dividers or zener diodes, but after it TMCs were not working. So I found out, that part of port F and port A are 5v tolerant and are free, so I tried to solder wires to them, but all of wires were too thick for this task(I didn't want to use single thread wires). The ones on photo were the smallest ones. 
[image]
[pins_MKS_ROBIN.zip](https://github.com/MarlinFirmware/Marlin/files/4080836/pins_MKS_ROBIN.zip)

 **sjasonsmith,** this file was not mine and all I do - I followed instruction in current source file... The person, who write this instuction was able to do this, but didn't do this for some reason...

daniyo27, 2020-01-18:
Thanks for the photo!  Looking at the photo and a pinout, it seems you hooked  black wire to VDDA? and Yellow to PA 0, white to PA3, purple to VDD4?, and orange to PA5?

Couldn't help but to notice that you had the BTT TMC2208 V3 and Fystec TMC2208's... which both seems to have different pinout than the Robin would accept. Specifically the VM rail.

[image]

[image]

Could this be a factor why your chips didn't work?  MKS is the only one with the unconventional pinouts... 

All TMC2208 datasheets for Fystec and MKS show they are 3.3v and 5V logic tolerant... unless you mean the TMC's are powered by 5V and the PC3, PA1, PF9, and PF8 are 3.3V logic only?  There seems to be someone with a Robin Nano board that the manufacturer replied to manually wire to the XS1 and XS2 of that board. [https://github.com/makerbase-mks/MKS-Robin/issues/146](url)

0Jmaster0, 2020-01-18:
Sorry, send you wrong file, this is correct one:
[pins_MKS_ROBIN.zip](https://github.com/MarlinFirmware/Marlin/files/4081797/pins_MKS_ROBIN.zip)
I made those changes on my work computer, not the home one, so I haven't copied last files from it :)
No, the chip layout is tricky too, the key mark on chip is top-left on photo, despite chip label is rotated :) This can be mind-blowing for these chips and I completely forgot about it :) Wiring diagram is following(look at datasheet and pinout and everything become clear)
X stepper - PE6
Y stepper - PE4
X stepper - PE2
E1 stepper - PE1
E2 stepper - PE0 
TMC-s are 5v tolerant, that's true, but if you, for some reason, remove them and replace with a4988 or whatever else and place jumpers on pads, you'll receive 5v on chip pins. If pin is not 5v tolerant - it'll kill pin/port/chip. Robin have 4 free ports, but I wanted 5 pins and didn't want any additional wires, so I done everything this way.
Bigtreetech or Fysetc - is not a problem, chip is the same. All you need is to find place where to wire pads under chip. Look for schematics for both of them for that info

daniyo27, 2020-01-18:
Thanks for the additonal info.  

I just wanted to double check, if I had 4 stepper drivers with TMC2208 I can use the 4 free ports?  Do I pull the jumpers for each driver?  Trying to avoid soldering if I can.

Thanks!

0Jmaster0, 2020-01-18:
Yes you can, but, in that way, cut out bottom pins(BTT have pins on top and bottom part of chip for it), that TMCs are using as UART, just to prevent possibility to hit chip with 5v jumper

daniyo27, 2020-01-18:
Oh, you mean don't allow the PDN/UART pins go into the socket and only allow it to go to XS1 and XS2.  Got it!  Do I need to pull any jumpers in Jm4_M2?

0Jmaster0, 2020-01-18:
Yes, that's correct!
 It's better to remove them, but if you control TMCs with uart they seem to be ignored in any way :)

daniyo27, 2020-01-18:
Ok! Thank you for the confirmation!  Just to confirm, the UART pins do output at 3.3V as long as the pins are not connected in the socket?  That's the only confusion I have now as you mentioned XS1 and XS2 are NOT 5V tolerant.

Will be doing this mod with TMC2209 when the drivers arrive! Will report back my results here! :)

0Jmaster0, 2020-01-19:
As for TMC2209 there is one more thing: they have slave address and you can connect up to 4 of them to single uart port(doesn't matter software or hardware), but I haven't tried it myself. Two remaining jumpers are forming slave address. If you use different pins for RX and TX, you can use zener diode overvoltage protection.

daniyo27, 2020-03-17:
Just wanted to report back huge success! A big thank you for the helpful hints! Everything was working on the first shot!  I was able to desolder the MKS TMC2209 UART pins and flip them around. I did not choose to go with one wire for everything so I stuck with one per TMC.

[image]

Seems like Marlin 2.0.5.1 makes my screen go to blank after the bootscreen. No issues with 2.0.4.4.
Attached are the pins file and instructions for softwareserialm and enabling it in configuration.h

[MKS_ROBIN.zip](https://github.com/MarlinFirmware/Marlin/files/4342127/MKS_ROBIN.zip)

thinkyhead, 2020-03-19:
Thanks for the heads up @daniyo27. Have a look at #17207 for a patch adapting your changes.