MarlinFirmware/Marlin #15877: [BUG] Fan0 turns off after initial tool/extruder change

eightheads, 2019-11-12:
### Description

I have been playing with dual extrusion lately, 2 extruders single nozzle, and have been having an issue where the initial tool change back from T1 to T0 causes the part cooling fan(Fan0) to turn off. There does not appear to be a M106 or M107 in the GCode to actually call for the fan change, but this happens every time but only on the initial change back from T1 to T0. 

1. Start print with fan off M106 S0
2. Print moves from T0 to T1 at the first layer change.
3. M106 S255 is called in GCode as it should
4. On tool change back to T0, the fan goes to 0. This is also shown on LCD showing fan as off.

**Expected behavior:** 
Expect for fan to stay at 255 as previously called. No fan change gcodes have been called.

**Actual behavior:**
Fan turns on at the second layer with M106 S255 as expected during the first T0 to T1 tool change, this is normal and expected. On second tool change T1 back to T0 fan turns off without GCode commanding it to. If I manually push M106 S255 fan will stay on the remainder of the print.

#### Additional Information
This is the section of GCode where this takes place, right at the T0 change before prime pillar.

```
    G1 X141.247 Y157.144 F6000    
    G1 X141.210 Y154.822 F6000    
    G1 Z0.400 F1002    
    G1 E5.0000 F2700    
    G1 X140.539 Y155.492 E0.0300 F2082    
    G1 X140.452 Y155.013 E0.0154    
    G1 X141.142 Y154.323 E0.0308    
    G1 X141.124 Y154.290 E0.0012    
    G1 X141.052 Y154.198 E0.0037    
    G1 X141.025 Y154.171 E0.0012    
    G1 X140.909 Y154.083 E0.0046    
    G1 X140.852 Y154.048 E0.0021    
    G1 X140.336 Y154.564 E0.0230    
    G1 X140.242 Y154.239 E0.0107    
    G1 X140.199 Y154.135 E0.0036    
    G1 X140.499 Y153.835 E0.0134    
    G1 X140.147 Y153.622 E0.0130    
    G1 X140.036 Y153.733 E0.0050    
    G1 E-30.0000 F2700    
    G1 Z0.600 F1002    
    ; process Color1    
    ; layer 4, Z = 0.400    
    T0    
    ; prime pillar    
    G1 X128.755 Y156.963 F6000    
    G1 Z0.400 F1002    
    G1 E30.0000 F2700    
    G1 X109.155 Y156.963 E0.6193 F2700    
    G1 X109.155 Y137.363 E0.6193    
    G1 X128.755 Y137.363 E0.6193    
    G1 X128.755 Y156.963 E0.6193    
    G1 E-5.0000 F2700    
    G1 Z0.600 F1002    
    G1 X109.382 Y156.823 F6000    
    G1 Z0.400 F1002    
    G1 E5.0000 F2700    
    G1 X109.295 Y156.736 E0.0039 F2700    
    G1 X109.295 Y156.170 E0.0179    
    G1 X109.948 Y156.823 E0.0292    
    G1 X110.513 Y156.823 E0.0179    
    G1 X109.295 Y155.605 E0.0544    
    G1 X109.295 Y155.039 E0.0179    
    G1 X111.079 Y156.823 E0.0797    
    G1 X111.645 Y156.823 E0.0179    
```
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3834154/Archive.zip)

eightheads, 2019-11-12:
Here is the full GCode file from above the the issue present. It happens at line 5771
[Gear_11_11_S3DOriginal.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3834180/Gear_11_11_S3DOriginal.gcode.zip)

ellensp, 2019-11-12:
The first thing I wanted to look at was the pins files, but you have BOARD_BIGTREE_SKR_V1_4  which I can find no information...

ellensp, 2019-11-12:
Looking through the marlin code It looks like each tool has its own fan speed (even with a single nozzle). You set a fan speed of 0 for T0 so it switched off every time it was selected.
your gcode in short (stripped out comments, everything not T and not M106 related. )

M106 S255      < set T0 fan 255
M106 S0          < reset T0 fan to 0
T0
T1
M106 S255     < set T1 fan to 255
T0
T1
T0
etc 

but it does not set t0 fan back to 255 anywhere..  

Can someone else confirm this?   (I still consider myself somewhat green)

thisiskeithb, 2019-11-12:
> The first thing I wanted to look at was the pins files, but you have BOARD_BIGTREE_SKR_V1_4 which I can find no information...

[image]

Coming soon 🙂

eightheads, 2019-11-12:
> Looking through the marlin code It looks like each tool has its own fan speed (even with a single nozzle). You set a fan speed of 0 for T0 so it switched off every time it was selected.
> your gcode [...]

This does not happen every time the tool switches back to T0, only after the first switch back to T0.
So:
T0 first layer, no fan, correct
T1 first layer, no fan, correct
T1 second layer, fan on, correct
T0 second layer, fan turns off, incorrect

If I turn the fan on again after this it will stay on the remainder of the print.

eightheads, 2019-11-12:
> > The first thing I wanted to look at was the pins files, but you have BOARD_BIGTREE_SKR_V1_4 which I can find no information...
> 
> [image]
> 
> Coming soon 🙂

Shhhh

InsanityAutomation, 2019-11-17:
Ill take a look at the tool change code. Originally this was written that if no fan speed was stored it would carry over the current. 

There was some debate on that one if they explicitly wanted 0 for the second tool, eg ABS and HIPS where they want a slow fan on the hips but none on the ABS. Not sure a good balance can be made to have it cover both scenarios...

Once set for the tool it held which is intended.

eightheads, 2019-11-17:
> Ill take a look at the tool change code. Originally this was written that if no fan speed was stored it would carry over the current.
>
> There was some debate on that one if they explicitly wanted 0 [...]

What would be the proper code to set both to a set speed at once?

Roxy-3D, 2019-11-17:
> What would be the proper code to set both to a set speed at once?

Until this sorts itself out and the community agrees on a direction...   Probably the best thing to do is change your Slic3r's Start Up GCode to set an explicit (and correct) fan speed for each of your tools.

That way...   no matter what gets decided by the community with respect to defaults and transitions...  Your platform will be doing the right thing for you.

eightheads, 2019-11-17:
> Until this sorts itself out and the community agrees on a direction... Probably the best thing to do is change your Slic3r's Start Up GCode to set an explicit (and correct) fan speed for each of [...]

Would that be:
M106 P0 S255
M106 P1 S255
?

Roxy-3D, 2019-11-17:
That will turn both fans on at the highest possible speed.   If that is what you want...  That is what that will do.

eightheads, 2019-11-17:
> That will turn both fans on at the highest possible speed. If that is what you want... That is what that will do.

There is only a single party cooling fan. I have been using M106 S255 but Marlin kills it after the tool change.

Roxy-3D, 2019-11-17:
If your 'Fan Count' is 1, I believe that is still correct.   When you switch tools, the other speed will be used.    But of course...  Your Milage May Vary.    Please check it out and see if things work as expected.

InsanityAutomation, 2019-11-17:
Fan1 is essentially taken over as the 2nd extruder fan, so fan0 for extruder 0, fan 1 for extruder 1. This basically allows singlenozzle to behave exactly like a dual hotend machine without complicated scripts in most cases. Just make sure a fan speed is set for each tool. Best method right now is to configure it as if it was a second hotend in youre slicer then everything just works as expected.

eightheads, 2019-11-18:
> Fan1 is essentially taken over as the 2nd extruder fan, so fan0 for extruder 0, fan 1 for extruder 1. This basically allows singlenozzle to behave exactly like a dual hotend machine without [...]

Replacing 
M106 S255 
with
M106 P0 S255
M106 P1 S255
does turn the part cooling fan on and it stays on throughout the print. 

I guess I will have to add a script to Simplify3D to handle this automatically as there are not separate extruder fan definitions currently.

boelle, 2019-11-25:
@InsanityAutomation as the noob i am i ask if things are working as intended or if there is a bug as such?

boelle, 2019-11-26:
@eightheads btw what version of marlin are you using?

eightheads, 2019-11-26:
> @eightheads btw what version of marlin are you using?

I'm currently on a pull from 11/18

boelle, 2019-11-26:
1.1.9?

eightheads, 2019-11-26:
> 1.1.9?

2.0

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle

VanessaE, 2020-03-16:
@boelle  yes it is in fact still an issue, at least up to Marlin commit 484e1a624c178f1bb0996892d2c606f6cfebefbe.  Multiple extruders feeding one hotend (single nozzle) via a Y splitter, with exactly 1 part cooling fan.

Please re-open (you'll notice the last-referenced Prusaslicer issue is also still open).