MarlinFirmware/Marlin #15878: [FR] support analog sensor like FSR and Piezo

Zezoh, 2019-11-12:
Hello, 

Thank You for Your Hard Work, 

I would like to know if it is possible to support analog sensors like FSR and Piezo for zprobe, using the extra TH pin. 

I know that there are many options available like using external board etc, but I think it will be great if we can integrate it within Marlin FW. 

Suggested scenario:  

for Configuration.h :
under zprobe section 
#define FSR_Zprobe || Piezo_Zprobe  // or simply #define analog sensor for zprobe
#define analog sensor_pin
#define sensitivity 95 

For LCD: 
Analog sensor status ( if triggered or not ) 
Adjust sensitivity ( 0 -100 ) 

During boot, the printer will take the reading and store it as initial value

pyr0ball, 2019-11-12:
Doubtful for either, as the pullup resistor attached to that input will basically drown the pulse from the piezo element or any velostat-like device into nothing.

Attaching it to a non-pulled analog pin might be possible if:
 - The analog pin on the MCU is capable of high enough impedance
 - There's enough EMI shielding to eliminate noise on the line
 ~OR~
 - The Piezo/FSR is attached using an OpAmp that lives near the piezo element itself

Zezoh, 2019-11-12:
I have a delta style printer from dagoma.fr they are using FSR under bed.

How they get it with MKS_BASE board !!

Please check their modified copy of Marlin: 

[https://dist.dagoma.fr/get/sources/Marlin-Firmwares/1554222799/354ce2dc42499e971d723fb40dc809cf](url)

pyr0ball, 2019-11-12:
Can you check that link? It appears to be broken.

Which version of the MKS Base?

Any links to schematics or wiring diagrams for how they've implemented the FSR would be appreciated.

I'm betting they're using a daughterboard with an opamp network or maybe even just a straight PrecisionPiezo.

Zezoh, 2019-11-12:
The above link fixed.

MKS_BASE 1.0 +, the one I have is 1.5.

I think they just started to build it on v2.0 please check it here:

https://github.com/dagoma3d/Marlin/commit/93572471ad83da747055a163395d9e28ea85ea46#diff-f3f16ada73f2765518ee060b71e3de26

From the pins file you can find out the wiring, they defined Z_MIN_PROBE pin  69 , while FSR to pin 15. as reading / writing., once readings  within limits send trigger signal to Z_MIN.

Most of the work done in temperature.cpp file, for FSR reading. 

for more into you can check these videos: 

https://www.youtube.com/watch?v=JX2lC7f3KsA&t=93s
https://www.youtube.com/watch?v=tuzsbaajuBc&t=695s

Regards,

Zezoh, 2019-11-20:
No one interested in this feature :(

pyr0ball, 2019-11-20:
I am but I'm on vacation for a couple of weeks.

Zezoh, 2019-11-21:
Happy to know that :) 
@pyr0ball enjoy your vacation

coolcheat, 2020-03-15:
If possible, I am also interested in this feature (:

Zezoh, 2020-03-16:
@coolcheat 

I did it with Marlin 1.1.9

Please check below link: 

https://app.box.com/s/qvnxmbll9doqg4469b4bbgmk8i6yplbi

pyr0ball, 2020-03-16:
@Zezoh any chance you have this on a forked repository? It's much harder to check differences in code from a zipped copy

InsanityAutomation, 2020-03-16:
I've been in some discussion with dagoma on this, and their dev is on here occasionally. It's been ported to 2.0 and will show up in time, just alot in the queue as always.

Zezoh, 2020-03-16:
@pyr0ball please check below link:

https://github.com/Zezoh/Marlin/tree/bugfix-1.1.x

coolcheat, 2020-03-16:
@Zezoh  thanks for sharing!

But I'm actually on 2.x.bugfix so I guess I'll have to wait a bit until it's there (if I'm understanding @InsanityAutomation's comment).

Thanks.

coolcheat, 2020-06-24:
Would anyone know if this has been added to 2.x.bugfix yet?

Thanks.

pyr0ball, 2020-06-24:
@coolcheat to my understanding @InsanityAutomation is currently working on adding support, but it's not been added as of yet

0r31, 2020-06-29:
Hi,

@pyr0ball  and @coolcheat : The FSR integration implementation is here : https://github.com/dagoma3d/Marlin/commit/93572471ad83da747055a163395d9e28ea85ea46 (I guess that's where @Zezoh found the code he mentioned when opening this thread).

As @InsanityAutomation stated, i talked with him about this topic. **It is still experimental**. It works but could be improved (especially in term of implementation genericity + additional features such as saving the threshold to the eeprom).

Moreover, i've also worked on an alternative implementation: 
- The current one uses the temperature ISR to handle the analog raw value and trigger when needed (threshold based triggering).
- The alternative one doesn't use the temperature ISR anymore, all the stuff is directly done in endstops code. **It is still in testing phase** and is only applicable for some specific boards.

Regards,
Orel

pyr0ball, 2020-06-29:
@0r31 The alternative one using endstops would probably require a certain amount of signal conditioning for a good read. Without some amount of conditioning, you'll either end up with an extremely noisy input that could trigger at inopportune times, or will simply not ever trip due to low impedance dropout and/or pullup/pulldown resistors.