MarlinFirmware/Marlin #15880: [BUG] Enabling UBL prevents compile when using EEPROM Emulation (e.g. BTT SKR E3 DIP)

pyr0ball, 2019-11-12:
### Description

SanityCheck flags cause compile to error when UBL is enabled on a board that requires FLASH_EEPROM_EMULATION

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3837937/Marlin.zip)

1. Disable EEPROM_SETTINGS
2. Enable FLASH_EEPROM_EMULATION & AUTO_BED_LEVELING_UBL
3. Build

**Expected behavior:** 
Compilation completes without error

**Actual behavior:** [What actually happens]
```
\marlin\marlin\src\inc\SanityCheck.h:1243:6: error: #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
     #error "AUTO_BED_LEVELING_UBL requires EEPROM_SETTINGS. Please update your configuration."
```

thisiskeithb, 2019-11-12:
You still need to enable `EEPROM_SETTINGS`.

boelle, 2019-11-13:
will close this at its very clear that its a user error and not a bug

closed 2019-11-13 by boelle