MarlinFirmware/Marlin #15881: Tag for Marlin 1.1.9 changed / points to wrong commit

xblax, 2019-11-12:
I was trying to diff the official Marlin 1.1.9 version from Github against a modified Marlin version provided by a third party. I was very confused at first about the number of changes.

I then took a look at the 1.1.9 release page that is dated to Jul 31, 2018 https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.9 and noticed that the actual 1.1.9 tag points to a commit from Sep 2 (author date Aug 25) on the 1.1.x branch: https://github.com/MarlinFirmware/Marlin/commit/0110faa717e642574ade9eb7aed250ee2840e942

Looking at some forks I found out the tag is actually supposed to point to https://github.com/MarlinFirmware/Marlin/commit/824edc049bb1ede7eade0b98d6366cda102d2af0

I assume this was unintentional (otherwise it would be very bad practice to change a tag for a released version).

boelle, 2019-11-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

thisiskeithb, 2019-11-13:
@boelle: This is an issue with Github/Marlin’s release tagging and should be looked into.

boelle, 2019-11-13:
@thinkyhead to to fix this one?

thisiskeithb, 2019-11-13:
Why are you editing my comments?

boelle, 2019-11-13:
it just made more sense after i reopened it
but will not do again

thisiskeithb, 2019-11-13:
Editing my comment to remove a mention is unnecessary. Please don't do that again.

thinkyhead, 2019-11-14:
Thanks for the heads-up @xblax. I've moved the tag to the HEAD of 1.1.x.

closed 2019-11-14 by thinkyhead

xblax, 2019-11-14:
@thinkyhead 
That's not what I meant. The version tag should not be different from what was released originally. If I check out 1.1.9 I should get 1.1.9 as it originally was on Sep 22 2018. The same way I get 1.1.8 if I check out that tag now.

If enough fixes have gone onto the 1.1.x branch since then that would probably justify the release of 1.1.10 or 1.1.9.1.