MarlinFirmware/Marlin #15882: [BUG] (sKr v1.3 "USB device not recognized after uploading Marlin )

Zatarines, 2019-11-12:
Howdy All, 

    I am replacing my Ultimaker v2.1.4 board with SKR v1.3 w/TMC2208 UART. I am not able to connected any Win10 PC to the SKR board via USB after re-flashing to Marlin 2.0bugfix. Getting a Win10 message" USB device not recognized. the last usb device you connected to this computer malfunctioned, and win does not recognize it." I tried to uninstall and reinstall the driver and that didnt work.  However, When I use the original firmware from Bigtreetech without my UM2 clone settings, the SKR board recognizes fine in the Win10. On the other hand, when I set my printer's settings to the Bigtreetech's Marlin version, it compiles an error message for Pins related to the Ulti_controller. 

Between the two versions, I preferred to troubleshoot this Marlin version but Inconveniently,  Marlin 2.0bugfix compiles all my UM2 settings without error, but I does not connected the board with USB. 

- I tried so far changing the in configuration.h: #define SERIAL_PORT 0, -1

- configuration_adv.h   //  #define SDCARD_CONNECTION ONBOARD , every possible combination.

- Tried to define a different controller in case that was the issue. 

- re-install USB driver on Win10

Can someone relate to this issue and point me in the right direction ?
thank you in advance,

3Dangel, 2019-11-12:
you would probably need to attache your configuration.h for ppl here to help you....my cent

3Dangel, 2019-11-13:
mine is set to:

#define SERIAL_PORT -1
 */
#define SERIAL_PORT_2 0

Zatarines, 2019-11-13:
@3Dangel 
Unfortunately it is not that, I did define it likewise and even tried switching order. It is not that

Zatarines, 2019-11-13:
I followed several guides on how to set it up, they have Very similar instructions but mine set identical don't work.

3Dangel, 2019-11-13:
but do you have the original sd card that came with your v1.3?
rename the .CUR back to bin and try it again and see if that gives you connection to the PC

Zatarines, 2019-11-13:
@3Dangel 
If I use the .cur original file the USB connects without any issue. I’ve done that before to assure that it was HW

3Dangel, 2019-11-13:
have you tried to use SKR's github's?
or it was from here?
I would try from SKR's and see if that works
if it does, then compare your configuration.h to see if any difference... 
sorry that I won't be any more hep that this....

but for sure, if you still have issue with the FW, you would need to upload your config file here...

cheers

Zatarines, 2019-11-13:
@3Dangel 
my bad on the huge txt, it is all 

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
[... 2116 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

ellensp, 2019-11-13:
I've found that the supplied 256meg sd card is unreliable. I found that uploading new firmware to it results in currupting the fimrware on the controller and then usb doesnt work any more.  Try different sd cards.

Zatarines, 2019-11-13:
@ellensp 
I am not uploading with the supplied sd card. I read in some post people resolved just switching SD cards, but in my case not a solution. thks

3Dangel, 2019-11-13:
You have a lot of functions enabled. 
Why fon’t you try ONLY the basic settings and say enable repeal full grading controller. ( I saw you have the ultimaker’s enabled )
Just upload the FW with the very basics item and see if that gets the connection. 
It might be something conflicting. 
I suggest you to use SKR’s FW so it is download and compile and test. 
If it it good then you need to find line by line what you have enabled to find the one that caused issue. 

Sleep time 01:04 am
Good luck

thinkyhead, 2019-11-13:
@Zatarines — To provide configurations, you ZIP them and drop the ZIP file on your reply. These text boxes treat some characters as special formatting, and pasting config text just makes a mess.

Zatarines, 2019-11-13:
@3Dangel 
when you mention SKR's FW, is that from the Bigtreetech Marlin version? I tried that and it gives a compiling error for the Ulti_board pins.

Zatarines, 2019-11-13:
Maybe this Skr v1.3 is worth crap, I looked into their bigtechtree github ''issues'' a dozen of opened issues that not one finds the solution. Is it a coincidence? maybe, but I have just got a board replacement from Amazon and it is the same issue; after reflashing the Marlin the board does not respond to the USB

3Dangel, 2019-11-14:
My board came from AliExpress
Without issue.
And flashed it back and forth with different settings.
You did not mention if new downloaded FW from Marlin and enabling minimal
settings, would it still the same?

Zatarines, 2019-11-14:
@3Dangel 
It did work with new downloaded FW and minimal settings(just the BOARD_BIGTREE_SKR_V1_3 changed), but if I change anything else in the FW and reupload, the USB does recognize. Weird, right?

3Dangel, 2019-11-14:
Then it is not the HW issue.
And you will be the one to find out what it is triggering the conflict.
Once you find out what causes, then reopen the bug report , I supposed.

Glad you find that out!!!

Zatarines, 2019-11-14:
@3Dangel 
thank you. are you able to my FW setting in your environment to see if you get the same issue?

3Dangel, 2019-11-14:
I’m away from home for work for 10days.
I won’t mind to try and help out!
My board is still outside.

But I think it is an issue of the settings with SD, or LCD display.

Cheers

Cid427, 2019-11-17:
"re-install USB driver on Win10" - This may possibly be the problem.

Windows 10 should not require any USB drivers to be installed.

Reference bottom of page 14 of the SKR1.3 manual on Big Tree Tech Github.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BIGTREETECH%20SKRV1.3%20guide%202019%20-%206.pdf

Zatarines, 2019-11-18:
Howdy all, 
  It looks like the reason that I am having the issue with the USB not connecting after flashing FW with Marlin2.bigfix is that when the #define ULTI_CONTROLLER is enabled there's some sort of Bug. Because I tried the configuration with  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled instead and it works fine. Is this a Bug in the firmware? @thinkyhead 

Thank you in advance for the attention,

boelle, 2019-11-24:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Zatarines, 2019-12-01:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3907874/Marlin.zip)
I notice that when the Ultiboard controller is not connected the SKR, the USB seems to work fine, but as soon as I connect the EXP1 and EXP2 the computer no longer recognizes the USB connection.

ManuelMcLure, 2019-12-01:
That’s probably the display drawing too much current.

boelle, 2019-12-02:
@Zatarines does it work without display?

Zatarines, 2019-12-03:
@boelle 
The USB connection does work withou the display plugged onto the board. I tried switching EXP1 and EXP2 and reversing many combinations, but nothing seems to make a difference. Also, while doing the switching and reverse that LCD never shows signal of being powered at all. The LCD is working with the UM2 board.
regards,

boelle, 2019-12-03:
> That’s probably the display drawing too much current.

Zatarines, 2019-12-03:
I am using the same power supply that I had on the original board. if drawing too much current, is there any proactive approach to it ?

boelle, 2019-12-03:
if you can try with a different display

Zatarines, 2019-12-03:
I am able to use a rep-rap full display and TFT24 without display issues. Or you meant? "using different display with Ultimaker display firmaware settings"

boelle, 2019-12-03:
no just different hardware, one of your display are drawing so much current that usb stops working

Zatarines, 2019-12-03:
I am using one at time. For testing purposes I used a TFT24 that works okay, then tested the Raprep fulldisplay and it works alright. 
The issue is that the Ultimaker controller does not turn on at all with the same hardware and causes the SKR board to not recognize the USB when plugged to the computer. Also showing the same symptoms even with the power supply connected to the hardware.

boelle, 2019-12-03:
> I am using one at time.

yes and one of the display draws to much current and the usb stops working

Dont use the display that makes usb not work

Zatarines, 2019-12-03:
THe HW is a Ultimaker 2 printer which has the display in question attached to it. The display works fine with previous versions of marlin and the original ultimaker board. I needed to replace the mainboard because it fried some components for the temp sensor. Now, I am working with Marlin2 for the SKR 32bit board. Hence, I opened this issue.

3Dangel, 2019-12-03:
So there is the new variable here which is the SKR board.
So it is not the “same” thing.
Most likely the SKR is not able to support much of sharing power to your
LCD hence USB stops working.
So in the end is not the Marlin issue. It is hardware issue and it is not
the same main board you are using and comparing to.

Cheers

boelle, 2019-12-03:
@3Dangel i bet the next thing you will hear is that it used to work fine with old version of marlin :-D

TerawattX, 2019-12-04:
I had this issue when compiling with VSCode and PlatformIO 4.1.0.  There were a few posts about this issue on the SKR repo (which looked like it wasn’t mostly the regular Marlin code with some additional files).  Ultimately the solution was to downgrade PlatformIO to 4.0.3 and everything worked normally.

I also noticed PlatformIO has a 4.1.1 in the works that fixes issues with some boards, so I suspect it might be related.

boelle, 2019-12-07:
@Zatarines did downgrade platformio make any difference?

boelle, 2019-12-07:
@Zatarines since 2.0 was just released a few days ago has this changed this issue at all?

Zatarines, 2019-12-07:
@boelle 
I did just tried with the new release yesterday, and I tried with a new board Mks sgen_l 1.0 with the same issue. I am not sure where to start but I am trying to understand if the Exp1 and exp2 pin assignment in Marlin 2.0 is the matching the sequence of the ulticontroller specsheet.

boelle, 2019-12-07:
and you are sure that you connect the display corretly? sometimes the connectors are rotated 180 degree and will not work

Zatarines, 2019-12-07:
I tried inventing them as well in all kind of positions. the screen doesn't show any sign of life

boelle, 2019-12-07:
its just strange that it works when not connecting the display

and its not bad because it works on other boards..... 

the only thing i can see is that the other boards are able to supply more current and hence it works on other boards

thiscodedbox, 2019-12-18:
Hi, I'm experiencing the same problem, in 2.0.0 release, replacing a UM2+ stock board with an SKR v1.3, and the LCD is not working. I worked through the pinouts on the Ulticontroller to try to determine the issue, then I wired it up with individual jumpers to test. There is one pin that stops the USB detection, that's pin 3 on EXP 2, labeled reset.  Oddly enough, if I spin the selector switch, it detects the USB every once in a while. I'm investigating further, my config is not unique; default with #define ULTI_CONTROLLER uncommented. I don't think it's a power issue, but I'll try external 5v on the Ulticontroller to eliminate that possibility.

Zatarines, 2019-12-18:
@thiscodedbox 

   Did you get the screen to turn on at all?

thiscodedbox, 2019-12-18:
No, not at all, no backlight, nothing.

Zatarines, 2019-12-18:
I am following your posts to see if your external 5v test eliminate the possibility .

thiscodedbox, 2019-12-19:
External power doesn't seem to help, I'm suspicious of the 3.3v the Ulticontroller needs, I tried plumbing the 3.3v from the ICSP header because it measured closer to 2.7 on the appropriate pin on EXT2.  My goal is to get the display working first as I think I already have the encoder/button working when I bypass pin 3 on EXT2 as mentioned above.  Since I still have a working Ultimaker board, I broke out the connector to see the bare minimum connections needed to make the screen work:

(this is from the Ulticontroller PCB Hooked up to the original UM2 PCB running the latest tinkergnome):

```
             EXP1                           EXP2
 
            -----                           -----  
    (5v) x | 0 9 | x (GND)                 | 0 9 |
           | 8 7 | x (reset)      (3.3v) x | 8 7 |
  (scl5) x | 6 5 |                         | 6 5 | 
  (sda5) x | 4 3 |                         | 4 3 |
           | 2 1 |                         | 2 1 | 
            -----                           -----

```

Now I'm going to try to make the display work on the SKR with just these pins in play to reduce complication.

Zatarines, 2019-12-23:
@thiscodedbox 
where you able to make the screen work on your SKR?

thiscodedbox, 2019-12-23:
No, unfortunately i continued to have the same issues. There are few issues open and closed regarding using the E1 headers for hardware spi to run i2c OLED displays, but it's beyond my ability at this time. I'm likely going to go with a different display.

boelle, 2020-01-07:
@Zatarines is this still a problem?

Zatarines, 2020-01-07:
@boelle yes, still having the same issue.

Grogyan, 2020-01-07:
@Zatarines Bigtree tech have released the SKR1.4, which is pin for pin identical to the SKR1.3.

It does however have a separate header for the I2C.
Maybe worth trying?

boelle, 2020-01-10:
@Zatarines so you use 2.0.1 or bugfix 2.0.x ?

Zatarines, 2020-01-10:
I am using bugfix 2.0x

boelle, 2020-01-23:
how often do you update? bugfix is updated almost daily

Zatarines, 2020-01-24:
@boelle 
I changed the board to a Mks lgen, please close this issue.

closed 2020-01-24 by boelle