MarlinFirmware/Marlin #15883: LPC1768: dtostrf function not working, (arduino floating point not working)

robbycandra, 2019-11-13:
**Description**

I use LPC1768
dtostrf function not working, because sprintf is not working for floating point.

**Steps to Reproduce**

Try M114 D with M114_DETAIL enabled
It also affect PowerLoss Recovery

thinkyhead, 2019-11-13:
Strange. Arduino recommends using `dtostrf` because `%f` is not supported by the `printf` functions.

thinkyhead, 2019-11-13:
This is what the LPC platform's `arduino.cpp` does for `dtostrf`…
```cpp
char *dtostrf (double __val, signed char __width, unsigned char __prec, char *__s) {
  char format_string[20];
  snprintf(format_string, 20, "%%%d.%df", __width, __prec);
  sprintf(__s, format_string, __val);
  return __s;
}
```
…and sure enough, it is trying to use `%f`. Looks like an LPC platform bug.

thinkyhead, 2019-11-13:
Looking at other platforms, I see that most of them implement `dtostrf` in just the same way. Considering that `dtostrf` has been working, and this includes the LPC1768 machines I am testing, then perhaps some breaking change has recently been made to the LPC platform…?

p3p, 2019-11-13:
It was my fault, I disabled printf float support, thinking that I way using dtostrf from the link `m` library, .. I'm sure I had a good reason to implement it this way originally.

thinkyhead, 2019-11-13:
> I disabled `printf` `float` support

Where is that disabled?

p3p, 2019-11-13:
In the platform, just patched it in release 0.1.2

robbycandra, 2019-11-13:
Problem Fixed; Thanks

closed 2019-11-13 by robbycandra